networkupstools/nut #1321: input voltage = 256 when no input

Mic-Pod, 2022-03-01:
Hi,
I have noticed that when power goes down, nut is showing `input.voltage: 256.0` instead of `input.voltage: 0`
OS: Ubuntu Server 20.04.4
NUT version: 2.7.4

`upsc ups
input.voltage: 256.0
input.voltage.nominal: 230
ups.status: OB DISCHRG`

Mic-Pod, 2022-03-01:
When using apcupsd, values are displayed correctly:

```
apcaccess status
DATE     : 2022-03-01 20:14:45 +0100
STATUS   : ONLINE
LINEV    : 230.0 Volts
```

```
apcaccess status
DATE     : 2022-03-01 20:15:33 +0100
STATUS   : ONBATT
LINEV    : 0.0 Volts
```

jimklimov, 2022-03-02:
Thinking of APC, could this be another case of LogMin/LogMax mismatch?
However if this is a packaged "vanilla" NUT 2.7.4 release, it might be not
as strict about this ***@***.*** fixed that later) and the problem could be
elsewhere...

> Message ID: ***@***.***>
>

Mic-Pod, 2022-03-07:
Is there anything I can do to help troubleshoot this issue?

jimklimov, 2022-04-24:
I think one idea for troubleshooting is to try a local build from current NUT master branch, just to check if recent years of bug-fixes solved this one too.