networkupstools/nut #1322: Eaton Ellipse 1200 Pro - No matching HID  UPS found

Impatience71, 2022-03-02:
Hi all, 
I tried to connect my new Eaton Ellipse Pro 1200 with the standard installation of Nut. 
(UPS driver controller 2.7.4, Generic HID driver 0.41, USB communication driver 0.33) on a raspberry 4 (latest bullseye armhf version)
lsusb shows: 
Bus 001 Device 004: ID 0463:ffff MGE UPS Systems UPS

The ups.conf is configured very simple:

`[eaton1]
       driver = usbhid-ups
       port = auto
       desc = "Eaton Ellipse Pro 1200"
       vendorid = "0643"
       productid = "FFFF"
       serial = "xxxxxxxxxx"
`
And the output of upsdrvctl  is: 

pi@raspberrypi:/etc/nut $ sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Driver failed to start (exit status=1)

Whats my mistake ?
Thanks Peter

jimklimov, 2022-03-04:
Some issues could be due to USB devfs permissions (are udev rules set up to
let nut runtime user r/w access to the device?) which can be worked around
by starting the driver with `-u root` in the worst case, or a `user=root`
in `ups.conf` global section.

More errors and causes can be seen with added `-D` options for debug
verbosity, but note that in case of `upsdrvctl` it means this wrapper's
verbosity - not the driver programs' one (at some point wrapper logs the
command it launches drivers by - try running that directly and add `-DDDDD`
there).

Finally, the latest release (and so packages) is a bit dated, much work is
wrapping up for a next one. If you can build custom NUT locally from
current master branch (wiki and several recent issues detail how), that
could be helpful too.

Impatience71, 2022-03-04:
Hi Jim, 
many thanks for your thoughts. I made the  following: 
- download the latest version (master) and compiled and installed all 

The permissions in /dev/bus/* are like this: 

`pi@raspberrypi:/dev/bus/usb/001 $ ls -al
insgesamt 0
drwxr-xr-x 2 root root    140  3. Mär 19:10 .
drwxr-xr-x 4 root root     80  1. Jan 1970  ..
crw-rw-r-- 1 root nut  189, 0  4. Mär 13:43 001
crw-rw-r-- 1 root nut  189, 1  4. Mär 13:43 002
crw-rw-r-- 1 root nut  189, 2  4. Mär 13:43 003
crw-rw-r-- 1 root nut  189, 3  4. Mär 13:46 004
crw-rw-r-- 1 root nut  189, 4  4. Mär 13:43 005`

The nut-scanner gives the following (no surprise, worked also before)
`pi@raspberrypi:/usr/local/ups/bin $ sudo ./nut-scanner 
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
[nutdev1]
	driver = "usbhid-ups"
	port = "auto"
	vendorid = "0463"
	productid = "FFFF"
	product = "Ellipse PRO"
	serial = "G361M36084"
	vendor = "EATON"
	bus = "001"`

Then I tried to run the usbhid-ups driver direct: 

`pi@raspberrypi:/usr/local/ups/bin $ sudo ./usbhid-ups -u root -DDDDDDD -a eaton1
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4757-g9e5e34d3)
USB communication driver (libusb 1.0) 0.42
   0.000000	[D3] do_global_args: var='pollintervall' val='1'
   0.000155	[D3] do_global_args: var='maxretry' val='3'
   0.000236	[D3] main_arg: var='driver' val='usbhid-ups'
   0.000306	[D3] main_arg: var='port' val='auto'
   0.000428	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000536	[D3] main_arg: var='desc' val='Eaton Ellipse Pro 1200'
   0.000631	[D3] main_arg: var='vendorid' val='643'
   0.000727	[D5] send_to_all: SETINFO driver.parameter.vendorid "643"
   0.000799	[D3] main_arg: var='productid' val='FFFF'
   0.000871	[D5] send_to_all: SETINFO driver.parameter.productid "FFFF"
   0.000934	[D3] main_arg: var='serial' val='G361M36084'
   0.001029	[D5] send_to_all: SETINFO driver.parameter.serial "G361M36084"
   0.001180	[D1] debug level is '7'
   0.003184	[D5] send_to_all: SETINFO device.type "ups"
   0.003273	[D2] Initializing an USB-connected UPS with library libusb-1.0.24 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.42')
   0.003329	[D1] upsdrv_initups (non-SHUT)...
   0.031529	[D2] Checking device 1 of 6 (1D6B/0003)
   0.033036	[D2] - VendorID: 1d6b
   0.033106	[D2] - ProductID: 0003
   0.033163	[D2] - Manufacturer: Linux 5.10.63-v7l+ xhci-hcd
   0.033227	[D2] - Product: xHCI Host Controller
   0.033287	[D2] - Serial Number: 0000:01:00.0
   0.033342	[D2] - Bus: 002
   0.033401	[D2] - Device: unknown
   0.033458	[D2] - Device release number: 0510
   0.033519	[D2] Trying to match device
   0.033579	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.033655	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.033716	[D2] Device does not match - skipping
   0.034246	[D2] Checking device 2 of 6 (1E7D/3201)
   0.038221	[D2] - VendorID: 1e7d
   0.038291	[D2] - ProductID: 3201
   0.038347	[D2] - Manufacturer: ROCCAT
   0.038406	[D2] - Product: ROCCAT Valo
   0.038464	[D2] - Serial Number: unknown
[... 44 lines cut ...]
   0.445585	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.445764	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.445937	[D2] Device does not match - skipping
   0.446156	[D2] Checking device 6 of 6 (1D6B/0002)
   0.447089	[D2] - VendorID: 1d6b
   0.447269	[D2] - ProductID: 0002
   0.447442	[D2] - Manufacturer: Linux 5.10.63-v7l+ xhci-hcd
   0.447621	[D2] - Product: xHCI Host Controller
   0.447799	[D2] - Serial Number: 0000:01:00.0
   0.447970	[D2] - Bus: 001
   0.448141	[D2] - Device: unknown
   0.448320	[D2] - Device release number: 0510
   0.448497	[D2] Trying to match device
   0.448668	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.448851	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.449024	[D2] Device does not match - skipping
   0.449249	[D2] libusb1: No appropriate HID device found
   0.449415	No matching HID UPS found`

Whats irritating me is that the vendor ID is one time "0463" and the second time just "463" Dont know if that plays a role bcs the strings are not identical. 

Do you have an idea what I can try next ? 

Thanks a  lot for your help !
Kr Peter

Impatience71, 2022-03-04:
David Zomaya gave me the hint to remove product and vendor ID from the ups.conf. I did that and I have no errors anymore when the usbhid-ups tries to talk with the UPS. Thats a huge step forward. Will try now to bring the other pieces to run. 
Thanks all for your support !

Kind regards Peter

Grandma-Betty, 2022-03-10:
Seems you're having the same issue as me on ARM based OS:
https://github.com/networkupstools/nut/issues/1004

Impatience71, 2022-03-11:
Yeah, think so. I have the issue on two different raspi 4. The driver itself (usbhid-ups) provides many HID objects. Everything looks ok. But the upsdrvctl runs into "No matching HID UPS found" error.  No idea what can be the reason.  I thought that the upsdrvctl is only a wrapper around the selected ups driver. But seems not.

Impatience71, 2022-03-16:
Hi Jim, 
your hint was the real one ! I let the upsdrvctl run with " -u root" and it works. I thought the sudo is enough, but it was not. Need to make all start and stop procedures round on the weekend and integrate it with Grafana on my HA but then I am happy. Thanks a lot for your support and the time you spent !
Great community.  Wish you a nice evening.

jimklimov, 2022-04-24:
The original post was poorly formatted for markdown so I did not read it right the first time.

Seems there was a typo in original ups.conf section, with `vendorid = "0643"` (not `0463`) so *that* did not match, I suppose.

jimklimov, 2022-04-24:
Regarding running as root - that is a fallback option if nothing else helps easily, but not a "proper" one (and potentially not safe, to have a root daemon in the system when it does not require to be).

Usually you should configure the udev rules or equivalent, which would tell the OS to set permissions for the device-FS node for NUT run-time accounts; distro-packaged NUT usually installs that into the OS, docs in source should tell how to do the same.

Note that recent iterations of the master branch include #1342 for more efficiently parsed "hwdb" instead of text-format rules. Practical testing how this is supported across the board would be welcome :)

SmartPhoneLover, 2025-05-16:
Did it improved somehow in the latest 2.8.3 NUT release? I'm about to receive my new 1200 PRO version next week.

jimklimov, 2025-05-16:
Also note there's a regression found recently with NUT v2.8.3 that can cause `usbhid-ups` to not process device poweroff commands by default, see #2955. This bit worked in 2.8.2 and checked with the fix in that PR too.