networkupstools/nut #1324: Account for POWERDOWNFLAG in nut shutdown script

sonicaj, 2022-03-06:
NUT debian package right now does not account for `POWERDOWNFLAG` in https://github.com/networkupstools/nut/blob/master/scripts/systemd/nutshutdown.in the shutdown script.

Let me know please if something else is required from my end. Thanks!

jimklimov, 2022-03-11:
Maybe I misunderstood the question, but the `POWERDOWNFLAG` is a configuration option (pathname for a touch-file)  of `upsmon`. The referenced script calls `upsmon -K` to check if kill-power mode was requested and then calls `upsdrvctl shutdown` to tell UPS to cut power and maybe restart later.

It may have been the case in much older NUT versions that the filename would be "hard-coded" and directly consulted by shutdown scripts, but it is not required to be so direct anymore.

sonicaj, 2022-03-12:
@jimklimov yes that is accurate that it is a configuration option. My concern is that the configuration option is not respected in standard debian installation, like we should only shutdown if kill power mode is requested and we have the file specified in `POWERDOWNFLAG` available.
```
upsmon creates this file when running in primary mode when the UPS needs to be powered off. You should check for this file in your shutdown scripts and call upsdrvctl shutdown if it exists.

This is done to forcibly reset the secondary systems, so they don’t get stuck at the "halted" stage even if the power returns during the shutdown process. This usually does not work well on contact-closure UPSes that use the genericups driver.
```

If we look at current shutdown script, we have
```
❯ cat /usr/lib/systemd/system-shutdown/nutshutdown
#!/bin/sh
/sbin/upsmon -K >/dev/null 2>&1 && /sbin/upsdrvctl shutdown
❯
```
which shows that `POWERDOWNFLAG` is not respected. We should account for `POWERDOWNFLAG` in this systemd generated script.

jimklimov, 2022-03-12:
What I meant (and if my memory did not fail me) was that on one hand
`upsmon` as a (primary/master) daemon uses the setting to touch that file
if it decides to power down the rack. On another, `upsmon -K` as a
command-line tool checks if the file from setting exists, and returns
success or failure. So it is effectively an internal implementation detail
now.

Why should the string be baked into the (generated) script? What if a
deployment changes the path by editing their upsmon.conf file - should the
script parse it?

sonicaj, 2022-03-12:
@jimklimov i am totally fine with `upsmon -K` handling this internally, but the problem is that it is not happening. Please have a peek at https://jira.ixsystems.com/browse/NAS-115025. Let me know if you would require something from my end. Thanks!

jimklimov, 2022-03-24:
Weird, I think I posted back a few days ago... but seems got to retry:

I believe with the config-change detailed in that issue, they tell `upsmon` to touch one filename instead of another. Then `upsmon -K` checks that this name exists and tells the UPS to shut off. NUT is working as intended and documented, but not how the use-case intended. Seems like another case of "Computers do what you tell them to, not what you wanted them to" ;)

I think one way to avoid `upsmon` telling the UPS to kill power is to set it up as slave/secondary `upsmon` watching an UPS exposed by the driver and server running on localhost. IIRC this trick was recommended for server racks watching an SNMP-connected UPS directly, but only one server actually responsible for power cycling.

sonicaj, 2022-04-12:
I see, thank you for clarifying @jimklimov. I was imagining a different usage but that's not true. Thanks again!

closed 2022-04-12 by sonicaj

jimklimov, 2022-10-11:
I believe with the config-change detailed in that issue, they tell `upsmon`
to touch one filename instead of another. Then `upsmon -K` checks that this
name exists and tells the UPS to shut off. NUT is working as intended and
documented, but not how the use-case intended. Seems like another case of
"Computers do what you tell them to, not what you wanted them to" ;)

I think one way to avoid `upsmon` telling the UPS to kill power is to set
it up as slave/secondary `upsmon` watching an UPS exposed by the driver and
server running on localhost. IIRC this trick was recommended for server
racks watching an SNMP-connected UPS directly, but only one server actually
responsible for power cycling.