networkupstools/nut #1326: Unable to communicate with Fry's Electronics MEC0003

dresnu, 2022-03-10:
Hello, I cannot get nut to work with a no name UPS that identifies as "Fry's Electronics".

Any help would be welcome. This is the output of the usb device query:

Bus 001 Device 006: ID 0001:0000 Fry's Electronics MEC0003
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000
  bcdDevice            1.00
  iManufacturer           0
  iProduct                1 MEC0003
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     624
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10

jimklimov, 2022-03-11:
Please check if `nutdrv_qx` or `nutdrv_atcl_usb` would handle it? Unfortunately, the very non-random vendorid/productid here do not give reliable pointers to what the device really is and what protocol(s) it talks. If it really is the MEC as reported above, AFAIK that is the USB interface chip which is not related to a particular protocol in use.

dresnu, 2022-03-12:
I have already tried the two drivers you suggested, with multiple options, but the readings are always 0. The UPS came with a cd containing a program called UPSMART, which supposedly has a linux implementation too, but does not work. On windows in order to make it communicate I have to use a MEGA(USB) method, I don't know if that rings any bells.

jimklimov, 2022-03-12:
That"MEGA(USB)" might imply Megatec (original Qx protocol author?
promoter?) and then trying with new nutdrv_qx or older blazer, riello and
several other drivers.

I think UPSMART was mentioned in earlier issues, can search for (and link
to) those discussions too...

lws-team, 2022-03-30:
I have a Powercool rack UPS that reports like this, I was able to get it mostly working by

a) build NUT from git, the distro versions seem too old
b) choose `hunnox` subdriver of nutdrv_qx

Whether you will have the same ride I don't know, because no two UPSes seem to ship with the same firmware (let alone USB UPS Device Class)

estatistics, 2024-02-04:
even older, upsmon is running successfully in linux in my distro only with sudo, without sudo, upsmon produces communication error

jimklimov, 2024-02-04:
Normally upsmon should be started as root, it forks to drop privileges for the most part but leaves a piece for `SHUTDOWNCMD` to fire. I suppose the latter could be a `sudo something` to avoid `root` for `upsmon` itself completely - it should be able to run without `root` too (pretty okay for monitoring-only of remote NUT servers). Permissions here may be an OS thing (firewalls based not only on IP/port but also app name, or e.g. SELinix profiles).