networkupstools/nut #1338: CPS wrong output voltage

pedro042, 2022-03-24:
I recently bought OR1500ERM1U and I"m very happy it's supported by Nut. Thanks a lot!

I see a minor issue in upsc:
_input.voltage: 232.0
input.voltage.nominal: 230
output.voltage: 253.0_

The value 253 in output.voltage is actually high transfer value. Probably easy to fix.

Version 2.7.4-5004-g1f143e5e (recemt/current git branch master)

jimklimov, 2022-03-24:
Nice to hear it (mostly) works for you :)

As for the issue at hand, can you please share a bit more about the configuration (which driver, some debug logs per troubleshooting hints in the documentation)?

So far I'd have two guesses: we are either mapping a wrong value indeed (whether our community's mistake, or firmware differences over time so "they" report a data point at a different node than the one which worked before), or in case of USB connections it may be or not be an issue with LogMin/LogMax processing per #1040 and related trail.

pedro042, 2022-03-24:
Of course, here is log, ups.conf and udev rules
[ups.zip](https://github.com/networkupstools/nut/files/8342710/ups.zip)
You're right, it's maybe firmware issue.

I'm not alone - for example #1191 - nominal voltage 120, output voltage 265 or #520 - 4 years old comment:

_input.transfer.high: 139
input.transfer.low: 88
input.voltage: 116.0
input.voltage.nominal: 120
**output.voltage: 140.0**_

But it differs - transfer high is 139 and output 140

1ockwood, 2022-03-31:
FWIW, I'm seeing this as well on a CP1500PFCLCD. The reported output voltage fluctuates around 136v. The display on the UPS itself, however, reports the expected output voltage ~120v. Let me know if there's any other info I can provide that may be helpful. 

```
device.mfr: CPS
device.model: CP1500PFCLCDa
device.type: ups
driver.name: usbhid-ups
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 120.0
input.voltage.nominal: 120
output.voltage: 136.0
ups.mfr: CPS
ups.model: CP1500PFCLCDa
ups.productid: 0601
ups.vendorid: 0764
``` 

[image]

pedro042, 2022-04-01:
Hi,

I' think your **reported** output voltage doesn't fluctuate. It's (reported) constantly 136V or, sometimes, zero. It's probably some issue in your monitoring software, IMO.

Your UPS display is right. Your input voltage is 120, your output voltage is 120V too. If this type of UPS ("line interactive") is not running on battery, the current simply pass thru; output is the same as input. The UPS start acting when input voltage is bellow low transfer setting or above high transfer setting or if there is an power outage (zero input power is, of course, bellow low transfer setting).
Output voltage can differ from input voltage only if line interactive UPS is running on battery.
Here is important part of my debug log:
```
0.009244     [D1] Path: UPS.Output.LowVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 0, Size: 16, Value: 207
0.009253     [D1] Path: UPS.Output.LowVoltageTransfer, Type: Input, ReportID: 0x10, Offset: 0, Size: 16, Value: 207
0.009258     [D1] Path: UPS.Output.HighVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 16, Size: 16, Value: 253
0.009261     [D1] Path: UPS.**Output.HighVoltageTransfer**, Type: Input, **ReportID: 0x10, Offset: 16**, Size: 16, Value: **253**
0.009452     [D1] Path: UPS.**Output.Voltage**, Type: Feature, **ReportID: 0x12, Offset: 0**, Size: 16, Value: **253**
```
On line 4, there is ReportID 0x10 offset 16 - HighVoltageTransfer 253
On line 5, ReportID 12, offset 0, Output Voltage 253
Based on the debug - I'm not sure what ReportID exactly is, does NUT really read diferrent data and UPS reports the same value 253 or NUT actually reads the same data? Or ReportID 0x10 offset 16 is the same part of data as ReportID 12 offset 0 ?

Someone tried PowerPanel? What reports PowerPanel?

jes1417, 2022-04-13:
I see this is as well and have seen it since the UPS was put in over a year ago.
[image]
and from the server
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 525
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.6
battery.voltage.nominal: 12
device.mfr: CPS
device.model: OR500LCDRM1Ua
device.serial: GA4JW2000855
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 118.0
input.voltage.nominal: 120
output.voltage: 135.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 63
ups.mfr: CPS
ups.model: OR500LCDRM1Ua
ups.productid: 0601
ups.realpower.nominal: 300
ups.serial: GA4JW2000855
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

jimklimov, 2022-04-14:
Just in case, would you have a chance to build a recent NUT source from
github and check in-situ if the current drivers also mis-report that? (and
what is the expected value range for you - circa 220-240V or 110-127V?)
Thanks.

pedro042, 2022-04-23:
Hi,
thank you, just rebuilt, nothing changed from my point of view. Output voltage still reported 253 V
_commit bce1d5201679e0dd64a839ceaaf47dc14c3c6a16
Merge: 573a03aba 22a32f1a3
Author: Jim Klimov <jimklimov@gmail.com>
Date:   Tue Apr 19 19:39:24 2022 +0200_

battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 9425
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.4
battery.voltage.nominal: 24
device.mfr: CPS
device.model: OR1500ERM1U
device.serial: GBULX2000140
device.type: ups
driver.name: usbhid-ups
driver.parameter.offdelay: 60
driver.parameter.ondelay: 300
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0-rc2-109-gbce1d5201
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.voltage: 230.0
input.voltage.nominal: 230
**output.voltage: 253.0**
ups.beeper.status: disabled
ups.delay.shutdown: 60
ups.delay.start: 300
ups.load: 8
ups.mfr: CPS
ups.model: OR1500ERM1U
ups.productid: 0601
ups.realpower.nominal: 900
ups.serial: GBULX2000140
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

45413, 2022-07-12:
As a fellow owner of a CP1500PFCLCDa (US Model - ProductID: 0601), I believe this issue is related to issue #439 and fix #1245.  This addresses an issue with CPS firmware returning and invalid LogMin/Max value but the fix is only applied for product ID 0x0501: https://github.com/networkupstools/nut/blob/587af14347cb1a85a093eca603c2c3c29cab9c89/drivers/cps-hid.c#L285.

@flashydave posted in #439 2 days ago saying it should be an easy fix and would raise a PR, my guess just adding in the additional ProductID `0x0601` in the if statement so it runs `cps_fix_report_desc` and doesn't just return.

flashydave, 2022-07-12:
Exacrly right!

I am away from my PC so if you want to do the change then im happy for you to
proceed.

Dave

closed 2022-07-22 by pedro042

icservis, 2024-01-21:
Issue is still there:

`root@pve:~# nut-scanner
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
Failed to create client: Daemon not running
[nutdev1]
	driver = "usbhid-ups"
	port = "auto"
	vendorid = "0764"
	productid = "0601"
	product = "OR600ERM1U"
	serial = "GA6NT2000017"
	vendor = "CPS"
	bus = "003"`

Values:
`root@pve:~# upsc CyberPower
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 3425
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: CPS
device.model: OR600ERM1U
device.serial: GA6NT2000017
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 253.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 21
ups.mfr: CPS
ups.model: OR600ERM1U
ups.productid: 0601
ups.realpower.nominal: 360
ups.serial: GA6NT2000017
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764`

jimklimov, 2024-01-21:
@icservis : Hello, can you please try an exploratory fix https://github.com/networkupstools/nut/issues/1338#issuecomment-1182304941 perhaps with the master-branch codebase (2.8.0 is almost 2 years ago now)?

Note that per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests you don't necessarily have to install a custom build over the system-provided package just to test if it helps; you can later install over that if it actually helps :)

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports mentioned in this issue (although the ones posted here seemed to be good already).

PR #1245 is presumed to have fixed the broken CPS transfer voltage reports discussed in this issue, at least the high one. Both values were reported reasonable in #1512 and #2718 testing, though possibly on models too different to this issue?...