networkupstools/nut #1343: Eaton Ellipse Eco 1200 VA [error: Driver not connected]

xaabi6, 2022-03-29:
Hello folks!

First thing first, thank you very much for this project, it's amazing 👍 

I'm opening this issue because I'm having trouble setting up a connection between my nut server which is an OpenWrt router (TP-Link Archer C7 v5) and the UPS stated on the title. I've set it up following some videos and taking a look at the issues I found here, but no matter what I do, I always get the same error. I also tried connecting a USB hub between the nut server and UPS (as mentioned [here](https://github.com/networkupstools/nut/issues/515#issuecomment-522319999)), explicitly running the service as root (mentioned [here](https://github.com/networkupstools/nut/issues/1322#issuecomment-1069678613)), rebooting the router or service multiple times and whatever I've found on related issues, but to no avail...

In addition, when I launch the `lsusb` command, I don't see any Eaton device. This also happened yesterday when I tried to configure it on an ODROID-C2 (also ARM architecture).

Also, if I run `upsdrvctl -D -u root start` command, this is the output:
> Network UPS Tools - UPS driver controller 2.7.4
   0.000000     Starting UPS: ellipse1200
   Network UPS Tools - Generic HID driver 0.41 (2.7.4)
   USB communication driver 0.33
   No matching HID UPS found
   0.016366     Driver failed to start (exit status=1)

And when I try to stop it using `upsdrvctl -D -u root stop`:
> Network UPS Tools - UPS driver controller 2.7.4
   0.000000     Stopping UPS: ellipse1200
   0.001003     Can't open /var/run/nut/usbhid-ups-auto.pid: No such file or directory

Finally, these're the packages I installed:
- nut ---> 2.7.4-22
- nut-common ---> 2.7.4-22
- nut-driver-usbhid-ups ---> 2.7.4-22
- nut-server ---> 2.7.4-22
- nut-web-cgi ---> 2.7.4-22
- luci-app-nut ---> git-20.108.38431-8f34e10

Regards

jimklimov, 2022-04-10:
Well, if something on the system precludes even the OS from seeing the USB device, I am not sure NUT can do much.

Can it be that the set of kernel built-in or modular drivers in your OpenWRT build is too constrained (embedded storage size and all) so it just does not include USB at all or much enough? Or that the USB port on the appliance is broken, powers down to save power, etc.? Are there other USB devices that would work (flash, printer, ...)?

Just in case, could you try this UPS and USB cable with a different computer, to rule out hardware issues with cable and port on the UPS side?

xaabi6, 2022-04-10:
Thank you very much for your tips Jim, I'll try to debug the situation again ASAP.

Actually, before connecting the UPS to my router, I did connect it to a PC using Eaton's software and worked fine for several weeks. I'll try installing NUT in my laptop and connecting it to the UPS, let's see what happens 👍

xaabi6, 2022-04-14:
Hi, I finally tried with the PC it used to work and amazingly, now it doesn't work, so, I'll buy a new cable to try it out and I hope it solves the problem.

xaabi6, 2022-04-30:
Fresh news, I purchased a new cable (it seems a good quality cable), but still doesn't work with any of my devices, so I think that the problem is in the UPS, I'll contact the commerce and see what they can do for me.

xaabi6, 2022-05-05:
Problem solved, I've got my UPS replaced with a new one and now is working flawlessly 🎉

closed 2022-05-05 by xaabi6

jimklimov, 2022-05-05:
For posterity: replaced by same model?

Any guesses - was the issue with hardware (some cable got loose or chip
shot), or is there also a different firmware version that could have fixed
it?

Thanks!

xaabi6, 2022-05-05:
Yes, it's been replaced by the same model.

The product revision stated in the package was the same in both cases (Rev. 01), the only difference was the manufacturation date, the faulty one was from 2021-05-27 and the new one from 2021-09-15.

I'm not sure about the problem the old one had, I guess it could be related to an internal loose cable, bad USB plug...