networkupstools/nut #1358: Bug in snmp-ups.c causes bogus “unhandled ASN0x80” messages to be logged

smarsching, 2022-04-08:
There is a bug in `snmp-ups.c` that causes the following message to be written to syslog repeatedly (when using the `ietf` SNMP driver):

```
unhandled ASN 0x80 received from .1.3.6.1.2.1.33.1.7.7.1`
```

The problem is in `nut_snmp_get_int`: When this function processes a response of type `ASN_OBJECT_ID`, it tries to resolve this object ID. In case of the `upsTestId` (a.k.a. 1.3.6.1.2.1.33.1.7.1.0 as defined in `ietf-mib.c`), this does not work because the returned OID should be interpreted as a constant instead of a pointer to another value.

From the IETF UPS MIB:

```upsTestId OBJECT-TYPE
    SYNTAX     OBJECT IDENTIFIER
    MAX-ACCESS read-write
    STATUS     current
    DESCRIPTION
            "The test is named by an OBJECT IDENTIFIER which
            allows a standard mechanism for the initiation of
            tests, including the well known tests identified in
            this document as well as those introduced by a
            particular implementation, i.e., as documented in the
            private enterprise MIB definition for the device.

            Setting this variable initiates the named test. Sets
            to this variable require the presence of
            upsTestSpinLock in the same SNMP message.

            The set request will be rejected with an appropriate
            error message if the requested test cannot be
            performed, including attempts to start a test when
            another test is already in progress.  The status of
            the current or last test is maintained in
            upsTestResultsSummary. Tests in progress may be
            aborted by setting the upsTestId variable to

            upsTestAbortTestInProgress.

            Read operations return the value of the name of the
            test in progress if a test is in progress or the name
            of the last test performed if no test is in progress,
            unless no test has been run, in which case the well
            known value upsTestNoTestsInitiated is returned."
    ::= { upsTest 1 }
```

and 

```
upsWellKnownTests     OBJECT IDENTIFIER ::= { upsTest 7 }

upsTestNoTestsInitiated OBJECT-IDENTITY
    STATUS     current
    DESCRIPTION
            "No tests have been initiated and no test is in
            progress."
    ::= { upsWellKnownTests  1 }
```

So, the OID `1.3.6.1.2.1.33.1.7.7.1` represents a constant, that should not really be resolved.

The code still works, because `nut_snmp_get_int` falls back to using the last part of the OID as the integer value, when the recursive call to `nut_snmp_get_int` fails, but it leads to the annoying

```
unhandled ASN 0x80 received from .1.3.6.1.2.1.33.1.7.7.1`
```

message being logged each time this function is called for the `upsTestId`.

I would be happy to submit a PR that fixes this issue, but I think it might make sense to first discuss in which way it should be fixed. I see three possible solutions:

1. Remove the call to `upslogx` in `nut_snmp_get_int` and replace it with a call to `upsdebugx`. This will have the effect that the message is not going to be logged when the driver is running normally (without debugging enabled). The downside is that the message will not be logged either when `nut_snmp_get_int` fails for a valid reason, so it isn’t my first choice.

2. Add a new parameter `supress_error_message` to `nut_snmp_get_int` and do not log the message when this parameter is set. We can then set this parameter when making the recursive call to `nut_snmp_get_int`, thus supressing the message when we don’t want it. This is my preferred solution.

3. Use `nut_snmp_get_oid` instead of `nut_snmp_get_int` when retrieving `upsTestId`. However, I don’t think that this could easily be done within the constraints of the current code architecture, so it probably is much more difficult to do it this way then using solution number two.

jimklimov, 2022-04-08:
Good research, thanks! For option 3 (to avoid hardcoding certain OIDs with option 2) we can look into bit-map flags in snmp-ups.h and call one of those methods based on a bit in the mapping table. Can't look into it now, but maybe there already are provisions for that?

CC: @aquette

autumnwalker, 2022-11-30:
Noticing the same issue here on nut-2.8.0-x86_64 (Slackware).

jimklimov, 2022-11-30:
Just in case, did recent #1710 or #1716 PRs happen to address this problem?

autumnwalker, 2022-11-30:
Unable to confirm - running as a plugin on Unraid which is based on Slackware 15. The most recent package available is 2.8.0.

jimklimov, 2023-01-06:
Hopefully when NUT v2.8.1 is released and plugin is updated, this issue can get checked again :)

jimklimov, 2025-03-19:
Tagging with DMF, because the likely solution would involve adding a flag - and that may conflict with FTY/DMF branch changes. Those would hopefully be upstreamed as part of NUT v2.8.4, so this issue can be addressed consistently along with that.

smarsching, 2026-03-02:
I just checked, and this issue is still present in NUT 2.8.1 (on Ubuntu 24.04). The message in syslog is just slightly different now:

```
Warning: type error exception (OID = .1.3.6.1.2.1.33.1.7.7.1
```