networkupstools/nut #1360: I am completely lost

aplayerv1, 2022-04-09:
So I cloned, I built it, I make install and i keep getting the same thing 

   _0.000000     debug level is '3'
   0.010833     Checking device (0001/0000) (002/002)
   0.025996     - VendorID: 0001
   0.026028     - ProductID: 0000
   0.026033     - Manufacturer: unknown
   0.026039     - Product: MEC0003
   0.026044     - Serial Number: unknown
   0.026049     - Bus: 002
   0.026054     - Device release number: 0100
   0.026059     Trying to match device
   0.026156     Device matches
   0.026217     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.026275     Trying megatec protocol...
   0.026288     send: Q1
   0.046054     received 47 (40)
   0.046083     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.046133     Status read in 1 tries
   0.046159     Supported UPS detected with megatec protocol
   0.046185     send: F
   0.058903     received 22 (35)
   0.058926     read: #000.0 0.0 00.00 00.0
   0.058943     Ratings read in 1 tries
   0.058951     send: I
   0.077892     received 39 (35)
   0.077921     read: #
   0.077940     Vendor information read in 1 tries
   0.077963     No values provided for battery high/low voltages in ups.conf

   0.077995     Using 'guestimation' (low: 0.000000, high: 0.000000)!
   0.078018     Battery runtime will not be calculated (runtimecal not set)
   0.078070     send: Q1
   0.099012     received 47 (40)
   0.099047     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.099170     dstate_init: sock /var/run/nut/blazer_usb-ups open on fd 9
   0.099214     send: Q1
   0.119923     received 47 (40)
   0.119953     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   2.101308     send: Q1
   2.126906     received 47 (40)
   2.126940     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   4.103334     send: Q1
   4.129025     received 47 (40)
   4.129172     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   6.103409     send: Q1
   6.123937     received 47 (40)
   6.123961     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   8.105487     send: Q1
   8.130973     received 47 (40)
   8.131003     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  10.107550     send: Q1
  10.133030     received 47 (40)
  10.133063     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  12.109616     send: Q1
  12.134972     received 47 (40)
  12.134999     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  14.111695     send: Q1
  14.137929     received 47 (40)
  14.137962     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  16.113747     send: Q1
  16.134868     received 47 (40)
  16.134891     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  18.114221     send: Q1
  18.138857     received 47 (40)
  18.138879     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000_

jimklimov, 2022-04-10:
Documentation includes chapters how to troubleshoot and report issues, boils down to telling which device, driver, config, versions etc. are used.

Here I guess use of a recent github build (which is good feature-wise - has more drivers and sub-drivers to try than 2.7.4 release did), and trying an USB UPS with possibly a variant of Megatec "Qx" protocol, with the older "blazer" driver. Although it does not pull much useful data from the device, so the variant tried probably does not match it.

"Which device" is important here, since "USB Vendor ID 0000 / Product ID 0001" is very generic and probably not an established vendor who registered and licensed a name from USBIF organization. Like with computer networking, "that machine uses MAC address range for a Realtek NIC" does not always map to a certain decision like "...so it is a Mac mini from 2000 with NFS v1" while a hypothetical "Mac Mini (R) 2000 NIC" might be a good guess for that.

But the "MEC0003" is something we see often as "Fry's Electronics" (not sure if the name comes from ID numbers or elsewhere). Still, by practical context this tells us about the USB chip used for electric interface and nothing about the protocol bytes it passes as implemented by the UPS vendor - and these were used with (variants of) standard USB HID protocol, variants of "Qx", and possibly some proprietary protocols. Please search the issue tracker with that keyword for more ideas which drivers (and their options) to try. In particular, for "Qx" protocol variants, see about the "nutdrv_qx" driver as it now aims to consolidate logic from many older drivers and has ways to select from a dozen subdrivers to try for protocol nuances.

aplayerv1, 2022-04-10:
so here was the problem, it does work with blazer and it does work with nutdrv_qx, after months and months of trying various ways, I decided to you know what lets try windows just for shits and giggles, I downloaded upsmart used megatec protocol, and boom it worked finally it worked, and so I now know its megatec, and i switched to linux and out of the blue it reads all data. like HUH what where when, was there an initialization that nut couldn't do?

aplayerv1, 2022-04-10:
battery.charge | Battery charge (percent of full) | 100
-- | -- | --
battery.voltage | Battery voltage (V) | 26.30
battery.voltage.high | Description unavailable | 26.00
battery.voltage.low | Description unavailable | 20.80
battery.voltage.nominal | Nominal battery voltage (V) | 24.0
device.mfr | Description unavailable |  
device.model | Description unavailable |  
device.type | Description unavailable | ups
driver.name | Driver name | blazer_usb
driver.parameter.pollinterval | Description unavailable | 2
driver.parameter.port | Description unavailable | auto
driver.parameter.synchronous | Description unavailable | no
driver.version | Driver version - NUT release | 2.7.4
driver.version.internal | Internal driver version | 0.12
input.current.nominal | Nominal input current (A) | 0.0
input.frequency | Input line frequency (Hz) | 50.2
input.frequency.nominal | Nominal input line frequency (Hz) | 0
input.voltage | Input voltage (V) | 224.0
input.voltage.fault | Description unavailable | 0.0
input.voltage.nominal | Nominal input voltage (V) | 0
output.voltage | Output voltage (V) | 224.0
ups.beeper.status | UPS beeper status | disabled
ups.delay.shutdown | Interval to wait after shutdown with delay command (seconds) | 30
ups.delay.start | Interval to wait before (re)starting the load (seconds) | 180
ups.firmware | UPS firmware | V3.8
ups.load | Load on UPS (percent of full) | 39
ups.mfr | UPS manufacturer |  
ups.model | UPS model |  
ups.productid | Product ID for USB devices | 0000
ups.status | UPS status | OL
ups.temperature | UPS temperature (degrees C) | 29.0
ups.type | UPS type | offline / line interactive
ups.vendorid | Vendor ID for USB devices | 0001

aplayerv1, 2022-04-10:
I hope this may help people in the future

jimklimov, 2022-04-10:
Thanks a lot for the report. It rings a bell a but, that some other ticket
mentioned running UPSmart before NUT as possibly fixing their issue, but it
could be any other amount of things they did as well.

Now wondering if it is possible to snoop what UPSmart does on the wire so
we can too. Maybe some "Hey device, you're licensed and may talk!"

In particular, wondering if your UPS would still be recognized if you
power-cycle it (maybe even taking battery out) or if such trick to beat
sense into it is a one-time event?

jimklimov, 2022-04-13:
Also, is there a "brand name" for the device so it might be mentioned in HCL as somehow working after all?

jimklimov, 2022-04-19:
Hi, while trawling the documentation before a release, I stumbled upon `usb_set_altinterface(=0)` optional setting for the device section in `ups.conf` - comments say some devices needed the call even if the interface remains at default zero; others can get stuck due to it - so it is not called by default. Can you check if it is the missing bit to initialize your UPS from NUT alone?

jimklimov, 2022-04-25:
Also, @aplayerv1 : is this discussion about same problem and device as in #1218 ?

AliMickey, 2022-04-26:
So I got nut to connect to the ups (Digitech) finally using the following config:
```
[ups]
        driver = blazer_usb
        port = auto
        vendorid = 0001
        productid = 0000
        pollinterval = 15
```

However, It is not reading any values.
```
Init SSL without certificate database

battery.voltage: 0.00
battery.voltage.high: 0.00
battery.voltage.low: 0.00
battery.voltage.nominal: 0.0
device.mfr:
device.model:
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: no
driver.parameter.vendorid: 0001
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 0.0
input.frequency: 0.0
input.frequency.nominal: 0
input.voltage: 0.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 0.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware:
ups.load: 0
ups.mfr:
ups.model:
ups.productid: 0000
ups.status: OL
ups.temperature: 0.0
ups.type: online
ups.vendorid: 0001
```

I tried the listed config on `https://networkupstools.org/stable-hcl.html` however it keeps saying `noscanlangid` is not a valid option for the driver. Removing it causes a `Device not supported` error.

😩

jimklimov, 2022-04-26:
I was just updating the HCL/DDL over past weekend, posted yesterday, with
reports collected over the years.

For Digitech, there were reports of success with `nutdrv_qx
protocol=hunnox` (more details in #674) which dictates a build of NUT from
master branch or one of recent 2.8.0-rc* snapshots. Packages built from
2.7.4 sources did not have these features yet.

jimklimov, 2022-04-26:
Also, please post separate reports for separate devices. Unfortunately, Vendors like 0000 and 0001 and ffff are radically not-unqiue and correspond to all sorts of real0life products and protocols.

jimklimov, 2022-04-26:
> @jimklimov <https://github.com/jimklimov> Thank you for the response. I
> have tried all the configs in the relevant issue threads. I suppose i was
> missing the specific build requirement.
>
> Do you know when 2.8.0 will be released as stable? If its only a few
> weeks, I will wait instead of reinstalling/making nut.

Well, I hope to do so soon(ish) now, having passed a few rehearsals and
bug/doc/api/... last-minute fixes. It then can take time to propagate
through distributions to become reasonable packages.

AliMickey, 2022-04-26:
Appreciate your hard work. I will try an RC release and see if it actually works or not.

Thanks!

closed 2023-01-06 by jimklimov