networkupstools/nut #1395: Improve nutdrv_qx with Q6 protocol details

jimklimov, 2022-04-25:
Follow research from https://github.com/networkupstools/nut/issues/1039 to add more data points (originally investigated for blazer_ser)

shangjiaxuan, 2024-12-23:
I recently acquired a piece of documentation from SANTAK reseller on their CASTLE series that have their `Q6` documentation in it.

I hope this helps to understand the fields in the returned string.

The document is in Chinese, but I think translation software are good enough these days.

[梦幻城堡及EDX 1~80KS通讯协议中文版2019.pdf](https://github.com/user-attachments/files/18226209/EDX.1.80KS.2019.pdf)

whc2001, 2025-02-06:
> I recently acquired a piece of documentation from SANTAK reseller on their CASTLE series that have their `Q6` documentation in it.
>
> I hope this helps to understand the fields in the returned [...]

Seems like we finally have a true breakthrough for the official documentation. Nice job, thanks!

jimklimov, 2026-02-12:
Some Q6 sopport was addressed in #2798 and #2737

closed 2026-02-12 by jimklimov