networkupstools/nut #1411: nJoy Keen 2000 support

perteraul, 2022-05-01:
Hi there, 

Would love to know if there's a way of using the nJoy Keen 2000 support — any of the in-list drivers don't work / are not compatible with this UPS. 
https://www.njoy.global/product/keen-2000?variants%5B53%5D=1&el=resources

nJoy provides the drivers, but I have no idea on how to use them without setting up a VM and install them there..
https://cdn.njoy.global/hive_files/wp/products/g50Brh.zip

Thanks a bunch!

jimklimov, 2022-05-01:
Looking at the names in that archive, that's Windows drivers.

Recently the NUT HCL and DDL were updated, with a couple of entries
introducing nJoy devices as something that exists :)

I have no idea if all their devices have the same protocol or not, and
can't say OTOH if those which worked were ok with plain 2.7.4 release (in
packages for past years) or needed subdrivers updated with 2.8.0 (released
last week).

Jim

jimklimov, 2022-05-10:
Note: there are reports of nJoy Keen 600 working : https://networkupstools.org/ddl/nJoy/Keen-600.html #867

And also an nJoy Aten line: #1281

closed 2022-05-11 by jimklimov

perteraul, 2022-05-13:
Hm... I don't see a way to get it working through Unraid, even after selecting any drivers mentioned to be working. Thanks!

jimklimov, 2023-06-27:
Hello, have you had any time to check if newer NUT (2.8.0 release or custom-build of yet newer master iterations) recognized the device?

I suppose per dating of the original post (and sudden revelation of unRaid being involved) you might have been on an older package...

AlbertBence, 2024-10-11:
I brought one for my TrueNAS Scale server.
After it didn't worked, this is what the manufacturer, nJoy, told me:

> Your Keen 2000 USB date of manufacture is October 2023. Only those manufactured starting with May 2024 have a communication interface compatible with popular NAS systems via HID USB for Windows.

I brought it today brand new.

jimklimov, 2024-10-12:
Did you try `nutdrv_qx` reported in the DDL link earlier?

AlbertBence, 2024-10-12:
Yes

jimklimov, 2024-10-12:
Which NUT version did you try? What configuration, commands, debug outputs were seen? What USB ID is in use?

Per #2650 at least some device samples did work with `usbhid-ups`, with CyberPower HID - so it is worth giving it a try anyway...

AlbertBence, 2024-10-13:
I use TrueNAS Scale's built in module (UI interface), which uses NUT server in the background, so I can't anwser most of those questions. The alert I got was that it can't connect to the UPS. I tried the following drivers:
-nJoy ups 2 Keen 600 USB (nutdrv_qx)
-nJoy ups 2 Keen 600 USB (blazer_ser)
-PowerWalker ups 2 PR1500LCDRT2U USB (usbhid-ups)

(But it's important to notice, that I wrote here to notify others of the problem, not to solve it myself, because it's just to much time for me for too little change)