networkupstools/nut #1412: Eaton Ellipse Eco 650USB - connection refused

binderth, 2022-05-01:
I'm using the Eaton Ellipse Eco 650USB, which is connected to a Raspbian Bullseye (Linux 11) - Raspberry Pi 3.

I just installed nut:
```
sudo apt-get install nut-client nut-server usbutils
```
which gave me `v2.7.4-13`

My `/etc/nut/ups.conf`:
```
maxretry = 3

[eaton]
        driver = usbhid-ups
        port = auto
        productid = ffff
        desc = "Eaton Ellipse Eco 650USB"

```

`sudo upsdrvctl start` gives me this:
```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: MGE HID 1.40
```

but: `upsc eaton` (or `sudo upsc eaton`) leads to:
```
Error: Connection failure: Connection refused
```

lsusb:
```
Bus 001 Device 004: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
`
What is causing this?

jimklimov, 2022-05-01:
I think it also needs vendorid (0x0463) to match.

Regarding rights, did you set up udev rules so the devfs node would be
handed by kernel to nut runtime user?

jimklimov, 2022-05-01:
Also you may need "usb quirks" set up in certain Linux kernel versions,
see
https://wiki.archlinux.org/title/Network_UPS_Tools#CyberPower_UPS_keeps_disconnecting/reconnecting

binderth, 2022-05-02:
I did screw up and edited the wrong config file for `upsd.users`, so of course, there wasn't any present for `upsmon.conf`. 
My second error was to think `upsc usbname@localhost` would use those files, but is independent! ;)

So, now it's running smoothly! ;)

> Regarding rights, did you set up udev rules so the devfs node would be
> handed by kernel to nut runtime user?

I did not. Apparently, this is optional for some LINUX flavours? Raspi OS 'Bullseye' handles the USB for a understandable way for nut, I guess?