networkupstools/nut #1413: getting "cached" values?

binderth, 2022-05-02:
Running nut v2.7.4 on a Raspberry Pi RaspiOS Bullseye connected to a Eaton Ellipse Eco 650USB.

I'm getting no errors whatsoever in the logs, but the values don't change. 
For example: I added another device to the UPS and the `ups.load` still shows "6". `battery.runtime` stays put to 2156, which are a bit more than 35mins - but I setup the device yesterday and even the nut-server runs more than 35mins.

I just ran a test and unplugged the UPS. Then `upsc` updated both `ups.load` to 10 and `battery.runtime` to 2070 (?) - but not they're stuck at those again? And also `battery.charge` went from 100 to 92 - and also stays there since.

Is there something I miss? My use case is to send current metrics of my UPS via MQTT, so I can use them on my home automation.

jimklimov, 2022-05-02:
Technically, NUT does cache last received readings, but they should refresh
often.

Some settings to look at are interrupt vs polling mode - some devices fare
better with one or the other, and pollfreq when applicable. Can't say about
defaults quickly, probably 5, 10 or 30 sec, but a raised driver debug level
would show entering the update loop in-vivo.

Otherwise it may be also the device's precision and update frequency for
exposed values.

binderth, 2022-05-03:
oh. sorry. It was a misunderstanding from my side. `battery.runtime` doesn't mean, how long the UPS runs already ("uptime") but how long it would last in the `ups.load`. So everything is fine! ;)

except, I didn't find the debug logs, but that's a different story! ;)

closed 2022-05-03 by binderth

jimklimov, 2022-05-04:
Debugs go to stderr, so check in journalctl of your service (assuming
systemd).

jimklimov, 2024-06-27:
FWIW, https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity was added to wiki later :)