networkupstools/nut #1415: nut 2.8.0 build failure

chenrui333, 2022-05-07:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>build error</summary>

```
dummy-ups.c:238:24: error: no member named 'st_mtim' in 'struct stat'
                                        if (datafile_stat.st_mtim.tv_sec != fs.st_mtim.tv_sec) {
                                            ~~~~~~~~~~~~~ ^
dummy-ups.c:238:45: error: no member named 'st_mtim' in 'struct stat'
                                        if (datafile_stat.st_mtim.tv_sec != fs.st_mtim.tv_sec) {
                                                                            ~~ ^
2 errors generated.
```

</details>

full build log, https://github.com/Homebrew/homebrew-core/actions/runs/2287073413
relates to Homebrew/homebrew-core#101035

jimklimov, 2022-05-07:
That would be MacOS? Which version? Hope I can find some clues in public
headers. Maintaining compat with obscure proprietary systems is hard
guesswork :\

Aldernately, can you locate and experiment how tge file modification time
field is called in `struct stat`, if present?

jimklimov, 2022-05-10:
Thanks for the nudge, I managed to get a MacOS build environment (on CircleCI) willing to build NUT, so iterated some more attempts to see and fix (or quickly hide) some more platform-specific issues which issued warnings and CI is set to treat those as fatal. Will post a PR soon-ish when that job finally passes :)

closed 2022-05-11 by jimklimov

chenrui333, 2022-07-03:
Thanks @jimklimov！

commit 4bec48f8 (2022-05-11): drivers/dummy-ups.c: use better portable st_mtime [fallout of #1383, fix #1415]
commit 9e6d3c79 (2022-05-11): Merge pull request #1420 from jimklimov/issue-1415

drivers/dummy-ups.c: use better portable st_mtime [fallout of #1383, fix #1415]
PR #1421 (2022-05-13): CircleCI project setup and some fixes for MacOS
https://app.circleci.com/pipelines/github/networkupstools/nut?filter=all

Closes: #1415 
Closes: #1419 

I'd say this is a "usable PoC quality", in that builds apparently pass, but implementation (detection instead of magic-passing of env vars for build settings) can be refined so it just works out of the box... later, and better done by someone with an actual Mac :)

CC @clepple : there were some fixes for macos driver, can you give those a second look please? and/or check on a live system, to make sure it still behaves well? :)
commit 3b56d81e (2022-05-13): drivers/dummy-ups.c: use better portable st_mtime [fallout of #1383, fix #1415]