networkupstools/nut #1416: Salicru SPS 850 ADV T supported by usbhid-ups

abelgomez, 2022-05-08:
I have a few [Salicru SPS 850 ADV T UPS](https://www.salicru.com/sps-850-adv-t.html), and this model seems to work with the new `salicru-hid` subdriver for `usbhid-ups`, but the product id (0302) is different from other supported Salicru USB HID UPSs (0201, 0202, 0203, 0300). I suppose all the others variants from [the same ADVANCE T family](https://www.salicru.com/sais/sps-advance-t.html) will work too since all of them seem to use the same PowerMaster software (but I cannot check if the productid is indeed different).

I'm new to nut, and I don't know what else I need to provide/test, but you can find below the output for the `upsc`, `upsrw` and `upscmd` commands (anyway, notice that instant commands are commented in the `salicru-hid` subdriver, and thus, no commands are available).

```
root@host:~# upsc salicru
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.runtime: 4500
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 26.00
battery.voltage.nominal: 24
device.mfr: 1
device.model:  850
device.serial: 000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0302
driver.parameter.synchronous: auto
driver.parameter.vendorid: 2e66
driver.version: 2.8.0-21-gf8462e88a
driver.version.data: Salicru HID 0.3
driver.version.internal: 0.47
driver.version.usb: libusb-0.1 (or compat)
input.frequency: 50.2
input.voltage: 228.0
input.voltage.nominal: 230
output.frequency: 50.2
output.voltage: 22.8
output.voltage.nominal: 24
ups.beeper.status: enabled
ups.load: 11
ups.mfr: 1
ups.model:  850
ups.productid: 0302
ups.realpower.nominal: 595
ups.serial: 000000000000
ups.status: OL
ups.vendorid: 2e66
root@host:~# upsrw salicru
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

root@host:~# upscmd -l salicru
Instant commands supported on UPS [salicru]:

root@host:~#
```

I tried to set the value of `battery.runtime.low` and `battery.charge.low` using `upsrw`, but although the command returns OK, no values seem to be modified.

I tested the driver on a Debian system, installing the stable Debian packages (nut 2.7.4), and overwriting the drivers with freshly built ones from the master branch (> 2.8.0) as explained here: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu (i.e., running `make install` from within the `drivers` and the `scripts/udev` directory). The mixture of versions does not seem to be an issue.

I didn't want to deviate from the standard packages of the distribution (as long as possible), since the UPSs are connected in a production environment (and I wanted to avoid installing complex stuff from sources). Now I'm on a test environment, and I can make software tests, but I will not have physical access to the UPS in the short term.

Just specifying both `vendorid` and `productid` in `/etc/nut/ups.conf` works with the code in the master branch:

```
[salicru]
    driver = usbhid-ups
    port = auto
    vendorid = 2e66
    productid = 0302
    desc = "Salicru SPS 850 ADV T"
```

In any case, I have explicitly added this model to the nut sources in a fork: https://github.com/networkupstools/nut/compare/master...abelgomez:salicru-sps-850-adv-t.

I can make a PR if you are interested.

jimklimov, 2022-05-08:
Looks great, and a PR from that branch looks useful too, thanks.

If you can also post a PR to nut-ddl repo with the upsc data dump and
comments above for other points, would be great also :)

abelgomez, 2022-05-08:
Sure!

Regarding the PR for nut-ddl, just a `Salicru/Salicru__SPS_850_ADV_T__usbhid_ups__2.8.0__01.dev` file with the above `upsc` dump is enough?

Which points do you want me to include in the comments? I said a lot of things that maybe are not interesting for the nut-ddl  repo 😄

jimklimov, 2022-05-10:
Github-eMail  intergration hicced up on me again :( Sorry about the delay.

Part of the idea with nut-ddl is to help people adding or troubleshooting device vs. driver support. so they would benefit at having as much context as possible. The files in that repository support explicit data usable by dummy-ups driver (effectively the data dumps from upsc), as well as structured comments that could be parsed by some tools, as well as unstructured comments for people to see.

I suppose the lack of value-setting and instant commands is something to at least comment there in layman fashion, not sure OTOH if there are structured-comment ways to specify lack of supported (working) items. There may be more details documented at https://networkupstools.org/ddl/#_file_naming_convention FWIW

abelgomez, 2022-10-11:
Hi @jimklimov ,

sorry for not advancing with this device, but I finally had physical access to the UPS, and I ended up by using the serial interface directly (rather than the USB HID) which indeed does work with the manufacturer's software (as opposed to the USB interface).

Thus, I'm no longer able to do further tests.

Nevertheless, since I already prepared a dev file in my clone of `nut-ddl` (which I never pushed), I can do it now, and I'll do a PR in a few minutes with the information I already gathered above (just in case it's useful).

Please, check the format since I created the file months ago, and let me know if further adjustments in the PR are needed.

abelgomez, 2022-10-11:
Regarding the `Sanity check list`, I tried to check all relevant points.

W.r.t. variables, as I said, it seems that writing values was not possible using `upsrw`, so I did not further inspect, and I didn't include comments for them.

hakonalstadheim, 2023-11-12:
I believe I have a device with the same firmware, https://www.salicru.com/sps-2000-adv-t.html. I'm not too familiar with git, but should I add a similar report to https://github.com/networkupstools/nut-ddl aswell? Managed to get git head compiled and running on a raspberry pi, driver.version: 2.8.1-94-gd59c05349.

jimklimov, 2024-10-07:
@hakonalstadheim : if the offer still stands - yes, thanks, that would be helpful :)