networkupstools/nut #1426: Support for MicroLink APC UPSs

EetuRasilainen, 2022-05-13:
Hi there,

as you know APC is still manufacturing MicroLink-only UPSs. I have purchased a APC SMC1000I and just found out that it could only be used as a generic USB-HID UPS with NUT right now. And unfortunately the SMC-series is not capable of ModBus so there are still APC devices out there which can only use MicroLink for a full operation.

So I think it would be great if MicroLink-UPSs could be supported directly by NUT at least through a serial connection. That's why I'd like to raise this _feature request_ here.

For testing purposes I have created a custom serial cable [with this pinout](https://pinoutguide.com/UPS/apc_0625_cable_pinout.shtml) and used the following Python tool for checking the MicroLink communication with my unit:

https://github.com/klaasdc/apcups-serial-test

I was able to get information about my unit and even set the battery installation date through MicroLink in a Python shell (with the help of [some sample code](https://github.com/klaasdc/apcups-serial-test/issues/1)). So it seems like the protocol description is there, it's just written in Python and needs to be ported to C. Unfortunately I am only familiar with Python and not with C :(

jimklimov, 2022-05-14:
Cross-posted to nut-upsdev mailing list, hoping someone would pick up the
challenge.

andy25225, 2025-05-17:
Hello,

I have an SMC100i-2uc here.
Modbus is available on it. I think you might just need to upgrade your firmware to at least 15.5, and then enable it.
Look for the "cP" (Communication Protocol) option on the device menu and ensure it is set to 1.

TrevorPH, 2026-09-22:
I have an SMC1500i from ~2018 (note, not an SMC1500i-2UC or SMC1500iC which are newer and different) from the original series. Only basic UPS info is available via usbhid-ups and most of the interesting stuff is missing. It has no menu option to enable/disable modbus even on firmware UPS 15.something that was the latest available in Aug 2026. I've also tried using microsol-apc from 2.8.5 and that also does not communicate with the UPS. I've tried using both a USB-A to USB-B cable and also an APC 940-1525A. My current recommendation for anyone looking at these is to avoid. Or I probably have one for sale very very cheap!

jimklimov, 2026-09-23:
@TrevorPH : Just in case, to clarify - did you also check if the `apcmicrolink` driver can manage this device? There were multiple changes related to APC support in general after the 2.8.5 release, but you might have to dive into https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to give them a spin (can run tests against your device from the build workspace without changing the OS installation until you like the results).

TrevorPH, 2026-09-23:
Thanks Jim, I will put the battery back in it and build from source and see what I get. I am using Fedora 43 & 44 and neither ship apcmicrolink as part of their distro package so BFS is the next step.

tmittelstaedt, 2026-09-26:
Excellent, I've been waiting for someone to test this.

TrevorPH, 2026-10-02:
OK, well that took a while. I started by trying to adapt the original Fedora 44 rpm to build from a -b master tarball but it just gave all sorts of weird errors and would not build. So I dragged an ancient sacrificial machine out and stuck F44 on that then installed all the dev tools and prereqs and built from a git checkout. After a bit of faffing about I got it built, installed, configured and running and the SMC1500I is working correctly.

[image]