networkupstools/nut #1427: 2.8.0 upsmon reports Cyberpower PR1500LCD/RMCARD205 as unavailable

awused, 2022-05-14:
* FreeBSD 13.0-release
* NUT 2.8.0 (installed via ports)
* UPS: Cyberpower PR1500LCD + RMCARD205
  * Previously used the same network card with another UPS without issue.
  * Was on firmware version 1.1.2, tried updating to 1.3.4. No apparent change. Not going to update to 1.3.5 as I still have a Linux client on 2.7.4.

With 2.7.4 I had no connectivity problems, but with 2.8.0 as soon as I start nut_upsmon it begins broadcasting `Communications with UPS cyberpower@localhost lost` and `UPS cyberpower@localhost is unavailable`. It never seems to establish a connection. All other tools seem to work just fine over snmp v1. My ups.conf is extremely basic, only specifying driver, community, and port. I did try specifying `synchronous = no` but it made no difference.

<details>
<summary>upsc output (2.8.0)</summary>

```
battery.charge: 100
battery.current: 3.20
battery.runtime: 3960
battery.runtime.elapsed: 0
battery.temperature: 34
battery.voltage: 27.60
battery.voltage.nominal: 24
device.contact: Administrator
device.description: UPS SNMP Card
device.location: Server Room
device.mfr: CYBERPOWER
device.model: PR1500LCD
device.serial: PYTLZ2000045
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.42.127
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: cyberpower MIB 0.51
driver.version.internal: 1.21
input.frequency: 60
input.voltage: 118.90
output.current: 3.20
output.frequency: 60
output.voltage: 118.90
ups.delay.reboot: 0
ups.delay.shutdown: 180
ups.delay.start: 0
ups.firmware: CR01203J9O1
ups.load: 26
ups.mfr: CYBERPOWER
ups.model: PR1500LCD
ups.serial: PYTLZ2000045
ups.status:
```
</details>

Notably ups.status is blank, and querying the IDs from https://github.com/networkupstools/nut/blob/master/drivers/cyberpower-mib.c for `ups.status` I get these values, which should show up as `OL` if I understand the code correctly. That is what I see on 2.7.4.
```
SNMPv2-SMI::enterprises.3808.1.1.1.4.1.1.0 = INTEGER: 2
SNMPv2-SMI::enterprises.3808.1.1.1.2.1.1.0 = INTEGER: 2
SNMPv2-SMI::enterprises.3808.1.1.1.7.2.7.0 = NULL
SNMPv2-SMI::enterprises.3808.1.1.1.2.2.5.0 = INTEGER: 1
```

I also have a Linux machine still on 2.7.4, it interprets less data but doesn't report it as unavailable.
<details>
<summary>upsc output (2.7.4)</summary>

```
battery.runtime: 255600.00
battery.runtime.elapsed: 0.00
device.mfr: CYBERPOWER
device.model: PR1500LCD
device.serial: PYTLZ2000045
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.42.127
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: cyberpower MIB 0.1
driver.version.internal: 0.97
ups.firmware: CR01203J9O1
ups.mfr: CYBERPOWER
ups.model: PR1500LCD
ups.serial: PYTLZ2000045
ups.status: OL
```
</details>

There are no errors from `snmp-ups -DDDDD` and nothing that looks out of the ordinary besides the empty ups.status. Running `upsmon -DDDDD` just gives me an uninformative error. The socket is properly created at `/var/db/nut/snmp-ups-cyberpower` and is user/group readable, opening up the permissions completely doesn't change anything.
```
Network UPS Tools upsmon 2.8.0
kill: No such process
   0.000133     UPS: cyberpower@localhost (secondary) (power value 1)
   0.000242     Using power down flag file /etc/killpower
   0.000411     [D1] debug level is '5'
   0.000416     [D1] debug level is '5'
   0.000856     [D1] Saving PID 13271 into /var/db/nut/upsmon.pid
   0.002830     [D1] Trying to connect to UPS [cyberpower@localhost]
   0.003025     UPS [cyberpower@localhost]: connect failed: Connection failure: Connection refused
   0.003031     [D2] do_notify: ntype 0x0005 (COMMBAD)
   0.003034     Communications with UPS cyberpower@localhost lost
   0.003208     [D3] Current power value: 1
   0.003213     [D3] Minimum power value: 1
```

jimklimov, 2022-05-14:
This does look odd, thanks for the report and cross-check vs. 2.7.4.

A couple of things that seem of interest:
* synchronous mode (no in 2.7.4, auto in 2.8.0) - if the newer driver sees more data to post to upsd, buffers can fill up and comms stop. With sync=auto, it should try to detect the situation and switch to the safer slower mode. Try to play with explicit setting there.

* "connection refused" in upsmon makes me think upsd data server on localhost is down (or stalled?)

* pollinterval=2 may be too intensive and overwhelms the UPS network card with work, especially if you have several clients polling it?

awused, 2022-05-14:
* Synchronous mode was one of the things I tried in the report. Switching it to an explicit "no" to match 2.7.4 did not change anything. Setting it to "yes" also doesn't change anything.
* The UPS data seems fresh when I run upsc. I see the battery runtime and ups load change over time.
* Changing pollinterval to 5 or 10 doesn't change anything either.

I did notice if I run the nut service normally instead of snmp-ups directly, I get a different error, so the connection refused may just have been a result of my debugging steps. Here are the results from upsmon with a normal `service nut onestart`.

```
   0.000000     UPS: cyberpower@localhost (secondary) (power value 1)
   0.000117     Using power down flag file /etc/killpower
   0.000298     [D1] debug level is '5'
   0.000304     [D1] debug level is '5'
   0.000721     [D1] Saving PID 81891 into /var/db/nut/upsmon.pid
   0.002733     [D1] Trying to connect to UPS [cyberpower@localhost]
   0.003035     Login on UPS [cyberpower@localhost] failed - got [ERR ACCESS-DENIED]
   0.003047     [D3] Current power value: 1
   0.003050     [D3] Minimum power value: 1
   5.033320     [D2] pollups: cyberpower@localhost
   5.033340     [D3] get_var: cyberpower@localhost / status
   5.033448     [D2] parse_status: []
   5.033453     [D2] do_notify: ntype 0x0005 (COMMBAD)
   5.033455     Communications with UPS cyberpower@localhost lost
   5.033659     [D3] Current power value: 1
   5.033667     [D3] Minimum power value: 1
```

jimklimov, 2022-05-15:
>    0.000000     UPS: ***@***.*** (secondary) (power value 1)
>    0.003035     Login on UPS ***@***.*** failed - got [ERR ACCESS-DENIED]
>    5.033340     [D3] get_var: ***@***.*** / status
>    5.033455     Communications with UPS ***@***.*** lost

This access denied looks like a mismatch of user/pass in upsmon.conf vs
upsd.users, or unsafe permissions on those?..

awused, 2022-05-15:
Those are not the issue. My username and password match and have remained unchanged for years with a previous UPS. They even worked just fine on 2.7.4 on this same machine and UPS combo right up until I switched to 2.8.0. The files are all owned by uucp:uucp and aren't world readable.

Reading the logs it does look like it queries ups.status, gets an empty string, and then treats it as if the UPS is unavailable, but as I pointed out in my initial report, all the values are available over snmpwalk and should result in a standard "OL" status.

awused, 2022-05-16:
Found the code responsible for the empty status being treated as unavailable: https://github.com/networkupstools/nut/blob/3e226165562a4fe289b1b48c4f9aec477ba2553f/clients/upsmon.c#L1667

Whatever broke construction of ups.status seems to be the root cause. The values that should be used for it show up properly with snmpwalk and are translated correctly with 2.7.4.

awused, 2022-05-23:
I did some digging around, and found the culprit. The `&& (strcmp(info_lkp->info_value, "NULL"))` clause should be removed from [here](https://github.com/networkupstools/nut/blob/86af0b98c1532b2a28a9c537a3c189e18aa72faa/drivers/snmp-ups.c#L2134) since the string "NULL" is a valid value. With ups.status in cyberpower's MIB, "NULL" is the first string encountered in [cyberpower_power_status](https://github.com/networkupstools/nut/blob/79fd9daa6cc53c4b4c3302e3335ab34cffd09541/drivers/cyberpower-mib.c#L36), so it never checks other values.

I can't tell if the "NULL" string should be changed to "" or if `su_find_infoval` should be updated to not stop iteration on "NULL" strings, so I've only patched it locally. Removing the check seems to work fine for me.

awused, 2022-05-23:
This means 2.8.0 is incompatible with Cyberpower UPSes, at least using snmp-ups. None of the other MIB files seem to have "NULL" strings except at the end of the lists, so it should only be Cyberpower that's broken by it. This also looks like it affects `su_find_valinfo`

jimklimov, 2022-05-23:
Thanks, great catch! CC @aquette just in case he remembers why that strcmp clause is there since at least 2005 ;)

As for Cyberpower, could be a fallout of my recent commit that sorted this table, sorry about that unexpected outcome. However, the `"NULL"` entry was not exactly last before it, either.

closed 2022-05-24 by jimklimov

awused, 2022-12-05:
I updated to Fedora 37 and was surprised to still be running into this bug nearly seven months later, are there any plans to cut a new release to fix this and the other bugs introduced with 2.8.0?

jimklimov, 2022-12-06:
Sorry about that - there were I think two more issues of this sort which we wanted to weed out first but reporters/wannabe-testers did not report in sync with developers trying to investigate/fix the issues.

Also note recent discussions here and in mailing list, that Fedora 37 packaging of NUT 2.8.0-x (up to -5 at least, reportedly) is broken. Nearly a one-line fix needed for PID paths, but does not work out of the box.

awused, 2022-12-06:
Honestly I didn't even run it, I saw the version number and stopped to confirm the source was still unpatched and uninstalled it before rebooting. I'm lucky I caught it before rebooting.

jimklimov, 2022-12-06:
Well, it is always an option to take the `configure` line from RPM SPEC file (take care about pid-related paths with Fedora 37 recipes though) and roll your own build from current master that had the NULL fixed. Or conversely, amend the spec to `rpmbuild` the current NUT as a package. Or save that PR's commit as a patch file and have `rpmbuild` (with amended spec) apply it over vanilla 2.8.0 sources.

PR #1432 (2022-05-24): Update cyberpower-mib.c C table ordering
Closes: #1427

Introduced by  #1381 for addressing #1377 (raised in #590 discussion).
commit 6a6888e9 (2022-05-24): Update cyberpower-mib.c

Closes: #1427