networkupstools/nut #1428: Total Energy Consumed is missing

Sesshoumaru-sama, 2022-05-15:
I am using NUT as add-on with home assistant to monitor my two APC UPS system - conneced via network using snmp.
There are a lot data provided about current, voltage and load, but I miss the information about energy, like total consumed energy (OID .1.3.6.1.4.1.318.1.1.1.4.3.6.0).

Downstream they say that the missing Energy parameters are due to NUT not providing it in first place: [DownstreamQuestion](https://github.com/hassio-addons/addon-nut/issues/227)

Is this a known limitation? Something to improve? Some way to verify?
Thanks in advance

jimklimov, 2022-05-17:
Not sure I've seen this complaint before (there were similar questions about accounting power use based on known V/A readings).

Generally this is fixable, along the lines of:
* check if docs/nut-names.txt has a standard name for the concept; if not - discuss proper generic neutral naming via nut-upsdev mailing list and raise a PR to "register" it
* use this name (or temporarily one from 'experimental.*' namespace) to publish the value accounted by the UPS as mapped from apc-mib.c in this case

Possibly MIB mappings for metered ePDUs (Eaton/MGE) can already serve similar values, so maybe the standard name exists. Not at PC to check quickly :)

Sesshoumaru-sama, 2022-05-17:
I checked the nut-names.txt for "energy" but there was nothing.
Its usually filed under the "Output" header I think as somehting like "TotalEnergyUsed".

So my take on it would be 
| output.energy.totalused    | Nominal value of totally used power (kwh [times 100])     | 49000 {=490kwh)
Note: The APC units I have report kwh * 100 as value for OID .1.3.6.1.4.1.318.1.1.1.4.3.6.0)

The proper proceeding now would be to write to some mailing list?

jimklimov, 2022-05-17:
I think there should be standard units like kWh (maybe not integer) rather than vendor multipliers that are not a power of 1000 :)

NUT mappings can multiply and otherwise convert readings on the fly.

Can you please subscribe to https://alioth-lists.debian.net/cgi-bin/mailman/listinfo/nut-upsdev to propose this to community?

Sesshoumaru-sama, 2022-05-17:
Subscribed.
Will post request later the day.

Sesshoumaru-sama, 2022-05-19:
Did send it. Hopefully somebody can relate.