networkupstools/nut #1433: Debian Package still 2.7.4-13

jackc94, 2022-05-26:
Desperate to make use of the nut driver enumerator but version 2.8.0 hasn’t made it to the Debian Packages yet… are you able to assist?

jimklimov, 2022-05-26:
Well, at a minimum you can run it as a script once, so it parses your ups.conf and defines the nut-driver@something instances. This would probably rely on having the generic units installed (from NUT sources) instead of ones packaged earlier.

Caveat possible: current units would start NUT daemons in foreground mode (type simple, not forking); 2.7.4 lacked that option, but for a simple fix-back you can set a `-D` option for equivalent with a bit of debug to journal.

I suppose it is not too often a "simple end user" would need the NDE service or timer to re-parse the config regularly (we needed that on a large monitoring system).

jackc94, 2022-05-26:
@jimklimov Thanks so much for such a speedy response. I’m not going to lie… I didn’t really understand much of that…!?

biergaizi, 2022-07-03:
I filed a bug report at Debian to raise attention to the issue.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1014287

NilByte, 2022-07-04:
I have been watching this for progress.

https://tracker.debian.org/pkg/nut

biergaizi, 2022-07-07:
I just uploaded my unofficial Debian and Ubuntu packages to an unofficial repo, using openSUSE's Open Build System, based on my Debian non-maintainer upload patch. 

To install the package on Debian 11:

    # Trust signing key
    wget 'https://download.opensuse.org/repositories/home:/niconiconi/Debian_11/Release.key' -O /etc/apt/trusted.gpg.d/nut-unofficial.asc

    # Add Repo
    echo 'https://download.opensuse.org/repositories/home:/niconiconi/Debian_11/ ./' >> /etc/apt/sources.list

To install the package on Ubuntu 22.04

    # Trust signing key
    wget 'https://download.opensuse.org/repositories/home:/niconiconi/xUbuntu_22.04/Release.key' -O /etc/apt/trusted.gpg.d/nut-unofficial.asc

    # Add Repo
    echo 'https://download.opensuse.org/repositories/home:/niconiconi/xUbuntu_22.04/ ./' >> /etc/apt/sources.list

Note that these packages are 100% untested and may not function correctly. Don't forget to report back.

jimklimov, 2022-07-07:
(got stuck in Drafts a few months ago) NDE relies on having the systemd
unit definitions updated with NUT 2.8.0, in particular to run each driver
as a separate service instance; later there were tweaks in both units and
daemons for "foreground mode" as a more efficient way to daemonize in
systemd.

This can be ported back to a 2.7.4 setup probably (at least the unit
layout, before `-F` mode introduction - after all that's the codebase where
NDE effort started), but if you "didn’t really understand much of that"
then it seems building 2.8.x locally or waiting for a package would be more
efficient and feasible.

biergaizi, 2022-07-11:
Good news, everyone!

> The maintainer have merged your work:
> https://salsa.debian.org/debian/nut/-/commit/89ba505c47313daae843012a9dcec883fe11b390
>
> He did other improvements and new version was uploaded to experimental and now is waiting the ftp master review (may take days or weeks) for new binaries:
>
> https://ftp-master.debian.org/new/nut_2.8.0-1~exp1.html

closed 2022-07-17 by jimklimov

biergaizi, 2022-10-22:
> Subject: nut 2.8.0-4 MIGRATED to testing
>
> FYI: The status of the nut source package
> in Debian's testing distribution has changed.
>
>   Previous version: 2.7.4-14
>   Current version:  2.8.0-4

https://tracker.debian.org/news/1376801/nut-280-4-migrated-to-testing/

jimklimov, 2022-10-22:
Great news, thanks for the heads-up! Just like a birthday present, just half a year in the making :)