networkupstools/nut #1437: Salicru SLC 3000 Twin RT2 not supported by usbhid-ups

fsckawk, 2022-06-01:
### My setup:

- OS: Linux 4.18.0-305.3.1.el8 x86_64 GNU/Linux CentOS Linux release 8.4.2105
- UPS: Salicru SLC 3000 Twin RT2 (2e66:0102)
- NUT: Just downloaded from github + libusbx-1.0.23-4
Network UPS Tools - UPS driver controller 2.8.0-98-g3744cb047
Network UPS Tools - Generic HID driver 0.48 (2.8.0-98-g3744cb047)
USB communication driver (libusb 1.0) 0.43
Using subdriver: Salicru HID 0.3
Note: I've added this to drivers/salicru-hid.c, but it made no difference:
`{ USB_DEVICE(SALICRU_VENDORID, 0x0102), NULL },`
- Relevant info from "udevadm info --attribute-walk --name=/dev/usb/hiddev1":
```
    ATTRS{serial}=="0000000000000000"
    ATTRS{manufacturer}=="Salicru"
    ATTRS{product}=="TWIN3000-R"
    ATTRS{serial}=="0000:00:14.0" 
    ATTRS{manufacturer}=="Linux 3.10.0-1062.1.1.el7.x86_64 xhci-hcd"
    ATTRS{product}=="xHCI Host Controller"
```
- ups.conf
```
[upsslc]
        driver=usbhid-ups
        port=auto
        vendorid=2e66
        productid=0102
        desc = "Salicru SLC 3000 Twin RT2"
```

### Output from "upsc":
```
device.mfr: Salicru
device.model: TWIN3000-R
device.serial: 0000000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0102
driver.parameter.synchronous: auto
driver.parameter.vendorid: 2e66
driver.version: 2.8.0-98-g3744cb047
driver.version.data: Salicru HID 0.3
driver.version.internal: 0.48
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
ups.mfr: Salicru
ups.model: TWIN3000-R
ups.productid: 0102
ups.serial: 0000000000000000
ups.status: OB
ups.vendorid: 2e66
```

#### Output from "/usr/local/ups/bin/usbhid-ups -D -u root -a upsslc":

```
Network UPS Tools - Generic HID driver 0.48 (2.8.0-98-g3744cb047)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000086	[D1] debug level is '1'
   0.001301	[D1] upsdrv_initups (non-SHUT)...
   0.006268	Using subdriver: Salicru HID 0.3
   0.006292	[D1] 2 HID objects found
   0.006487	nut_libusb_get_report: Success
   0.006509	[D1] Can't retrieve Report 00: Resource temporarily unavailable
   0.006534	[D1] Path: ff000001.ff000002, Type: Input, ReportID: 0x00, Offset: 0, Size: 8
   0.006739	nut_libusb_get_report: Success
   0.006744	[D1] Can't retrieve Report 00: Resource temporarily unavailable
   0.006750	[D1] Path: ff000001.ff000002, Type: Output, ReportID: 0x00, Offset: 0, Size: 8
   0.006764	[D1] Detected a UPS: Salicru/TWIN3000-R
   0.006850	[D1] upsdrv_initinfo...
   0.006857	[D1] upsdrv_updateinfo...
   0.757015	[D1] Got 0 HID objects...
   0.757048	[D1] Quick update...
   0.757231	[D1] Group and/or user account for this driver was customized ('root:nut') compared to built-in defaults. Fixing socket '/var/state/ups/usbhid-ups-upsslc' ownership/access.
   0.757248	[D1] Group access for this driver successfully fixed
   0.757262	[D1] upsdrv_updateinfo...
   1.507501	[D1] Got 0 HID objects...
   1.507532	[D1] Quick update...
   2.758609	[D1] upsdrv_updateinfo...
   3.508807	[D1] Got 0 HID objects...
   3.508837	[D1] Quick update...
```
As you can see, it stucks in a loop, getting 0 HID objects. If I increase the debug level...

```
   0.757823	[D1] upsdrv_updateinfo...
   1.508007	[D2] nut_libusb_get_interrupt: Connection timed out
   1.508018	[D1] Got 0 HID objects...
   1.508024	[D1] Quick update...
   2.759135	[D1] upsdrv_updateinfo...
   3.509375	[D2] nut_libusb_get_interrupt: Connection timed out
   3.509392	[D1] Got 0 HID objects...
   3.509396	[D1] Quick update...

```
upsrw and upscmd -l both return nothing...

jimklimov, 2022-06-02:
Interesting... seems like some usb layer issue; try the option for "pollonly" (iirc the name) to avoid interrupts. Some systems seem to only work with one or the other method.

jimklimov, 2022-07-13:
Hm, one more idea popped up: since you've had to add a `productid` to the sources, chances are that the value is not listed in your `udev.rules` (or newer `hwdb`) and so the system default handler grabs the device and keeps the NUT driver out.

inode64, 2023-03-28:
I have managed to make the Salicru ADV3000-RT UPS work, following the ideas of issue #1856, the only thing that does not appear is "Not supported", and neither is the battery autonomy, for the rest, everything seems correct.

```
Bus 001 Device 002: ID 2e66:0101 Salicru ADV3000-RT
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x2e66 
  idProduct          0x0101 
  bcdDevice            0.02
  iManufacturer           3 Salicru
  iProduct                1 ADV3000-RT
  iSerial                 2 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           25 Spanish
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 583 lines cut ...]
  19.527784	[D5] update_status: !LB
  19.527788	[D5] update_status: !CAL
  19.527792	[D5] update_status: !FSD
  19.527801	[D5] send_to_all: SETINFO driver.state "quiet"
  21.326211	[D5] send_to_all: SETINFO driver.state "updateinfo"
  21.326234	[D1] upsdrv_updateinfo...
  21.326240	[D1] Quick update...
  21.326498	[D3] send: QGS
  21.391790	[D5] read [  0]: (8 bytes) => 28 32 32 39 2e 31 20 35
  21.399764	[D5] read [  8]: (8 bytes) => 30 2e 30 20 32 32 39 2e
  21.423778	[D5] read [ 16]: (8 bytes) => 31 20 35 30 2e 30 20 30
  21.431767	[D5] read [ 24]: (8 bytes) => 30 31 2e 38 20 30 31 34
  21.447770	[D5] read [ 32]: (8 bytes) => 20 33 36 35 2e 35 20 2d
  21.463770	[D5] read [ 40]: (8 bytes) => 2d 2d 2e 2d 20 30 38 32
  21.471781	[D5] read [ 48]: (8 bytes) => 2e 30 20 2d 2d 2d 2e 2d
  21.487768	[D5] read [ 56]: (8 bytes) => 20 30 32 30 2e 35 20 30
  21.495767	[D5] read [ 64]: (8 bytes) => 31 30 30 30 30 30 30 30
  21.503764	[D5] read [ 72]: (8 bytes) => 30 30 31 0d 00 00 00 00
  21.503771	[D3] read: (229.1 50.0 229.1 50.0 001.8 014 365.5 ---.- 082.0 ---.- 020.5 010000000001
  21.503779	[D5] update_status: OL
  21.503782	[D5] update_status: !LB
  21.503786	[D5] update_status: !CAL
  21.503790	[D5] update_status: !FSD
  21.503799	[D5] send_to_all: SETINFO driver.state "quiet"
```

jimklimov, 2026-02-12:
TODO: port SETINFO clues to NUT DDL dump