networkupstools/nut #1440: Ippon Model not support

Deadfuntik, 2022-06-04:
[image]
[image]

how to write the name of the model? not automatically detected

jimklimov, 2022-06-04:
Maybe the device does not tell... do tools like `lsusb` (or plug-event logs in dmesg, or debug in nut drivers startup) say any string about the device?

Deadfuntik, 2022-06-05:
is it possible to write it manually? I understand it is not critical, but it hurts the eye)
[image]

jimklimov, 2022-06-09:
In ups.conf, try setting `override.device.model` and/or `override.ups.model`