networkupstools/nut #1442: [HCL] Cyber Power Systems UT1500EI supported by usbhid-ups

AnrDaemon, 2022-06-14:
Datasheet: [https://www.cyberpower.com/my/en/File/GetFileSampleByType?fileId=DS-18100009-01&fileType=Download%20Center&fileSubType=FileOriginal](https://www.cyberpower.com/my/en/File/GetFileSampleByType?fileId=DS-18100009-01&fileType=Download%20Center&fileSubType=FileOriginal)
```
# upsc UT1500EI
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 3600
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 18.1
battery.voltage.nominal: 24
device.mfr: CPS
device.model: UT1500E
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.vendorid: 0764
driver.version: 2.7.2
driver.version.data: CyberPower HID 0.3
driver.version.internal: 0.38
input.transfer.high: 0
input.transfer.low: 0
input.voltage: 221.0
input.voltage.nominal: 230
output.voltage: 223.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: UT1500E
ups.productid: 0501
ups.realpower.nominal: 900
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

```
# upsrw UT1500EI
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 0

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 0

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```

```
# upscmd -l UT1500EI
Instant commands supported on UPS [UT1500EI]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

Quick functional tests have passed.

AnrDaemon, 2022-06-14:
`ups.conf`:
```
[UT1500EI]
	driver = usbhid-ups
	port = auto
	vendorid = 0764
```

jimklimov, 2023-03-31:
Not even the `productid = 0501`? I'd expect both values to be required if present...

AnrDaemon, 2023-03-31:
I suppose that's only required if you want to control multiple connected UPS units.