networkupstools/nut #1449: Question: USBHID-UPS with APC SMT3000RM2UC

rcdailey, 2022-06-18:
I'm not sure if this is the right place to ask this question, since I don't deal with NUT directly. I use it through Home Assistant's NUT server.

I'm not able to see Load Percentage using the usbhid-ups driver. Is this a limitation of my specific APC model? I have connected the USB port directly to the machine running Home Assistant.

Below is a screenshot of the current entities / data set I get from NUT. I am expecting a percentage of load somewhere, but not seeing it. This would be the same percentage load I see on the UPS small LED screen.

[image]

jimklimov, 2022-06-21:
Can't quickly say. It could well be a limitation of your NUT driver version, especially if HA bundle still contains 2.7.4 release and not 2.8.0 or newer (not sure OTOH if any mappings changed for this, not too online now to compare sources), or if the device does not serve the data on mapping entry we check for this info (if we do), or HA does not map all of NUT info...

rcdailey, 2022-06-22:
I went out on a limb and ran my own docker container for NUT server. I'm  using this one: https://hub.docker.com/r/instantlinux/nut-upsd

This has v2.8.0. Unfortunately, I'm still only seeing the same entities. No load percentage. Can you help me determine why I can't view this for my particular APC model?

Thanks for your patience and help thus far!

rcdailey, 2022-08-01:
@jimklimov Any chance to review my last few replies to see if you can help me understand why NUT isn't showing a load percentage for my UPS?

jimklimov, 2022-08-02:
Not at a PC now to check the command lines, so you might have to trawl other recent issue discussions - but generally a USB driver start with higherr debug levels, and/or using a gen-usb-subdriver script from sources, should walk the device and report data fields seen from it. Then you'd have to compare sources to see if your UPS has a data point the driver does not handle, or just does not report the value you're after.

rcdailey, 2022-08-02:
I'm not sure how to do that. Is the higher debug levels an option when starting NUT? I'll assume you mentioned checking recent issue discussions because you put instructions in there somewhere. Thanks for taking the time to help me out.

jimklimov, 2022-08-07:
Depending on NUT version, the 2.8.0 release and newer (git master so far) support a `debug_min` option in config files for `upsd` and driver daemons. For older releases you have to get into init scripts or systems daemons involved to bump the debug by adding half a dozen `-D` options. Better hush it back down after troubleshooting.

EetuRasilainen, 2022-12-22:
Unfortunately a newer NUT version is not going to help you here at the moment. On newer APC devices advanced information such as the current load and power consumption are exposed through both Modbus and MicroLink and not through USB-HID. 

MicroLink is not supported by NUT at all (see #1426) and Modbus support is not in a good shape right now. If you really need the UPS load/power consumption and some other advanced features  your best option is using apcupsd right now as they have proper Modbus support for newer APC SmartUPS devices such as yours.

rcdailey, 2022-12-22:
I recall trying to use apcupsd driver in NUT and had some issues with it. IIRC it didn't work over USB or something. Been a hot minute.

klada, 2022-12-22:
What I was saying is: don't use NUT, use apcupsd instead (unfortunately). APC UPS support is not in a good shape in NUT right now. For all "newer" APC UPS models (a.k.a. models introduced in 2009 or later [SMT, SMC, SMX, SRT lines]) only basic reporting is available in NUT when falling back to USBHID.

APC/Schneider have published their Modbus specification online, but of course we still need a volunteer to implement this in NUT. Right now I think APC is not putting any development resources into the NUT project so I doubt that APC UPS support within NUT is going to improve anytime soon.

jimklimov, 2022-12-22:
Actually it is not as gloomy today as it was a year or two ago - several
modbus drivers were contributed, just not an APC among them yet. So there
are precedents to learn from :)

rcdailey, 2022-12-22:
I love NUT because it integrates so nicely with Home Assistant. I have no experience with apcupsd, but I'll check it out. Hopefully it integrates nicely and is easy to set up.

jimklimov, 2022-12-23:
There is a NUT apcupsd-ups driver, to retranslate... better than nothing,
to integrate with NUT ecosystem.

In this year's discussions however somebody complained... I think debian(?)
packaging botched it that either one or another package may be installed.
But that was a distro issue, programs worked together.

Jim