networkupstools/nut #1453: NUT 2.7.4 APC Smart 750

OrgoneBlaster, 2022-06-21:
Hello,

There are no input/output and load data displayed when I run the upsc command.

jimklimov, 2022-06-21:
Can you please check if there are mappings for the entries you expected at https://github.com/networkupstools/nut/blob/3744cb0476820d3bbb6dc3f4ba62feb94673d9b4/drivers/apc-hid.c#L322 (assuming you use usbhid-ups driver)?

If not, it can help if you "explore" your device with subdriver generation script (see docs in source), merge its output with the apc-hid.c to add missing lines in correct places, test it and post a PR.

If the entries are missing in the old release (set in stone) and appear(ed) in newer codebase, e.g. by your PR or if someone addressed it earlier, an upgrade is due :) Note that 2.7.4 was released about 6 years ago.

jimklimov, 2022-09-12:
Also, FWIW - which driver (config), media and protocol do you use? Various APCs can talk Serial, USB, SNMP... all differently.