networkupstools/nut #1455: upscode2 fails with Compaq T2400h, works at high debug level

marekm72, 2022-06-25:
The driver needs a minimum of three -D options to recognize responses from the UPS correctly.
At lower debug levels, just says "Missing UPCL after UPCL" - actually receives "UPCL\n" which fails to compare with "UPCL".
UPS responses on the serial port are the same, it's just that debug option changes something in the way it is received (with newline as part of the command).  After each "Supports command: UPxx" debug message there is an empty line.

nut-2.8.0-r0 package from Alpine Linux
Network UPS Tools - UPScode II UPS driver 0.90 (20220328-3481-gccbf3eb6c6)
Warning: This is an experimental driver.
Some features may not function correctly.

marekm72, 2022-06-26:
Looking at the code, I see the bug in upscode2.c line 1053:

		upsdebugx(3, "upscrecv: %zd bytes:\t'%s'", res-1, str_rtrim(buf, ENDCHAR));

which should be replaced with:

		str_rtrim(buf, ENDCHAR);
		upsdebugx(3, "upscrecv: %zd bytes:\t'%s'", res-1, buf);		

upsdebugx is a macro, its other args are only evaluated if debug level >= first arg.
str_rtrim has side effects (modifies the string in place and returns pointer to it).
The trailing newline should be removed regardless of debug level, of course.
It may be good to review the code (other drivers too) for other similar bugs.

jimklimov, 2022-06-26:
Nice catch! FWIW, it was changed from function to macro for a bit of efficiency in the default case (no debug - no overhead to call methods).

jimklimov, 2022-07-14:
Thanks, preparing a PR. FWIW, I gave up a bit on using a source formatter to simplify the searches, and went for this one-liner to trawl the code with my bare eyes (and some `less` search):

````
git grep -A10 -E 'upsdebug(x|_hex|_ascii|_with_errno)' | (H=false; while IFS= read LINE ; do case "$LINE" in *upsdebug*) H=true ;; esac; $H && echo "$LINE" ; case "$LINE" in *");") H=false; echo "---" ;; esac ; done )| uniq | less
````

Just in case similar needs to be done again...

closed 2022-07-17 by jimklimov