networkupstools/nut #1456: Can't claim USB device

nseries100, 2022-06-27:
When I try start NUT
`sudo upsdrvctl start
`

I get this error
_Can't claim USB device [1cb0:0038]@0/0: could not detach kernel driver from interface 0: Operation not permitted
Driver failed to start (exit status=1)_

When I manually change permission and then try to start again, then its ok - but its not persistent.
`sudo chmod 0666 /dev/bus/usb/001/003`

Any ideas? :)

Setup:
`uname -a`
Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

`dpkg -s nut`
Package: nut
Status: install ok installed
Priority: optional
Section: metapackages
Installed-Size: 264
Maintainer: Arnaud Quette <aquette@debian.org>
Architecture: all
Version: 2.7.4-8
Depends: nut-client, nut-server
Description: network UPS tools

etc
drwxr-xr-x  2 root nut     4096 Jun 22 10:28 nut

etc/nut
-rw-r--r-- 1 root root  1111 Jun 22 10:26 hosts.conf
-rw-r----- 1 root nut   1543 Jun 22 09:23 nut.conf
-rw-r----- 1 root nut   5700 Jun 22 09:03 ups.conf
-rw-r----- 1 root nut   4599 Jul 22  2021 upsd.conf
-rw-r----- 1 root nut   2225 Jun 22 10:21 upsd.users
-rw-r----- 1 root nut  15424 Jun 22 10:28 upsmon.conf
-rw-r----- 1 root nut   3887 Jun  1  2018 upssched.conf
-rw-r--r-- 1 root root  1452 Jun 22 09:46 upsset.conf
-rw-r--r-- 1 root root  3603 Jun  1  2018 upsstats.html
-rw-r--r-- 1 root root  6446 Jun  1  2018 upsstats-single.html

jimklimov, 2022-06-28:
Normally udev rules should get installed, to dynamically assign permissions to NUT-supported vendor/product IDs, so then the drivers can access them. Relies on consistent packaging to install the user accounts etc.

dehnert, 2023-09-12:
FWIW, I had a similar error, and in my case the relevant UPS just seemed to be missing from `/usr/lib/udev/rules.d/62-nut-usbups.rules`.

Adding the missing lines to a file in `/etc/` (and rebooting) seemed to work:
```
# cat /etc/udev/rules.d/63-nut-usbups.rules
# For some reason, missing from 
# /usr/lib/udev/rules.d/62-nut-usbups.rules

# Add my UPS:
# Bus 001 Device 004: ID 09ae:3024 Tripp Lite AVR750U         
ATTR{idVendor}=="09ae", ATTR{idProduct}=="3024", MODE="664", GROUP="nut"
```
(Not sure if there's a subtle issue there.)

One imagines either the applicable line is missing for you (I don't see any `1cb0` in my copy) or `/lib/` is the wrong path on your system, and either way probably adding something to `/etc/` can work. (There might be a better solution of course.)