networkupstools/nut #1460: Error installing nut -- Failed with result 'exit-code'

mabesignal, 2022-07-03:
Hi,

I'm unable to install nut with "sudo apt install nut" and fails with the errors below. After install "nut-scanner" is missing and appears I can't proceed and setup nut. Any advice or guidance is appreciated, thanks!

Selecting previously unselected package nut-client.
Preparing to unpack .../nut-client_2.7.4-5.1ubuntu2_arm64.deb ...
Unpacking nut-client (2.7.4-5.1ubuntu2) ...
Selecting previously unselected package nut-server.
Preparing to unpack .../nut-server_2.7.4-5.1ubuntu2_arm64.deb ...
Unpacking nut-server (2.7.4-5.1ubuntu2) ...
Selecting previously unselected package nut.
Preparing to unpack .../nut_2.7.4-5.1ubuntu2_all.deb ...
Unpacking nut (2.7.4-5.1ubuntu2) ...
Setting up libupsclient4:arm64 (2.7.4-5.1ubuntu2) ...
Setting up nut-client (2.7.4-5.1ubuntu2) ...
Job for nut-monitor.service failed because the control process exited with error code.
See "systemctl status nut-monitor.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nut-client, action "restart" failed.
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
   Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-07-03 15:39:42 EDT; 27ms ago
  **Process: 27100 ExecStart=/sbin/upsmon (code=exited, status=1/FAILURE)**

Jul 03 15:39:42 rock64 systemd[1]: Starting Network UPS Tools - power device monitor and shutdown controller...
Jul 03 15:39:42 rock64 systemd[1]: nut-monitor.service: Control process exited, code=exited status=1
Jul 03 15:39:42 rock64 systemd[1]: nut-monitor.service: Failed with result 'exit-code'.
Jul 03 15:39:42 rock64 systemd[1]: **Failed to start Network UPS Tools - power device monitor and shutdown controller.**
Setting up libusb-0.1-4:arm64 (2:0.1.12-31) ...
Setting up nut-server (2.7.4-5.1ubuntu2) ...
Job for nut-server.service failed because the control process exited with error code.
See "systemctl status nut-server.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nut-server, action "restart" failed.
● nut-server.service - Network UPS Tools - power devices information server
   Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-07-03 15:39:45 EDT; 27ms ago
  Process: 27218 ExecStart=/sbin/upsd **(code=exited, status=1/FAILURE)**

Jul 03 15:39:45 rock64 systemd[1]: Starting Network UPS Tools - power devices information server...
Jul 03 15:39:45 rock64 systemd[1]: nut-server.service: Control process exited, code=exited status=1
Jul 03 15:39:45 rock64 systemd[1]: nut-server.service: Failed with result 'exit-code'.
Jul 03 15:39:45 rock64 systemd[1]: **Failed to start Network UPS Tools - power devices information server.**
nut-driver.service is a disabled or a static unit not running, not starting it.
Setting up nut (2.7.4-5.1ubuntu2) ...
Processing triggers for systemd (237-3ubuntu10.53) ...
Processing triggers for libc-bin (2.27-3ubuntu1.6) ...

root@rock64:/etc# cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

root@rock64:/etc# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic

root@rock64:/etc# cat /proc/version
Linux version 4.4.132-1075-rockchip-ayufan-ga83beded8524 (root@runner-f725ff63-project-5943294-concurrent-0) (gcc version 7.3.0 (Ubuntu/Linaro 7.3.0-16ubuntu3) ) #1 SMP Thu Jul 26 08:22:22 UTC 2018

jimklimov, 2022-07-06:
Looking at the log (FWIW of an older Ubuntu installing older NUT) the exact issue is a distro/setup problem so something to solve with them (maybe backport packaging from newer releases).

Distro half of the issue is about installation order - client fails to start since there is no server to listen to, then the server is installed. Notably they would still fail to start until you configure the set for your devices, credentials, etc. so packaging should not depend on starting the daemons if it is not an upgrade. That is the setup half :)

jimklimov, 2022-07-17:
CC @bigon - can you please check if the problem can appear with the recent release packaging?