networkupstools/nut #1461: Get NUT 2.8.0 into the official Ubuntu repository

fatbasstard, 2022-07-04:
Hi,

Wild shot to put it here. But in the normal Ubuntu repositories (`https://packages.ubuntu.com/search?keywords=nut`) we've still got `2.7.4`, I'd like to get the `2.8.0` version in there. How can we arrange this?

biergaizi, 2022-07-04:
I think that reporting a bug to Ubuntu developers would be more effective than reporting a bug here. Here at the upstream, there's essentially no control over downstream distributions. I just reported a bug to Debian yesterday in #1433.

fatbasstard, 2022-07-05:
Hi @biergaizi,

Thanks for the pointer, exactly the reason why I still created the issue here. No clue where to go to and maybe somebody has "connections", so to speak ;)

biergaizi, 2022-07-05:
I just took a look at the Ubuntu package, it came directly from Debian. Once Debian updates the package, it should trickle down to Ubuntu soon. But it seems that the Debian maintainer of this package is not actively working on this so far (understandable since it's all volunteer works), I'll try to contribute a patch to accelerate the process.

biergaizi, 2022-07-07:
I just uploaded my unofficial Debian and Ubuntu packages to the Open Build System, based on my Debian non-maintainer upload patch. 

To install the package on Debian 11:

    # Trust signing key
    wget 'https://download.opensuse.org/repositories/home:/niconiconi/Debian_11/Release.key' -O /etc/apt/trusted.gpg.d/nut-unofficial.asc

    # Add Repo
    echo 'https://download.opensuse.org/repositories/home:/niconiconi/Debian_11/ ./' >> /etc/apt/sources.list

To install the package on Ubuntu 22.04

    # Trust signing key
    wget 'https://download.opensuse.org/repositories/home:/niconiconi/xUbuntu_22.04/Release.key' -O /etc/apt/trusted.gpg.d/nut-unofficial.asc

    # Add Repo
    echo 'https://download.opensuse.org/repositories/home:/niconiconi/xUbuntu_22.04/ ./' >> /etc/apt/sources.list

Note that these packages are 100% untested and may not function correctly. Don't forget to report back.

MattL0, 2022-07-07:
A ppa would be great!

jimklimov, 2022-07-07:
OBS should be able to serve as a (PPA?), RPM, DEB etc. style standard repo
for adventurous NUT users to update from regularly. At least, the on-prem
deployment maintained that for our 42ity project internal CI needs.

I've also seen promising Dockerfile efforts to containerize NUT (usually as
one driver + upsd server), but that feels like an overkill space-wise and
too complicated especially with random-usb device node passthrough or for
multiple-device monitoring, etc. It can be useful to test package installs
in many distros though, by just building a set of containers (and likewise
to prepare prereqs for a build quickly without "polluting" a workstation
main OS environment or keeping a persistent container as documented in NUT
sources for CI setup).

biergaizi, 2022-07-11:
Good news from the Debian side.

> The maintainer have merged your work:
> https://salsa.debian.org/debian/nut/-/commit/89ba505c47313daae843012a9dcec883fe11b390
>
> He did other improvements and new version was uploaded to experimental and now is waiting the ftp master review (may take days or weeks) for new binaries:
>
> https://ftp-master.debian.org/new/nut_2.8.0-1~exp1.html

closed 2022-07-17 by jimklimov

biergaizi, 2022-07-23:
I think this Ubuntu packaging bug should be reopened. Currently NUT 2.8.0 only exists as a Debian Experimental package, and has not been uploaded to even Debian Sid. Consequently, none of the Ubuntu repository has synced this version from Debian to Ubuntu.

biergaizi, 2022-10-22:
> Subject: nut 2.8.0-4 MIGRATED to testing
>
> FYI: The status of the nut source package
> in Debian's testing distribution has changed.
>
>   Previous version: 2.7.4-14
>   Current version:  2.8.0-4

https://tracker.debian.org/news/1376801/nut-280-4-migrated-to-testing/