networkupstools/nut #1462: scripts/python/app/NUT-Monitor is Missing From v2.8.0 Release Tarball

biergaizi, 2022-07-06:
When attempting to create a NUT package, I noticed that the file `scripts/python/app/NUT-Monitor` is missing from the release tarball, although the git record shows in exists in the v2.8.0 gin tree.

jimklimov, 2022-07-06:
Thanks for the nudge, though I won't take close care of this for a week at least.

FWIW among changes leading up to 2.8.0 there was a split of that app into two, for py2 and py3. I think the "legacy" named files (app, gnome manifest, etc) are covered by either symlinks created during build based on preferred python version on the platform, or by a wrapper script in some one case.

jimklimov, 2022-07-17:
Cheers, just in case: *how* did you make the package from tarball? In particular, wondering if you regenerated `configure` script or used the tarballed one?

I see that we do provide `configure.ac` and `m4/*` sources in the tarball, but do not include the `autogen.sh` with arcane recipes for whoever might want to regenerate it from scratch (involving augeas, perl and python stuff before calling common `autoreconf` tool).

jimklimov, 2022-07-17:
But overall yes, an `EXTRA_DIST` is missing to add the new `NUT-Monitor` wrapping script (selecting an implementation based on available python version) as well as for `nut-monitor.desktop` file to call it.

And also there is no `make install` target, perhaps to be handled by a `configure` script option like `enable-python-app` or some such.

closed 2022-07-20 by jimklimov

biergaizi, 2022-07-22:
> Cheers, just in case: _how_ did you make the package from tarball? In particular, wondering if you regenerated `configure` script or used the tarballed one?

I just downloaded the premade tarball from GitHub Release. I thought it's the standard for project maintainers to provide ready-to-use tarballs with all necessary build files pregenerated. I actually found this only when trying to patch Debian - which has a policy of, as we all familiar with, only using official tarballs.

> But overall yes, an EXTRA_DIST is missing to add the new NUT-Monitor wrapping script (selecting an implementation based on available python version) as well as for nut-monitor.desktop file to call it.

Alright.

jimklimov, 2022-07-23:
Yes, it is a standard procedure - and apparently an oversight of that feature change's review (mea culpa) and in a way CI (especially as adding "reference packaging" is a quest for the future, as a way to check what we deliver and what discrepancies arise).

commit 87ce3eeb (2022-07-20): scripts/python/Makefile.am: EXTRA_DIST the "dispatcher" script and applet manifest

Regression in #1354 for NUT v2.8.0 release
Closes: #1462