networkupstools/nut #1463: MUST EP2000 PRO is not supported

Temtaime, 2022-07-10:
Hello ! 

> \# lsusb
> Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics CH340 serial converter

Tried blazer_ser and nutdrv_qx.

```
Network UPS Tools - UPS driver controller 2.8.0
   0.000000     [D1] Starting UPS: ups
   0.000033     [D2] 1 remaining attempts
   0.000038     [D2] exec:  /lib/nut/nutdrv_qx -DDD -a ups -u nut
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000021     [D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000034     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'nut' specified on command line
   0.000039     [D1] debug level is '3'
   0.000632     [D1] upsdrv_initups...
   0.114128     [D3] send: 'QGS'
   1.115176     [D3] read: timeout (0)
   1.115196     [D2] qx_process_answer: short reply (input.voltage)
   1.115275     [D3] send: 'QGS'
   2.116321     [D3] read: timeout (0)
   2.116340     [D2] qx_process_answer: short reply (input.voltage)
   2.116424     [D3] send: 'QGS'
   3.117474     [D3] read: timeout (0)
   3.117493     [D2] qx_process_answer: short reply (input.voltage)
   3.117568     [D3] send: 'M'
   4.118597     [D3] read: timeout (0)
   4.118617     [D2] qx_process_answer: short reply (ups.firmware.aux)
   4.118693     [D3] send: 'M'
   5.119737     [D3] read: timeout (0)
   5.119755     [D2] qx_process_answer: short reply (ups.firmware.aux)
   5.119836     [D3] send: 'M'
   6.120882     [D3] read: timeout (0)
   6.120902     [D2] qx_process_answer: short reply (ups.firmware.aux)
   6.120977     [D3] send: 'M'
   7.122020     [D3] read: timeout (0)
   7.122037     [D2] qx_process_answer: short reply (ups.firmware.aux)
   7.122115     [D3] send: 'M'
   8.123163     [D3] read: timeout (0)
   8.123189     [D2] qx_process_answer: short reply (ups.firmware.aux)
   8.123278     [D3] send: 'M'
   9.124311     [D3] read: timeout (0)
   9.124328     [D2] qx_process_answer: short reply (ups.firmware.aux)
   9.124401     [D3] send: 'QS'
  10.124444     [D3] read: timeout (0)
  10.124463     [D2] qx_process_answer: short reply (input.voltage)
  10.124536     [D3] send: 'QS'
  11.125562     [D3] read: timeout (0)
  11.125580     [D2] qx_process_answer: short reply (input.voltage)
  11.125654     [D3] send: 'QS'
  12.126678     [D3] read: timeout (0)
  12.126695     [D2] qx_process_answer: short reply (input.voltage)
  12.126772     [D3] send: 'D'
  13.127816     [D3] read: timeout (0)
  13.127833     [D2] qx_process_answer: short reply (input.voltage)
  13.127905     [D3] send: 'D'
  14.128685     [D3] read: timeout (0)
  14.128706     [D2] qx_process_answer: short reply (input.voltage)
  14.128788     [D3] send: 'D'
  15.129831     [D3] read: timeout (0)
  15.129849     [D2] qx_process_answer: short reply (input.voltage)
  15.129929     [D3] send: 'Q1'
  16.130975     [D3] read: timeout (0)
  16.130995     [D2] qx_process_answer: short reply (input.voltage)
  16.131075     [D3] send: 'Q1'
  17.132122     [D3] read: timeout (0)
  17.132141     [D2] qx_process_answer: short reply (input.voltage)
  17.132217     [D3] send: 'Q1'
  18.132461     [D3] read: timeout (0)
  18.132480     [D2] qx_process_answer: short reply (input.voltage)
  18.132558     [D3] send: 'Q1'
  19.133606     [D3] read: timeout (0)
  19.133625     [D2] qx_process_answer: short reply (input.voltage)
  19.133698     [D3] send: 'Q1'
  20.134725     [D3] read: timeout (0)
  20.134744     [D2] qx_process_answer: short reply (input.voltage)
  20.134821     [D3] send: 'Q1'
  21.135847     [D3] read: timeout (0)
  21.135864     [D2] qx_process_answer: short reply (input.voltage)
  21.135942     [D3] send: 'Q1'
  22.136991     [D3] read: timeout (0)
  22.137011     [D2] qx_process_answer: short reply (input.voltage)
  22.137088     [D3] send: 'Q1'
  23.138132     [D3] read: timeout (0)
  23.138152     [D2] qx_process_answer: short reply (input.voltage)
  23.138231     [D3] send: 'Q1'
  24.139279     [D3] read: timeout (0)
  24.139300     [D2] qx_process_answer: short reply (input.voltage)
  24.139373     [D3] send: 'Q1'
  25.140428     [D3] read: timeout (0)
  25.140447     [D2] qx_process_answer: short reply (input.voltage)
  25.140524     [D3] send: 'Q1'
  26.141572     [D3] read: timeout (0)
  26.141592     [D2] qx_process_answer: short reply (input.voltage)
  26.141673     [D3] send: 'Q1'
  27.142719     [D3] read: timeout (0)
  27.142737     [D2] qx_process_answer: short reply (input.voltage)
  27.142746     [D2] find_nut_info: info type series not found
  27.142749     [D2] claim: cannot find series
  27.142754     [D2] find_nut_info: info type series not found
  27.142758     [D2] claim: cannot find series
  27.142762     [D2] find_nut_info: info type series not found
  27.142765     [D2] claim: cannot find series
  27.142839     [D3] send: 'Q1'
  28.144106     [D3] read: timeout (0)
  28.144124     [D2] qx_process_answer: short reply (input.voltage)
  28.144200     [D3] send: 'Q1'
  29.144462     [D3] read: timeout (0)
  29.144480     [D2] qx_process_answer: short reply (input.voltage)
  29.144557     [D3] send: 'Q1'
  30.145610     [D3] read: timeout (0)
  30.145630     [D2] qx_process_answer: short reply (input.voltage)
  30.145706     [D3] send: 'Q1'
  31.146733     [D3] read: timeout (0)
  31.146751     [D2] qx_process_answer: short reply (input.voltage)
  31.146827     [D3] send: 'Q1'
  32.147870     [D3] read: timeout (0)
  32.147890     [D2] qx_process_answer: short reply (input.voltage)
  32.147967     [D3] send: 'Q1'
  33.148460     [D3] read: timeout (0)
  33.148479     [D2] qx_process_answer: short reply (input.voltage)
  33.148559     [D3] send: 'Q1'
  34.149605     [D3] read: timeout (0)
  34.149626     [D2] qx_process_answer: short reply (input.voltage)
  34.149704     [D3] send: 'Q1'
  35.150732     [D3] read: timeout (0)
  35.150751     [D2] qx_process_answer: short reply (input.voltage)
  35.150826     [D3] send: 'Q1'
  36.151869     [D3] read: timeout (0)
  36.151888     [D2] qx_process_answer: short reply (input.voltage)
  36.151894     Device not supported!
  36.151898     Device not supported!
  36.154677     Driver failed to start (exit status=1)
```

Any advices how to setup this ups? :(

jimklimov, 2022-07-13:
Interesting, the only mention of `1a86` in NUT codebase I see is for the (recently added) `huawei-ups2000` driver.

It might be that the devices just happen to use the same USB interface chip, but give a shot to that driver - maybe the boxes are more closely related.

Temtaime, 2022-07-13:
Thanks, just compiled and tried it, not working :(

```
Network UPS Tools - UPS driver controller 2.8.0
   0.000000     [D1] Starting UPS: ups
   0.000023     [D2] 1 remaining attempts
   0.000029     [D2] exec:  /lib/nut/huawei-ups2000 -DDD -a ups -u nut
Network UPS Tools - NUT Huawei UPS2000 (1kVA-3kVA) RS-232 Modbus driver 0.02 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000     [D3] main_arg: var='driver' val='huawei-ups2000'
   0.000020     [D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000036     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'nut' specified on command line
   0.000041     [D1] debug level is '3'
   0.000626     [D2] upsdrv_initups
   0.016381     [D2] upsdrv_initinfo
   0.016387     [D2] ser_send_buf
   1.417537     unable to read from serial port /dev/ttyUSB0, retry...
   1.417556     [D2] ser_send_buf
   2.818698     unable to read from serial port /dev/ttyUSB0, retry...
   2.818714     [D2] ser_send_buf
   4.219866     unable to read from serial port /dev/ttyUSB0, retry...
   4.219882     unable to read from serial port /dev/ttyUSB0!
   4.221977     Driver failed to start (exit status=1)
```

jimklimov, 2022-07-17:
The "Unable to read" looks fishy. Did you stop the other driver (or any other process) using the `/dev/ttyUSB0` device node?

jimklimov, 2023-01-06:
Hm, one more thing - docs for huawei-2000 mention that a sufficiently recent linux kernel with a particular usb driver is required - got that in place?

CC @biergaizi

Temtaime, 2023-01-07:
It uses ch341 chip. yes.

> [    4.918685] usbcore: registered new interface driver ch341
> [    4.918694] usbserial: USB Serial support registered for ch341-uart
> [    4.918716] ch341 1-6:1.0: ch341-uart converter detected
> [    4.931453] usb 1-6: ch341-uart converter now attached to ttyUSB0

I made a dump. It seems to be something different from huawei.

There's connection initialization :

> <20230107 03:20.44:455 TX> 01 03 27 10 00 01 8F 7B 05 03 4E 21 00 01 C2 AC 06 03 4E 21 00 01 C2 9F 0A 03 75 30 00 01 9F 72
> <20230107 03:20.44:983 RX> 0A 03 02 00 00 1D 85
> <20230107 03:20.45:254 TX> 04 03 4E 20 00 07 12 BF 0A 03 75 30 00 1B 1E B9
> <20230107 03:20.45:745 RX> 0A 03 36 00 00 41 46 00 04 00 0C 02 58 08 B6 01 F3 08 C4 01 F3 00 04 00 5C 00 6C 00 0F 00 00 00 86 00 10 00 00 00 64 00 1F 00 00 00 00 00 00 00 00 00 02 00 01 00 00 00 01 4F 3B
>

biergaizi, 2023-01-07:
> Hm, one more thing - docs for huawei-2000 mention that a sufficiently recent linux kernel with a particular usb driver is required - got that in place?
> 
> CC @biergaizi

You got it backward. CH341 is an old and mature USB-to-serial/RS232 chip that has existed for around two decades and does not require any new driver, it's the go-to choice for millions of USB devices in all shapes and forms. In the case of Huawei UPS2000, it initially used a new and uncommon chip from MaxLinear/Exar, but in a later product revision, Huawei reverted to the generic CH341. Possible due to chip shortage or compatibility.

For the MUST EP2000 PRO, my best guess is that the protocol has been changed, and it's unrelated to the serial chip. Huawei UPS2000's cannot be used for other devices that happen to have the same serial chip in it.

> @Temtaime: I made a dump of com port when ups communicates with its software for windows. It seems to be something different from huawei.

Just to make sure the serial port is functional on Linux, perhaps you can try replaying the captured binary sequence of handshake, and see if at least you get a respond.

jimklimov, 2023-01-07:
Oh, sorry... Ok, thanks for clarification  :)

jimklimov, 2023-01-09:
@Temtaime : one more idea to test is if the UPS supports USB HID standard in fact (although use of `/dev/ttyUSB0` in the original post does not keep me hopeful), e.g. probe it by something like:
````
:; usbhid-ups -s must -d1 -DDDDDD -x port=auto -x vendorid=1a86 -x productid=7523
````

...would it data-dump anything useful?

jimklimov, 2023-01-09:
Otherwise, it remains that they use *some* serial(-over-USB) protocol that is not in Megatec Qx family. There are other serial-capable drivers in NUT that you could try.

https://en.must-ee.com/archives/category/upper-computer-software suggests that EP2000 is handled by "WinCC" and EP2000PRO by "PowerMonitor", names which do not ring a bell. Other devices are handled by "ViewPower" which nutdrv_qx man page does mention.

Perhaps you as a paying client can ask MUST for protocol/specs and written permission/license needed to develop a NUT driver (and find someone to code one), or better yet entice them to step up and contribute a NUT driver (promoting their sales along the way). One leverage point is that the two softwares mentioned above are for Windows, so users on other OSes are not covered. Given how their website is about inverters for off-grid survival (solar on trailers, etc.) nowadays some Raspberry Pi and HomeAssistant managing stuff sounds like a reasonable target audience.

pavlozelinskyi, 2023-09-28:
Has anyone found a way to connect this type of remote control to home assistant?

WildRat, 2023-09-28:
> Has anyone found a way to connect this type of remote control to home assistant?

I use Modbus Home Assistant extention. 
[image]
All information needed to setup sensors can be found here:
[EP3000Plus Modbus RTU Protocol1.0 - Eng.xlsx](https://github.com/networkupstools/nut/files/12747904/EP3000Plus.Modbus.RTU.Protocol1.0.-.Eng.xlsx)

pavlozelinskyi, 2023-09-28:
> I use Modbus Home Assistant extention. [image] All information needed to setup sensors can be found here: [EP3000Plus Modbus RTU Protocol1.0 - [...]

Can you show me an example of one of the sensors?  for some reason it shows "unavailable"

WildRat, 2023-09-28:
Here some of my sensors:
[image]

pavlozelinskyi, 2023-10-05:
> Here some of my sensors: [image]

Thank you very much!  Everything works great!
[image]

hamster66, 2024-07-06:
Hi all.
Are there any changes to the topic? Are there any chances that this device will receive a working driver for NUT?