networkupstools/nut #1464: Possible bashism in ./scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in

bigon, 2022-07-10:
Hello,

`./scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in` shebang is set to `/bin/bash` but seems to use bash only construct

The shebang should explicitly be set to `/bin/bash` or the code changed

```
$ checkbashisms ./scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in
possible bashism in ./scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in line 835 (alternative test command ([[ foo ]] should be [ foo ])):
    egrep -v '(^$|^#)' | \
    sed -e 's,^['"$TABCHAR"'\ ]*,,' \
        -e 's,^\#.*$,,' \
        -e 's,['"$TABCHAR"'\ ]*$,,' \
        -e 's,^\([^=\ '"$TABCHAR"']*\)['"$TABCHAR"'\ ]*=['"$TABCHAR"'\ ]*,\1=,g' \
        -e 's,=\"\([^\ '"$TABCHAR"']*\)\"$,=\1,' \
        -e 's,^\(\[[^]'"$TABCHAR"'\ ]*\]\)['"$TABCHAR"'\ ]*\(#.*\)*$,\1,' \
    | egrep -v '^$' \
    | egrep '([\[\=]|^[^ '"$TABCHAR"']*$|^[^ '"$TABCHAR"']*[ '"$TABCHAR"']*\#.*$)'
possible bashism in ./scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in line 835 ([^] should be [!]):
    egrep -v '(^$|^#)' | \
    sed -e 's,^['"$TABCHAR"'\ ]*,,' \
        -e 's,^\#.*$,,' \
        -e 's,['"$TABCHAR"'\ ]*$,,' \
        -e 's,^\([^=\ '"$TABCHAR"']*\)['"$TABCHAR"'\ ]*=['"$TABCHAR"'\ ]*,\1=,g' \
        -e 's,=\"\([^\ '"$TABCHAR"']*\)\"$,=\1,' \
        -e 's,^\(\[[^]'"$TABCHAR"'\ ]*\]\)['"$TABCHAR"'\ ]*\(#.*\)*$,\1,' \
    | egrep -v '^$' \
    | egrep '([\[\=]|^[^ '"$TABCHAR"']*$|^[^ '"$TABCHAR"']*[ '"$TABCHAR"']*\#.*$)'
```

jimklimov, 2022-07-10:
Thanks! Reviewed, but... Given how this complains about sed regex with no shell test commands around (probably the bracket-slash-bracket sequence), it is a bug in the checkbashisms ;)

At least, the script is intentionally CI tested with many interpreters including dash, ksh and busybox and whatever /bin/sh is on several OSes, and works as expected.

Also, typo in your report - shebang is /bin/sh not /bin/bash here.

bigon, 2022-07-11:
OK then, if you already have CI testing this and that everything is working, I guess we can close this

closed 2022-07-11 by jimklimov