networkupstools/nut #1468: Eaton Ellipse Eco 650USB - missing some information

janschmidt76, 2022-07-12:
I've successfully connected an Eaton Ellipse Eco 650USB to a RPi3B+ running Debian 11 (aarch64) and queried the UPS with nut 2.7.4-13. I was wondering about some missing information, e.g. the serial number or input voltage. That lead me to the idea, to give the new package version 2.8.0-exp1 a try. But sadly, some information is still lacking. Is this because the UPS is simply not providing this information or are there any additional tweaks to get this?

```
pi@eaton-ups-outbound:~ $ upsc eaton
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 2156
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 650
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000109)
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 0
ups.mfr: EATON
ups.model: Ellipse ECO 650
ups.power.nominal: 650
ups.productid: ffff
ups.realpower: 0
ups.serial: 000000000
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463
```

aquette, 2022-07-12:
What does lsusb -v returns?

janschmidt76, 2022-07-13:
I've attached the lsusb output.

[lsusb_output.txt](https://github.com/networkupstools/nut/files/9098853/lsusb_output.txt)

aquette, 2022-07-13:
iSerial is 0000000 there too, so either a bug of libusb on rpi, either from the device, and not a privilege issue since you're root. I'd suspect the first, but I'm biased 😅 You can try on a x86 to confirm. Another option is to build with libusb 0.1

jimklimov, 2022-07-13:
I think I saw only 00000000's over USB from Eaton units on various systems and libusb versions...

janschmidt76, 2022-07-14:
I've tried the same package on a laptop running Debian 11. upsc output is the same. I think even the Eaton Companion shows 00000000 for the serial. That might be a UPS firmware issue. But that's not that important to me. I'm more annoyed about missing "relevant" information, like input voltage. Can I somehow give some more lowlevel debug information to see, whether the UPS is not providing the data or NUT is not interpreting it correctly?

```
root@debian-test:~# upsc eaton
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 2156
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 650
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Ellipse ECO
driver.parameter.productid: FFFF
driver.parameter.serial: 000000000
driver.parameter.synchronous: auto
driver.parameter.vendor: EATON
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000109)
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 6
ups.mfr: EATON
ups.model: Ellipse ECO 650
ups.power.nominal: 650
ups.productid: ffff
ups.realpower: 31
ups.serial: 000000000
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463
```

clepple, 2022-07-17:
Two ways to check for additional USB HID Usage IDs:

1. Run the NUT driver at least once after plugging the UPS USB cable into the host, and then stop NUT (potentially `systemctl stop nut-server`? key point is to not have `usbhid-ups` running, but I forget which versions of Debian move the drivers into their own service). Then check `sudo lsusb -vvv -d0463:` for information in place of `UNAVAILABLE`:
```
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID              10.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     909
         Report Descriptors:
           ** UNAVAILABLE **
```

2. Stop NUT if running (driver doesn't need to be run once beforehand as for the previous method). Run `sudo /lib/nut/usbhid-ups -a eaton -DD | tee /tmp/eaton-ellipse-eco-650usb.txt`

clepple, 2022-07-17:
The previous comment probably sounds a bit like voodoo, so I should explain:

`lsusb` can't fetch additional descriptors (such as the HID Report Descriptor) unless it has R/W privileges to the device node *and* no kernel or user-space programs have the device open (technically, just the interface open, but most USB devices only have one interface). Running the `usbhid-ups` driver once will detach the kernel driver from the UPS, and stopping the driver will allow other programs (such as `lsusb`) to interact with it.

The second method is a shortcut version of this procedure since the Eaton USB VID is already known to the driver: https://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#_writing_a_subdriver

I'm suspicious of the six nearly identical configurations listed in `lsusb` for the UPS, but then again, they all seem to have the same wDescriptorLength, so it would be surprising if one of them provided a different, expanded set of variables.

janschmidt76, 2022-07-18:
I've attached both outputs. Sadly it looks to me, that the UPS is simply not giving the information on the input values
[eaton-ellipse-eco-650usb.txt](https://github.com/networkupstools/nut/files/9135519/eaton-ellipse-eco-650usb.txt)
[lsusb.txt](https://github.com/networkupstools/nut/files/9135521/lsusb.txt)
.

jimklimov, 2022-08-28:
Got hold of same model, also did not find the "UPS.PowerConverter.Input.anything" in it. Curiously, the boost/trim "input.*" voltages are mapped from "PowerConverter.Output". @aquette : is that intentional?