networkupstools/nut #1498: Get NUT 2.8.x into the official OmniOS repository

jimklimov, 2022-07-17:
@hadfl on #omnios IRC channel said he would take a look at this (needs to provide libusb first). NUT and dependencies are expected to appear in `omnios-extra` repository.

thenktor, 2024-05-06:
Any updates on this?

jimklimov, 2024-05-06:
Not on my side, hope @hadfl would say more :)

I did bump OpenIndiana recipes which were then iterated a few times by other OI community members - this can be a helpful starting point IMHO, whether regarding build arguments or possible OS-adaptation patches, or the recommended split into numerous packages based on the role of a set of programs or grouped by non-ubiquitous dependencies (e.g. stuff that depends on net-snmp has a different installation footprint than USB drivers).

Speaking of which, probably gotta bump for NUT v2.8.2 :)

One important note: there is a new `apc_modbus` NUT driver in recent releases that can use Serial and TCP RTU with stock libmodbus, but for USB it would require a custom build from a branch that AFAIK was not yet merged into upstream - stalled on no reply for SLA submission from patch author. This library can be built and linked statically into the driver program, to avoid conflicts with stock implementation - see https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol for details.

thenktor, 2024-05-07:
Thanks for your reply.
I've found NUT in the [pkgsrc repository for illumos](https://pkgsrc.smartos.org/), so I'll give that version a try.

discard-this, 2026-01-20:
Any news on this? I'd really love to have an "official" package available.