networkupstools/nut #1506: kstar MP 3k S

lukasa1993, 2022-07-18:
i have kstar MP 3k S any idea which driver might work ?

this is how it shows in lsusb

```Bus 001 Device 005: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P```

i tried with no luck these 2 configs:

```
[kstar]
driver = blazer_usb
port = auto
```

and

```
[kstar]
driver = nutdrv_qx
port = auto
```

jimklimov, 2022-07-30:
You might try same drivers with such options:
````
vendorid=067b
productid=2303
````
since such numbers are not currently known to the NUT codebase so no driver matches them as "theirs".
Is there any vendor info on what protocol they use, perhaps which (Windows?) software they ship to monitor the UPS?

ALF-2021, 2022-07-30:
Where I can find that option?

Thanks.

jimklimov, 2022-07-30:
Hey, I edited the GitHub issues post after initial submission - please check https://github.com/networkupstools/nut/issues/1506#issuecomment-1200199095 for the latest version ;}

lukasa1993, 2022-07-30:
tried 

```
[kstar]
driver = nutdrv_qx
port = /dev/ttyUSB0
vendorid = "067b"
productid = "2303"
```

with error:
`Can't connect to UPS [kstar] (nutdrv_qx-kstar): No such file or directory`

and 

```
[kstar]
driver = blazer_usb
port = /dev/ttyUSB0
vendorid = "067b"
productid = "2303"
```

with error:
`Can't connect to UPS [kstar] (blazer_usb-kstar): No such file or directory`

jimklimov, 2022-08-01:
The error wording seems to indicate that `upsd` (data server) could not connect to the socket file of the driver instance. Did you start both the driver and the data server? (And further some client like `upsmon` usually, to react to power events and shut down when needed)

lukasa1993, 2022-08-01:
> The error wording seems to indicate that `upsd` (data server) could not connect to the socket file of the driver instance. Did you start both the driver and the data server? (And further some client like `upsmon` usually, to react to power events and shut down when needed)

:( probably i don't know how to do that, any good guide i can follow ?

lukasa1993, 2022-08-05:
> Do you mean a guide about starting NUT parts? Tons on the internet, including project's docs.
> […](#)

cant find anything which would help me

jimklimov, 2022-08-08:
From your post I can assume you are running on some Linux distro, and probably with a packaged NUT (not custom build) so probably 2.7.4 release. Depending on distribution, that may be started by systemd services or by init-scripts. Those handle startup (and shutdown) of separate NUT components, for an all-in-one system that is typically the driver(s) for your power hardware, NUT data server (upsd), and its client to react to power events (upsmon).

Here you would need to ensure the driver startup, so look for an init-script or service (probably named `nut-driver` or similar) which in NUT releases 2.7.4 and older likely launches the `upsdrvctl` program to start all the driver(s) you have defined. Release 2.8.0 may have it more granular, and allows to pass debug for troubleshooting (but that may be needed at later stage).

lukasa1993, 2022-08-08:
i am on ubuntu server  22.04, i cant find `nut-driver` but `upsdrvctl` is there and with both types of configs these are results
```sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

Driver failed to start (exit status=1)
```

```sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```

jimklimov, 2022-08-09:
Can you try running the driver directly, like `nutdrv_qx -a kstar -DDDDDD` to see detailed debug?
Possibly do so from a custom NUT build workspace of current git codebase, its logging may be more sensible.

lukasa1993, 2022-08-24:
> Can you try running the driver directly, like `nutdrv_qx -a kstar -DDDDDD` to see detailed debug?
> Possibly do so from a custom NUT build workspace of current git codebase, its logging may be more sensible.

that was helpful managed to make it work in some sense

jimklimov, 2022-09-12:
Well, it seems to show *something* beside driver's own info - e.g. `input.frequency`, `input.voltage`, `output.voltage`, `ups.temperature` seem sane?

It may be that not "all" values are served by the driver for a device (if it lacks mapping knowledge or other ways to learn that info).

In particular for `battery.charge` it may be related to discussions in #1279 / #1652 or #1512

closed 2024-12-18 by jimklimov