networkupstools/nut #1513: Eaton 5SC shutting down when not supposed to

sfulham, 2022-07-22:
Hey guys,
I have an Eaton 5SC hooked up to my Dell Poweredge R710 via USB to NUT, on XCP-ng, and I've noticed that every night, at 10pm roughly, the servers are shutting down, even though the UPS still has power.
The only logs that I can find that are relevant are the logs from the shutdown script I am using, and that only tells me that the script was run because it got the signal to shutdown from NUT.

Any ideas or pointers?

sfulham, 2022-07-22:
Running `usbhid-ups -DDDDD -a eatonfivesc -k` results in https://pastebin.com/NdwZKVWP

jimklimov, 2022-07-23:
Some time ago there was a bug report about incorrect USB processing on Eaton devices (IIRC) that could cause weird status and upsmon reacted by shutdown. Not sure if that appeared after 2.7.4 release as in your package though.

sfulham, 2022-08-21:
Looks like the issue was a smart power plug I had the UPS connected to. I'm still not sure why it instantly shut down rather than waiting 300 seconds like I set in the config

closed 2022-08-21 by sfulham