networkupstools/nut #1516: Advice AIN850-3

moriel5, 2022-07-24:
While I barely understand what I am doing, and thus I still don't have full details (I have no idea how to fill the device variables, nor where to place them), I have managed to get the Advice AIN850-3 working with NUT, on OpenWrt, using the `blazer_usb` driver (the UPS appears as `MEC MEC0003` in `lsusb`.

I also have reason to believe that the AIN650-3 should be pretty much the same, and possibly the AIN1000-3, AIN1200-3, AIN1500-3 and the AIN2000-3 should be very similar, though I have no way to verify this.

As proof:
[image]

And this is the data I can see with NUT CGI:

battery.charge | : | 100
-- | -- | --
battery.voltage | : | 14.00
battery.voltage.high | : | 13.00
battery.voltage.low | : | 10.40
battery.voltage.nominal | : | 12.0
device.type | : | ups
driver.name | : | blazer_usb
driver.parameter.pollinterval | : | 2
driver.parameter.port | : | auto
driver.parameter.synchronous | : | no
driver.version | : | 2.7.4
driver.version.internal | : | 0.12
input.current.nominal | : | 3.0
input.frequency | : | 50.0
input.frequency.nominal | : | 50
input.voltage | : | 238.2
input.voltage.fault | : | 238.2
input.voltage.nominal | : | 230
output.voltage | : | 238.2
ups.beeper.status | : | enabled
ups.delay.shutdown | : | 30
ups.delay.start | : | 180
ups.load | : | 3
ups.productid | : | 0000
ups.status | : | OL
ups.type | : | offline / line interactive
ups.vendorid | : | 0001

moriel5, 2022-07-24:
This is what I got when running `upsdrvctl start`:

>Network UPS Tools - UPS driver controller 2.7.4
>Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
>Supported UPS detected with megatec protocol
>Vendor information unavailable
>No values provided for battery high/low voltages in ups.conf
>
>Using 'guestimation' (low: 10.400000, high: 13.000000)!
>Battery runtime will not be calculated (runtimecal not set)

I also have no idea whether this will be persistent, or if I need to configure something additional in order to get NUT to properly communicate with the PSU even after a reboot.

jimklimov, 2022-07-25:
Thanks for the report.

Just in case: is voltage around 238V correct for your location?

As for persisting the setup, if your device settings are saved in an ups.conf file in a place that survives reboot, that should be ok.

If you know your battery model/tech specifics, you can provide something like `override.battery.voltage.low = 12.3` in that device config section. Otherwise the defaults would be used.

For devices/protocols that do not report their own estimate of remaining runtime, NUT can try to guess that based on discharge speed - but it really needs to know some numbers for the guesstimation to be somewhat relevant, so IIRC the message is about that.

moriel5, 2022-07-25:
> Thanks for the report.
No problem, I'm glad to be of help.

> Just in case: is voltage around 238V correct for your location?
It certainly is, the official regulation is around 230V, and anywhere between 220V-240V technically counts.

> As for persisting the setup, if your device settings are saved in an ups.conf file in a place that survives reboot, that should be ok.

That's the thing, for some odd reason the automatic creation of `ups.conf` did not add the settings, I had to add them myself before NUT was able to communicate with the PSU, and the first time and, I tried restarting the services and the changes to the file were reverted (this was before I rebooted so as to allow the driver to even start up, so that was alright).

> If you know your battery model/tech specifics, you can provide something like override.battery.voltage.low = 12.3 in that device config section.

It is a 9Ah AGM battery pack, and I have access to the specsheet. I will be replacing the battery pack however, hopefully this week, with a smaller LiFePO4 battery pack (smaller due to size constraints and availability of cells, the business that will be making it for me only have 32700 LiFoPO4 cells).

>The docs directory in github sources can provide more details (e.g., there is also a default.* prefix that may be good for your case).
Thanks, I'll try taking a deeper look there.

> For devices/protocols that do not report their own estimate of remaining runtime, NUT can try to guess that based on discharge speed - but it really needs to know some numbers for the guesstimation to be somewhat relevant, so IIRC the message is about that.
That certainly makes sense.
Thank you very much for the help.

moriel5, 2022-07-26:
For some odd reason, I saw my previous post twice, and deleting the second quarter to instead delete the first, but in reality center both. This is really weird.

moriel5, 2022-07-26:
@jimklimov In the end, it did not persist, a reboot cleared away `ups.conf` again.

moriel5, 2022-07-26:
I'm going to try asking in the OpenWrt forums what file is supposed to be the source for UCI's automatic configuration of `/etc/nut/ups.conf`, and then set `upsdrvctl start` and `upsd` to start up automatically upon boot.