networkupstools/nut #1519: Tecnoware era plus 1100 no supported

ALF-2021, 2022-07-27:
Hello I'm trying to integrate UPS Tecnoware ERA PLUS 1100 in Home assistant on Raspberry but the log says me "No supported UPS detected"

On device compatibility list there is. 

Someone has the same problem?

Thanks.

jimklimov, 2022-07-28:
The HCL (and DDL) usually mention a driver the device was tested with. Can you try with the same?

Other than that, there can be bugs as NUT evolves, or your version could be one from before it was added, or different HW/FW can be marketed under same name...

Also Raspberries have a fair share in recent USB driver issues reports, but I am not sure if that says anything about HW/SW quality or just that they are popular for tinkering these days.

ALF-2021, 2022-07-28:
Hello,
I’m using this config:

devices:
  - name: tecnoware
    driver: blazer_usb
    langid_fix: 0x409
    desc: Technoware Era Plus 1100
    port: auto
    config: []
There is something wrong?

Thanks best regards.

ALF-2021, 2022-07-28:
Hello,
Below my log:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.11.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 8.4  (aarch64 / raspberrypi3-64)
 Home Assistant Core: 2022.7.7
 Home Assistant Supervisor: 2022.07.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
cont-init: info: running /etc/cont-init.d/02-set-timezone.sh
[22:29:07] INFO: Configuring timezone
cont-init: info: /etc/cont-init.d/02-set-timezone.sh exited 0
cont-init: info: running /etc/cont-init.d/nut.sh
[22:29:08] INFO: Setting mode to netserver...
[22:29:08] INFO: Generating /etc/nut/upsd.users...
[22:29:10] INFO: Configuring user: nutty
[22:29:12] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[22:29:16] INFO: Configuring Device named tecnoware...
[22:29:17] INFO: Starting the UPS drivers...
No supported UPS detected
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
Network UPS Tools - UPS driver controller 2.7.4
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Thanks. 

Best regards.

jimklimov, 2022-07-29:
Is it possible for you to log into that system, or tweak the init-scripts, and run the driver with some -D options? I wonder if it opens the USB device nodes or e.g. lacks access to any (system grabbed them and nothing like udev rules to reassign to nut account)?

Does HA have a way to generate ups.conf with a line to run the driver as root?

ALF-2021, 2022-07-30:
> Is it possible for you to log into that system, or tweak the init-scripts, and run the driver with some -D options? I wonder if it opens the USB device nodes or e.g. lacks access to any (system [...]

If you want I can share the log.

jimklimov, 2022-07-30:
Yes, a log of driver start with bumped debug verbosity could be helpful.

Note that NUT v2.7.4 is a pretty old release (was latest until this spring, but still quite old) so it may help to ask on Home Assistant community venues if someone can bump the plugin to use NUT v2.8.0 or master?

ALF-2021, 2022-07-30:
Which is the repo to install 2.8.0?

I have to use HACS?

Thanks. 

Best regards.

jimklimov, 2022-07-30:
Sorry, got no idea about HA. About this - ask them please.

ALF-2021, 2022-07-31:
Hello,
maybe this can help you to identify my UPS

Thanks. 

Best regards.

jimklimov, 2022-08-01:
One more note: in https://github.com/networkupstools/nut/issues/590#issuecomment-966777679 and further down users of Home Assistant shared how they built a custom container with current NUT. Maybe that could help with your troubleshooting?

ALF-2021, 2022-08-06:
This is my config. 

[image]

This is the log. 

[image]

Someone can help me?

Thanks.

jimklimov, 2022-08-09:
Looking at one detailed report in DDL https://networkupstools.org/ddl/Tecnoware/UPS_ERA_PLUS_1100.html I see:
````
driver.name | blazer_usb
driver.parameter.pollinterval | 2
driver.parameter.port | auto
driver.parameter.synchronous | no
````

Not sure if langid fix which you use applies to this case or should be omitted. 
Might also help to confirm that the vendorid and productid you specify are actually those reported by the UPS.

Running a recent version of NUT driver with debug mode, perhaps as a custom-built program (right from the build area, maybe as root to easily allow USB device access) on a real Linux, would help a lot in guessing what the problem can be.

ALF-2021, 2022-10-11:
Hello,
I don’t know. 

If you I can share with you some log file or make some test if you drive me. 

Thanks. 

Best regards.

jimklimov, 2022-10-11:
Never used HA, sorry. Google is our friend, I guess...

jimklimov, 2022-11-09:
To rule out some circumstances:
* is this running in a VM/container? (recent ESXi tickets have similar data dumps)
* if this is a custom build, did you `sudo make install` it to deliver in particular the new `udev.rules` or `udev.hwdb` file with NUT info? Do you have both of these files under `/etc` (from older package and newer custom build, might convey conflicting info then)?
* are your NUT drivers configure to run as the same user account you have in those rules?

The recently added #1694 feature seems to help data-mining here:
````
   0.001086     [D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
````

Seems you ran `configure` with default settings, so it did not customize the user account to `nut` or `ups` as packaging usually does. If your system has the `udev` rules delegating USB access to that account and not `nobody` - here is one possible culprit.

This one also seems suspicious:
````
  15.626267     Got disconnected by another driver: Device or resource busy
````

Do you have another instance of the driver running (or trying to)? Notably, with new NUT releases the `nut-driver-enumerator.(path|service)` can create service instances for each driver section in `ups.conf` which helps out-of-the-box setups but may confuse a bit the manual troubleshooting (you may have to `systemctl disable nut-driver@sectionname` or at least `stop` it while experimenting).

ynuyasha, 2022-11-10:
Thank you for your reply.

>     * is this running in a VM/container? (recent ESXi tickets have similar data dumps)
No it is an HP Genserver

>     * if this is a custom build, did you `sudo make install` it to deliver in particular the new `udev.rules` or `udev.hwdb` file with NUT info? Do you have both of these files under `/etc` (from older package and newer custom build, might convey conflicting info then)?
Yes after some digging i found that the default location that use make install in debian 10 is /lib/udev so i rerun with the correct command
`./configure --with-user=nut --with-group=nut --with-udev-dir=/etc/udev`
and now i have rules correctly placed in /etc/udev/

```
ll /etc/udev/rules.d/
totale 48
-rw-r--r-- 1 root root 18607 giu 27  2021 61-openmediavault-dev-disk-by-id.rules
-rw-r--r-- 1 root root 14508 nov 10 7:53 62-nut-usbups.rules
-rw-r--r-- 1 root root     0 dic  2  2017 70-persistent-cd.rules
-rw-r--r-- 1 root root  1348 mag 14  2021 99-openmediavault-md-raid.rules
-rw-r--r-- 1 root root  1647 mag 14  2021 99-openmediavault-nonrot.rules
-rw-r--r-- 1 root root  1378 nov  2  2021 99-openmediavault-scheduler.rules
```
> 
>     * are your NUT drivers configure to run as the same user account you have in those rules?
> 
now yes after rebuild with correct configure

```
root@openmediavault:~# cat /etc/udev/rules.d/62-nut-usbups.rules |grep 0665
ATTR{idVendor}=="0665", ATTR{idProduct}=="5161", MODE="664", GROUP="nut"
```
```
root@openmediavault:~# NUT_STATEPATH=/tmp /usr/local/ups/bin/blazer_usb -a ups-tecnoware -DDDD
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0-Windows-63-g55042401d)
   0.000000     [D1] debug level is '4'
   0.001093     [D1] Succeeded to become_user(nut): now UID=116 GID=122
```
> 
> Do you have another instance of the driver running (or trying to)? 

i would say no now but i'm not sure at 100%. There is a way to understand it at 100%?
i have purged nut deb packages before doing the installation with sources and after that i have disabled service auto created as you suggested with `systemctl disable nut-driver@ups-tecnoware` and i didn't find anymore logs inside syslog correctly

Actually the new output is this
```
root@openmediavault:~# NUT_STATEPATH=/tmp /usr/local/ups/bin/blazer_usb -a ups-tecnoware -DDDD
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0-Windows-63-g55042401d)
   0.000000     [D1] debug level is '4'
   0.001093     [D1] Succeeded to become_user(nut): now UID=116 GID=122
   0.072253     [D3] usb_busses=0x55b6fd340070
   0.072301     [D2] Checking device (1D6B/0003) (003/001)
   0.119897     [D1] libusb_open get iManufacturer failed, retrying...
   0.119947     [D1] libusb_open get iManufacturer failed, retrying...
   0.119968     [D1] libusb_open get iManufacturer failed, retrying...
   0.119985     [D1] libusb_open get iProduct failed, retrying...
   0.120000     [D1] libusb_open get iProduct failed, retrying...
   0.120016     [D1] libusb_open get iProduct failed, retrying...
   0.120041     [D1] libusb_open get iSerialNumber failed, retrying...
   0.120057     [D1] libusb_open get iSerialNumber failed, retrying...
   0.120072     [D1] libusb_open get iSerialNumber failed, retrying...
   0.120086     [D2] - VendorID: 1d6b
   0.120099     [D2] - ProductID: 0003
   0.120114     [D2] - Manufacturer: unknown
   0.120133     [D2] - Product: unknown
   0.120147     [D2] - Serial Number: unknown
   0.120166     [D2] - Bus: 003
   0.120180     [D2] - Device: 001
   0.120193     [D2] - Device release number: 0419
   0.120205     [D2] Trying to match device
   0.120219     [D3] match_function_regex: matching a device...
   0.120236     [D2] Device does not match - skipping
   0.120353     [D2] Checking device (0781/5583) (002/002)
[... 70 lines cut ...]
   0.302952     [D3] read: timeout
   0.303349     [D2] blazer_status: short reply
   0.303637     [D2] Status read 2 failed
  10.464224     [D3] send: error sending control message: Connection timed out
  10.464273     [D2] blazer_status: short reply
  10.464286     [D2] Status read 3 failed
  10.464297     [D2] Trying mustek protocol...
  10.464323     [D3] send: error sending control message: Device or resource busy
  10.464377     Got disconnected by another driver: Device or resource busy
```

and this is the syslog when i execute manually the blazer_usb:
```
Nov 10 07:41:34 openmediavault kernel: [ 1778.353199] usb 1-1.3: usbfs: process 5669 (Thread-13) did not claim interface 0 before use
Nov 10 07:41:34 openmediavault kernel: [ 1778.525438] usb 1-1.3: usbfs: process 5669 (Thread-13) did not claim interface 0 before use
Nov 10 07:41:34 openmediavault kernel: [ 1778.525454] usb 1-1.3: usbfs: process 5669 (Thread-13) did not claim interface 0 before use
Nov 10 07:41:34 openmediavault kernel: [ 1778.532126] usb 1-1.3: usbfs: process 16704 (blazer_usb) did not claim interface 0 before use
Nov 10 07:41:44 openmediavault kernel: [ 1788.689202] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd blazer_usb rqt 33 rq 9 len 8 ret -110
Nov 10 07:41:44 openmediavault kernel: [ 1788.689339] usb 1-1.3: usbfs: process 16704 (blazer_usb) did not claim interface 0 before use
Nov 10 07:41:45 openmediavault kernel: [ 1789.221339] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd Thread-13 rqt 33 rq 9 len 8 ret -110
Nov 10 07:42:00 openmediavault kernel: [ 1804.165202] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd Thread-13 rqt 33 rq 9 len 8 ret -110
Nov 10 07:42:15 openmediavault kernel: [ 1819.205466] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd Thread-13 rqt 33 rq 9 len 8 ret -110
Nov 10 07:42:29 openmediavault kernel: [ 1834.053373] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd Thread-13 rqt 33 rq 9 len 8 ret -110
Nov 10 07:42:44 openmediavault kernel: [ 1848.997387] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd Thread-13 rqt 33 rq 9 len 8 ret -110
```

jimklimov, 2022-11-10:
Thanks for the details. Did you restart udev service and replug the USB connection after `make install`? Or reboot the system for similar effect?

Helped me in a recent setup...

To check for running drivers, `ps -ef | egrep '(nut|ups)` usually helps :)

When/if the current problems get resolved, don't forget to reenable the nut-driver service instance ;)