networkupstools/nut #1520: Issue - Raspberry Pi - Random disconnects -  Data stale - Driver not connected

ansred, 2022-07-28:
Greetings,

I have deployed NUT server on the Pi recently, everything is up to date and connected two Synology DSM as NUT client to it.

The issue is sometimes the NUT server on the Pi gets disconnects.

```
root@raspberrypi:~# service nut-client status
? nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-07-25 21:15:17 EDT; 12h ago
    Process: 9413 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 9415 (upsmon)
      Tasks: 2 (limit: 4915)
        CPU: 2.205s
     CGroup: /system.slice/nut-monitor.service
             ??9414 /lib/nut/upsmon
             ??9415 /lib/nut/upsmon

Jul 25 22:09:27 raspberrypi upsmon[9415]: Poll UPS [ups@localhost] failed - Data stale
Jul 25 22:09:32 raspberrypi upsmon[9415]: Poll UPS [ups@localhost] failed - Data stale
Jul 25 22:09:37 raspberrypi upsmon[9415]: Poll UPS [ups@localhost] failed - Driver not connected
Jul 25 22:09:42 raspberrypi upsmon[9415]: Communications with UPS ups@localhost established
Jul 26 00:20:18 raspberrypi upsmon[9415]: Poll UPS [ups@localhost] failed - Driver not connected
Jul 26 00:20:18 raspberrypi upsmon[9415]: Communications with UPS ups@localhost lost
Jul 26 00:20:23 raspberrypi upsmon[9415]: Communications with UPS ups@localhost established
Jul 26 01:05:39 raspberrypi upsmon[9415]: Poll UPS [ups@localhost] failed - Driver not connected
```

To fix this, I have to run `upsdrvctl start` and everything is normal until 

```
root@raspberrypi:~# sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Duplicate driver instance detected! Terminating other driver!
Using subdriver: APC HID 0.96

```

Any idea what is causing this issue with the NUT server/client on the Pi?

jimklimov, 2022-07-29:
There were several reports of issues with Raspberry, some suggested that the USB port might be not stable there - if you have a chance to add a powered USB hub to handle the link, it might help. Trawl these issues for key words :)

Also, 2.7.4 is not quite "all up to date" - some distros take their time updating to the newest release (2.8.0 currently).

ansred, 2022-07-29:
Thanks for the response @jimklimov 

I see. Will look into that powered USB hub, even though the data usb cable is connected directly from the APC UPS to the Raspberry Pi. Do you have use a specific brand by any chance to share Amazon URL?

Do you have any good instructions on upgrading the Pi UP driver from 2.7.4 to the latest?

jimklimov, 2022-07-29:
Unfortunately no, don't have specifics except this idea was reminded of in some discussion a week or two ago (originally was earlier) so you might search by last-update of issue posts.

For updates in-place, it usually goes about finding package recipes for your OS and substituting current sources, and then `sudo make install` to overwrite. Here it could suffice to juggle just the built `usbhid-ups` driver.

Another approach is to keep defaults (maybe twezk runtime user), let it install in `/usr/local(/ups?)` and copy or symlink the config files, and so minimally intrude on packaged stuff (replacing just systemd units, maybe augeas lens files).

ansred, 2022-07-30:
It would be nice to have some kind a workaround maybe to have automated bash script to be triggered whenever these disconnects happens to run the command `sudo upsdrvctl start`

That should fix the issue, at least in the meantime while investigating the root cause of this strange issue.

Any chance you have such a script I can add to the Pi?

Thanks

jimklimov, 2022-07-31:
Currently NUT 2.8.0 should provide templated nut-driver@.service files so at least if a driver process dies, systemd should revive it (similarly for Solaris/illumos SMF).

It may be harder if the driver does not die but tries to reconnect somehow in vain, which may be riddled with system USB stack details (port resets, permissions changes, re-enumeration of devices...) so maybe ensuring that a "stale" driver gets fully recycled may be a solution on such systems.

jimklimov, 2022-08-02:
I looked for a script I had sometime ago, but the computer at remote location is offline now. It was a relatively trivial parser of upsc output executed from crontab, to (re)start drivers if deemed stuck. Current NUT master includes logic to kill old stuck drivers more aggressively when a new instance starts, which was added to make scripts like that more useful.

ansred, 2022-08-05:
That seems to be fixed the issue. I think we are good! Thanks for the help and closing

closed 2022-08-05 by ansred

jimklimov, 2022-08-09:
Cool! In the meanwhile, I found a copy of that script I had in mind - I had it committed :)

"As is" it is quite Solaris/illumos specific in what it does to nuke the USB connection and restart it really from scratch, but the general canvas can be adapted to Linux easily: https://github.com/networkupstools/nut/blob/master/scripts/Solaris/reset-ups-usb-solaris.sh.sample

fips21, 2024-10-15:
You can fix random disconnects by adding `pollonly = "enabled"` in the driver definition in /etc/nut/ups.conf.