networkupstools/nut #1533: on battery event missed if it happens before boot?

jmcisvt, 2022-07-29:
I have NUT installed and working on an openSUSE 15.3 server with an APC 1500AVR. Recently we had a power outages with several events over a few hours. The first event everything worked right, the battery ran down, the server shutdown properly. Then an hour later the power came back on but only for 10 minutes. In the logs I can see the server booted up and ran until the power went out again 10 minutes later. It went down hard. No shutdown. The battery was surely still too rundown to power the system. 

But there are no NUT events in the log for that session. Shouldn't there be an ON battery and LOW battery event?

jimklimov, 2022-07-30:
One guess would be that NUT driver was set up to poll (not go via interrupts), and by default does so every 30 sec, at least until power is critical (arguably OL LB should mean it too). Perhaps UPS took less than half a minute to expire? Did NUT driver even start and initially walk the device, to serve the data by then?

Many smarter UPSes have a setting to not power the load until they can guarantee a runtime long enough to facilitate a safe shutdown (so say an hour of downtime while charging when wall-power returns), and tell NUT to "FSD" the machine if you started manually too soon. Just because of scenarios like yours. Trade-offs...

jmcisvt, 2022-08-03:
Nvm, I think I figured it out. Nothing to do with NUT. The server has 2 power supplies. One of them was plugged directly into the wall! Looking more carefully at the boot logs I see NUT did not establish communications with the UPS on the 2nd boot, which makes sense.

closed 2022-08-03 by jmcisvt