networkupstools/nut #1535: [Legrand Keor SP 1000] Incorrect output voltage shown

D4rk4, 2022-07-31:
430v vs expected 230v

```
root@rpi-ups-office:~# upsc ups-office
Init SSL without certificate database
battery.charge: 100
battery.runtime: 1507
battery.voltage: 27.2
device.mfr: Legrand  
device.model: Legrand UPS  
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 10
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Legrand UPS
driver.parameter.productid: 0032
driver.parameter.synchronous: auto
driver.parameter.vendor: Legrand
driver.parameter.vendorid: 1CB0
driver.version: 2.8.0
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000109)
input.voltage: 239.0
output.voltage: 430.0
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 27
ups.mfr: Legrand  
ups.model: Legrand UPS  
ups.productid: 0032
ups.status: OL CHRG
ups.vendorid: 1cb0

```

```
root@rpi-ups-office:~# dpkg -l | grep ii | grep UPS
ii  libnutscan1:arm64                    2.7.4-13                         arm64        network UPS tools - scanner library
ii  libnutscan2:arm64                    2.8.0-1~exp1                     arm64        network UPS tools - scanner library
ii  libupsclient4:arm64                  2.7.4-13                         arm64        network UPS tools - client library
ii  libupsclient6:arm64                  2.8.0-1~exp1                     arm64        network UPS tools - client library
ii  nut                                  2.8.0-1~exp1                     all          network UPS tools - metapackage
ii  nut-client                           2.8.0-1~exp1                     arm64        network UPS tools - clients
ii  nut-server                           2.8.0-1~exp1                     arm64        network UPS tools - core system
```

jimklimov, 2022-07-31:
I wonder if a build vs libusb-1.0 (compared to 0.1 for older NUT packages) had to do with it, or some other changes? Maybe some mayhem with LogMin/LogMax correction, or getting this UPS interpreted as a 3-phase?..

There was recently a report about CPS HID not reporting a voltage item where older NUT does still.

jimklimov, 2023-06-26:
FWIW, linking to #1973 and #439 as possibly related issues (although not in CPS HID).

xorbug, 2023-09-21:
I recently started using nut with a Legrand Keor SP 800 and noticed that some reported values are wrong as well (as shown by `upsc`):

```
battery.charge: 100
battery.charge.warning: 30
battery.runtime: 2811
battery.voltage: 13.5
device.mfr: Legrand  
device.model: Legrand UPS  
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Legrand UPS
driver.parameter.productid: 0032
driver.parameter.synchronous: auto
driver.parameter.vendor: Legrand
driver.parameter.vendorid: 1CB0
driver.version: 2.8.0
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 175.0 <--- ~230 expected
output.voltage: 2160.0 <--- ~230 expected
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0 <--- current load expected (at least when OB i guess, but not the case)
ups.mfr: Legrand  
ups.model: Legrand UPS  
ups.productid: 0032
ups.realpower.nominal: 800
ups.status: OL CHRG
ups.vendorid: 1cb0
```

jimklimov, 2024-12-16:
Did anyone explore implementing a `fix_report_desc` sort of method for `drivers/legrand-hid.c`? Examples can be seen in CPS, as recently touched by PR #2717

reitermarkus, 2026-10-26:
In my case (Legrand Keor PDU 800VA, NUT version 2.8.3), the output voltage is too low instead of too high:

```
output.voltage: 25
```