networkupstools/nut #1539: Fry's Electronics/MEC0003 on Home Assistant Addon (Pi 4) , all 0 value. umm how i to do to get a value like UPSmart 😭

boyphongsakorn, 2022-08-07:
my settings
```
- config:
    - vendorid = "0001"
    - productid = "0000"
    - pollinterval = 10
  driver: nutdrv_qx protocol=hunnox
  name: myups
  port: auto
```

my logs
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.11.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 8.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2022.8.1
 Home Assistant Supervisor: 2022.07.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
cont-init: info: running /etc/cont-init.d/02-set-timezone.sh
[00:03:37] INFO: Configuring timezone
cont-init: info: /etc/cont-init.d/02-set-timezone.sh exited 0
cont-init: info: running /etc/cont-init.d/nut.sh
[00:03:38] INFO: Setting mode to netserver...
[00:03:38] INFO: Connected USB devices:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0001:0000 Fry's Electronics MEC0003
Bus 001 Device 003: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[00:03:38] INFO: Generating /etc/nut/upsd.users...
[00:03:39] INFO: Configuring user: hassio
[00:03:39] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[00:03:41] INFO: Configuring Device named myups...
[00:03:41] INFO: Starting the UPS drivers...
Using protocol: Q1 0.07
Can't autodetect number of battery packs [-1/0.00]
Battery runtime will not be calculated (runtimecal not set)
Network UPS Tools - UPS driver controller 2.7.4
[00:03:41] WARNING: UPS Shutdown will shutdown the host
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[00:03:42] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.005289	listening on 0.0.0.0 port 3493
   0.006654	Connected to UPS [myups]: nutdrv_qx-myups
[00:03:42] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000654	Using power down flag file /etc/killpower
   0.000984	UPS: myups@localhost (master) (power value 1)
   0.001280	debug level is '1'
   0.001307	Warning: running as one big root process by request (upsmon -p)
   0.002032	Init SSL without certificate database
   0.018007	Trying to connect to UPS [myups@localhost]
   0.522636	User upsmonmaster@127.0.0.1 logged into UPS [myups]
   0.023059	Logged into UPS myups@localhost
```

my value
0 (jokes) but really all 0

how i have to do to get real value like UPSmart.

boyphongsakorn, 2022-08-08:
ok i set protocol wrong 
```
- config:
    - vendorid = "0001"
    - productid = "0000"
    - pollinterval = 2
    - protocol=hunnox
  driver: nutdrv_qx
  name: myups
  port: auto
```
but `Device not supported` when i set protocol=hunnox,

boyphongsakorn, 2022-08-08:
[image]

UPSmart USB Debug log

boyphongsakorn, 2022-08-09:
i try to build NUT from [this](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu)

and now every driver can not connect what the hell 😭

jimklimov, 2022-08-09:
>    0.000984	UPS: ***@***.*** (master) (power value 1)
>    0.522636	User ***@***.*** logged into UPS [myups]

Try different sub-protocols, might not be a hunnox dialect. ID 0000/0001 is
a very generic no-name no-brand interface chip and the vendor did not care
to uniquely identify themselves.

boyphongsakorn, 2022-08-09:
> Try different sub-protocols, might not be a hunnox dialect. ID 0000/0001 is a very generic no-name no-brand interface chip and the vendor did not care to uniquely identify themselves.
> […](#)

wow ups no name is look so hard
but i think that is q1, UPSmart tell me is Mega(USB) but i try megatec_usb and sub-protocols megatec is not connecting

southalladam, 2022-08-28:
Here's a verbose output:
Bus 001 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0665 Cypress Semiconductor
  idProduct          0x5161 USB to Serial
  bcdDevice            0.02
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              4 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              32

jimklimov, 2022-08-29:
@southalladam : Is this a report fir this issue? IDs are all different...

jimklimov, 2022-09-12:
The bit about `Couldn't open device, some information will be missing` look suspicious - possibly kernel or another program (e.g. another copy of NUT driver) is blocking access to the USB HID device. Make sure your udev rules are applied, and if you're using NUT in a virtualized environment - that the host OS is not grabbing the device.

boyphongsakorn, 2022-09-24:
i think i end with sell this UPS and i buy very sure UPS that work with NUT (like APC)

jimklimov, 2022-10-11:
Should not hurt ro probe it with `nut-scanner` BTW. But IIRC it mostly
relies on IDs...

SlySven, 2022-11-05:
Ah, I've got a PowerCool 1200VA and I have found the UPSmart software {Version **1.9 (20170707)**} does work mostly (but not completely - the temperature is always 29.0 degrees Celsius for example). I can see that I have slightly older firmware in my unit (**V3.7** rather than **V3.8**) but my logs show that there are four different messages being sent out to the UPS but I can only work out what three of them are:
* `80 06 03 09 04 00 66` = i.e. the **Q1** command - gets a response of the form `(235.0 000.0 235.0 031 49.9 27.4 29.0 00001000`
* `80 06 0c 09 04 00 66` = i.e. **I** - gets a response of the form `#                           V3.7      `
* `80 06 0d 09 04 00 66` = i.e. **F** - gets a response of the form `#   .        .     . `
* `80 06 f3 09 04 00 66` = I do not know - gets a response of the form `BL100` - though that does change when the Mains supply is removed to be a number less, e.g. `BL040`

The UPSmart Windoze utility seems to cycle through the commands, one per second, in the sequence: **Q1**, **Q1**, **Q1**, **F**, the unknown one, and then **I** before repeating the cycle over again. @boyphongsakorn 's seems to be doing something similar but not exactly the same.

One thing I do not understand is how the `0x03` third byte in that seven byte sequence gets mapped to stand for "Q1" and likewise for the others - if I got that I might try hacking the code to also send the one i do not know to also be sent.

In practice with the previous NUT build I was trying based on commit 7ccee802 was only sending the **Q1** command and only getting all zeros back:
[buggy_nut_driver_ouput.txt](https://github.com/networkupstools/nut/files/9942989/buggy_nut_driver_ouput.txt)

jimklimov, 2022-11-05:
@SlySven : thanks for investigation (and willingness). Regarding 0x03=Q1, I suppose that stems from mappings like this:
* https://github.com/networkupstools/nut/blob/master/drivers/nutdrv_qx.c#L895-L914

Note that several subdrivers for different vendors/dialects (krauler, fabula, hunnox, fuji, snr) define different sets of such mappings, and ablerex considers "Q1" string in a different manner.

SlySven, 2022-11-05:
:man_facepalming: My config file was wrong - I didn't understand the difference between the `subdriver` and the `protocol` settings:

Now things are working - unfortunately I found the hard way that my distribution's packages (Devuan  4.0 "Chimaera") are too old to support it (it didn't support `hunnox` as a `subdriver` or a `protocol` or the `noscanlangid`) and getting a later one integrated into my system was a bit tricky (involving Duct-tape and symbolic links :laughing: ). However after that the **2.8.0** tagged version seems to work with a `ups.conf` containing:
```
[PowerCool]
        driver="nutdrv_qx"
        subdriver="hunnox"
        desc="PowerCool 1200"
        port="auto"
        vendorid="0001"
        productid="0000"
        bus="004"
        product="MEC0003"
        protocol="hunnox"
        langid_fix=0x0409
        novendor
        norating
        noscanlangid
```

jimklimov, 2022-11-05:
sounds great, thanks for the update

if you have access to distro's recipes (configure script options) you can probably make a drop-in replacement and avoid duct-tape :)

boyphongsakorn, 2022-11-17:
> 2.8.0

wait version 2.8.0 is seen work or better right? 🤔

SlySven, 2022-11-18:
> wait version 2.8.0 is seen work or better right? 🤔

Well it is working for ME with, as I say, my "PowerCool" 1200VA unit - which is (poorly) identified by that Vender/Product ID pair.

boyphongsakorn, 2022-11-19:
> Well it is working for ME with, as I say, my "PowerCool" 1200VA unit - which is (poorly) identified by that Vender/Product ID pair.

hm well i will try 2.8.0 but i don't know how to install 2.8.0 on Raspberry Pi OS ([Ubuntu packages](http://packages.ubuntu.com/nut)) is still 2.7.4

SlySven, 2022-11-20:
> > Well it is working for ME with, as I say, my "PowerCool" 1200VA unit - which is (poorly) identified by that Vender/Product ID pair.
> 
> hm well i will try 2.8.0 but i don't know how to install 2.8.0 on Raspberry Pi OS ([Ubuntu packages](http://packages.ubuntu.com/nut)) is still 2.7.4

True, that is the problem I had - I ended up with a load of symbol links so that the later version I compiled and installed into `/usr/local` was used by the Devuan system scripts - note that I am using Devuan so, by definition, I do not have to worry about `systemd` sticking its oar in:
```sh
stephen@Ripley:~$ ls -l /sbin | grep "^l" | grep local
lrwxrwxrwx 1 root root        24 Nov  5 17:25 upsd -> /usr/local/ups/sbin/upsd
lrwxrwxrwx 1 root root        29 Nov  5 17:26 upsdrvctl -> /usr/local/ups/sbin/upsdrvctl
lrwxrwxrwx 1 root root        26 Nov  5 17:26 upsmon -> /usr/local/ups/sbin/upsmon
lrwxrwxrwx 1 root root        28 Nov  5 17:26 upssched -> /usr/local/ups/sbin/upssched
```
and
```sh
stephen@Ripley:~$ ls -l /usr/bin | grep "^l" | grep local | grep -v root
lrwxrwxrwx 1 stephen stephen       24 Nov  6 04:02 al175 -> /usr/local/ups/bin/al175
lrwxrwxrwx 1 stephen stephen       27 Nov  6 04:02 apcsmart -> /usr/local/ups/bin/apcsmart
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 apcsmart-old -> /usr/local/ups/bin/apcsmart-old
lrwxrwxrwx 1 stephen stephen       30 Nov  6 04:02 apcupsd-ups -> /usr/local/ups/bin/apcupsd-ups
lrwxrwxrwx 1 stephen stephen       25 Nov  6 04:02 bcmxcp -> /usr/local/ups/bin/bcmxcp
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 bcmxcp_usb -> /usr/local/ups/bin/bcmxcp_usb
lrwxrwxrwx 1 stephen stephen       25 Nov  6 04:02 belkin -> /usr/local/ups/bin/belkin
lrwxrwxrwx 1 stephen stephen       28 Nov  6 04:02 belkinunv -> /usr/local/ups/bin/belkinunv
lrwxrwxrwx 1 stephen stephen       27 Nov  6 04:02 bestfcom -> /usr/local/ups/bin/bestfcom
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 bestfortress -> /usr/local/ups/bin/bestfortress
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 bestuferrups -> /usr/local/ups/bin/bestuferrups
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 bestups -> /usr/local/ups/bin/bestups
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 blazer_ser -> /usr/local/ups/bin/blazer_ser
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 blazer_usb -> /usr/local/ups/bin/blazer_usb
lrwxrwxrwx 1 stephen stephen       24 Nov  6 04:02 clone -> /usr/local/ups/bin/clone
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 clone-outlet -> /usr/local/ups/bin/clone-outlet
lrwxrwxrwx 1 stephen stephen       28 Nov  6 04:02 dummy-ups -> /usr/local/ups/bin/dummy-ups
lrwxrwxrwx 1 stephen stephen       25 Nov  6 04:02 etapro -> /usr/local/ups/bin/etapro
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 everups -> /usr/local/ups/bin/everups
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 gamatronic -> /usr/local/ups/bin/gamatronic
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 genericups -> /usr/local/ups/bin/genericups
lrwxrwxrwx 1 stephen stephen       25 Nov  6 04:02 isbmex -> /usr/local/ups/bin/isbmex
lrwxrwxrwx 1 stephen stephen       25 Nov  6 04:02 ivtscd -> /usr/local/ups/bin/ivtscd
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 liebert -> /usr/local/ups/bin/liebert
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 liebert-esp2 -> /usr/local/ups/bin/liebert-esp2
lrwxrwxrwx 1 stephen stephen       30 Nov  6 04:02 masterguard -> /usr/local/ups/bin/masterguard
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 metasys -> /usr/local/ups/bin/metasys
lrwxrwxrwx 1 stephen stephen       27 Nov  6 04:02 mge-shut -> /usr/local/ups/bin/mge-shut
lrwxrwxrwx 1 stephen stephen       28 Nov  6 04:02 mge-utalk -> /usr/local/ups/bin/mge-utalk
lrwxrwxrwx 1 stephen stephen       30 Nov  6 04:02 microdowell -> /usr/local/ups/bin/microdowell
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 microsol-apc -> /usr/local/ups/bin/microsol-apc
lrwxrwxrwx 1 stephen stephen       34 Nov  6 04:02 nutdrv_atcl_usb -> /usr/local/ups/bin/nutdrv_atcl_usb
lrwxrwxrwx 1 stephen stephen       28 Nov  6 04:02 nutdrv_qx -> /usr/local/ups/bin/nutdrv_qx
lrwxrwxrwx 1 stephen stephen       39 Nov  6 04:02 nutdrv_siemens-sitop -> /usr/local/ups/bin/nutdrv_siemens-sitop
lrwxrwxrwx 1 stephen stephen       30 Nov  6 04:02 nut-scanner -> /usr/local/ups/bin/nut-scanner
lrwxrwxrwx 1 stephen stephen       24 Nov  6 04:02 oneac -> /usr/local/ups/bin/oneac
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 optiups -> /usr/local/ups/bin/optiups
lrwxrwxrwx 1 stephen stephen       27 Nov  6 04:02 powercom -> /usr/local/ups/bin/powercom
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 powerpanel -> /usr/local/ups/bin/powerpanel
lrwxrwxrwx 1 stephen stephen       24 Nov  6 04:02 rhino -> /usr/local/ups/bin/rhino
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 richcomm_usb -> /usr/local/ups/bin/richcomm_usb
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 riello_ser -> /usr/local/ups/bin/riello_ser
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 riello_usb -> /usr/local/ups/bin/riello_usb
lrwxrwxrwx 1 stephen stephen       26 Nov  6 04:02 safenet -> /usr/local/ups/bin/safenet
lrwxrwxrwx 1 stephen stephen       23 Nov  6 04:02 skel -> /usr/local/ups/bin/skel
lrwxrwxrwx 1 stephen stephen       27 Nov  6 04:02 snmp-ups -> /usr/local/ups/bin/snmp-ups
lrwxrwxrwx 1 stephen stephen       24 Nov  6 04:02 solis -> /usr/local/ups/bin/solis
[... cut ...]
lrwxrwxrwx 1 stephen stephen       31 Nov  6 04:02 upssched-cmd -> /usr/local/ups/bin/upssched-cmd
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 usbhid-ups -> /usr/local/ups/bin/usbhid-ups
lrwxrwxrwx 1 stephen stephen       29 Nov  6 04:02 victronups -> /usr/local/ups/bin/victronups
```

I note that the later version is being used in Debian (testing) i.e. version 12 "Bookworm" - it is not clear that it will be backported by Debian into the current (stable) i.e. version 11 "Bullseye", but Ubuntu's policies might be different...

balves42, 2022-11-20:
> > Well it is working for ME with, as I say, my "PowerCool" 1200VA unit - which is (poorly) identified by that Vender/Product ID pair.
> 
> hm well i will try 2.8.0 but i don't know how to install 2.8.0 on Raspberry Pi OS ([Ubuntu packages](http://packages.ubuntu.com/nut)) is still 2.7.4

Hi @boyphongsakorn, i had the same problem as you. I couldn't find clear instructions regarding the requirements and and how to build NUT, so i documented it and forked a docker image that now fits my needs. Hopefully it will help you or others that find this comment: https://github.com/balves42/docker-tools Best of luck

jimklimov, 2022-11-21:
> I couldn't find clear instructions regarding the requirements and and how to build NUT, so i documented it...

Sorry to hear that, and a bit puzzling given how extensive the `docs/` directory contents are (and the resulting website content). I understand "getting lost in the maze", but if something is lacking - PRs welcome ;)

Current docs may be distro-agnostic a bit - so do not detail "clearly how to package NUT specifically for X" because X'es come and go too rapidly, and the best instruction for a custom drop-in updated package is to start from parameters in a distro's recipe. Or to forgo that and maybe only reuse same user, group and sysconfdir. 

Regarding "requirements", though, `docs/config-prereqs.txt` does detail build agent setup (packages and all) for many distros.

jimklimov, 2022-11-21:
And there's also NUT Wiki - linking your repo there ;)

balves42, 2022-11-21:
> Sorry to hear that, and a bit puzzling given how extensive the `docs/` directory contents are (and the resulting website content). I understand "getting lost in the maze", but if something is [...]

No sorry needed ;) I wasn't very clear, although the documentation is extensive (which i linked in some parts of my readme) it's exactly like you said, easy to get lost in the maze.
This is not a negative review, i got the information that i needed to get out of the "fry's electronics" rabbit hole and i just wanted to share with the world a possible solution :)

boyphongsakorn, 2022-11-24:
> Hopefully it will help you or others that find this comment: https://github.com/balves42/docker-tools

nut-scanner -U is not work, i can't scan USB Port
how do i scan USB Port using nut-scanner

boyphongsakorn, 2022-11-24:
> Hi @boyphongsakorn, i had the same problem as you. I couldn't find clear instructions regarding the requirements and and how to build NUT, so i documented it and forked a docker image that now fits [...]

oh use config in https://github.com/balves42/docker-tools and install nut 2.8.0 is work but why 🤔

boyphongsakorn, 2022-11-24:
and yeah.... like UPSmart Battery % is not work but nevermind i think i can see battery low on status

[image]

closed 2022-11-26 by boyphongsakorn

Domotica-Ba, 2023-07-27:
In my case this config works

```
devices:
  - config: []
    name: myups
    bus: 1
    desc: UPS Line Interactive Sinusoidale - 1000VA
    driver: nutdrv_qx
    subdriver: hunnox
    protocol: q1
    vendorid: 1
    productid: 0
    product: MEC0003
    langid_fix: 1033
    port: auto
mode: netserver
shutdown_host: false
list_usb_devices: true
```

jimklimov, 2023-08-31:
@Domotica-Ba : asking for the HCL: is there a specific known vendor and model for your UPS? The description "UPS Line Interactive Sinusoidale" matches almost everything on the market ;)