networkupstools/nut #1540: eaton eco ; connected via usb2 to ubuntu server (running on ESXi)

heapxor, 2022-08-07:
hello,
once booted into the latest ubuntu server i got following error

jimklimov, 2022-08-10:
Not sure, might be due to hypervisor (e.g. if the host intercepted the device and grabbed for itself, or some retries to connect do not pass in a timely fashion). I suppose a proper NUT installation would ensure the `udev` rules (or similar) to allow the `nut` run-time user account to access the device node for drivers, but in the log above it seems that the system (kernel) can not grab it.

On second thought, it *might* also mean that the NUT driver did start (did it?) and holds the device connection now, so the system can not take (back?) over for generic USB HID handler. But I wouldn't expect this to be the case some 8sec after boot.

fliespl, 2022-09-02:
I am struggling with same issue while trying to execute command on my CyberPower UPS (attached to proxmox machine).

`upscmd ups test.battery.start.quick` ends up with access denied (login info is correct - cause nut-monitor works correctly.

In the dmesg log:
[  756.168424] usb 3-3: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 161 rq 1 len 2 ret -75

```
root@pve:~# upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

```
root@pve:~# upsc ups@192.168.88.10
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 15
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 5580
battery.runtime.low: 3600
battery.type: PbAcid
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: CPS
device.model: VP700ELCD
device.type: ups
driver.flag.ignorelb: enabled
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 238.0
input.voltage.nominal: 230
output.voltage: 238.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 10
ups.mfr: CPS
ups.model: VP700ELCD
ups.productid: 0501
ups.realpower.nominal: 390
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```