networkupstools/nut #1560: Fix `configure` script for explicit `--with-nut-scanner` toggle

jimklimov, 2022-08-14:
As recently raised on the mailing list, some OS packaging of NUT did not include `nut-scanner` because the build environment did not require `libltdl` and nothing complained with the one used.

Solution to this issue would add an explicit `configure --with-nut-scanner` option (`yes` by default; failing the script if `libltdl` is missing and `yes` is required) to prompt distro maintainers to update build environments or recipes (maybe including an explicit `--without-nut-scanner` option for their next packaging iterations, if their OS lacks or does not want to provide this mechanism).

gdt, 2022-08-14:
Sounds good; as a packager I am happy with options being on or off, and if on and prereqs missing, loud fail.

(As far as "OS lacks or does not want to provide", it's more that packages avoid declared dependencies on things they don't need, just so they aren't forced to be installed unless needed.)

jimklimov, 2022-08-16:
PR posted, in the end I went for `auto` by default, but also tied into existing `--with-all` support (so that can be a blanket "yes" requirement for many features and dependencies by default, from which some packagers may deduct `--without-something`).

@gdt : at least regarding NUT, there are two aspects of "declaring dependencies on things": one is what it needs to get built (everything and a kitchen sink, more so if building with docs), and another is what end-user packages need at run-time. For the latter, NUT guidelines suggest making an horde of packages, grouped by dependencies (e.g. usb drivers, modbus drivers, python X11 desktop apps, etc.) - about a dozen of those. Depending on the OS and packaging tech, an "umbrella" package for "nut" might be delivered, which would install automatically just the sub-packages for which dependencies are present already (e.g. avoiding the niche `nut-driver-powerman` for most people).

Might make sense to define a package for nut-scanner (hard dep on libltdl; recommends libsnmp, libneon, libusb variant it was built against), if not yet addressed :)

gdt, 2022-08-16:
Yes, there is build-time and run-time dependencies, not the same.   And each functionality/option should explain what it needs in INSTALL or something pointed to b that.   And agreed that "--with-all" is useful to force failure.

However, the idea of "depend on everything and then make separate packages" only work for packaging systems that 99% expect users to run binaries.   pkgsrc is source-first, intending for users to build, and thus split packages are harder in that they need to build separately.   It ends up building again with more deps and just cherrypicking binaries.

And "install for which dependencies are present' is contrary to the core notion of "if you install this package, you get X" which implies "and if something is needed, it's installed quietly".  But agreed that things almost nobody wants are best left off.

Right now we have a base package, and then usb/snmp/cgi.   I should perhaps roll usb into base, as usb UPSes seem to be the norm, and the idea that someone's system doesn't need libusb anyway seems rare.

jimklimov, 2022-08-16:
> And "install for which dependencies are present' is contrary to the core notion of "if you install this package, you get X" which implies "and if something is needed, it's installed quietly". But agreed that things almost nobody wants are best left off.

Just to clarify - I don't necessarily disagree, and I know all systems are different and cater to their populations (and vice-versa).

On average end-users installing particular aspects of NUT via packages (even on systems with umbrellas or metapackages) are free to do so by providing an explicit list ("ok, here I want the data server and USB drivers and CGI and upsmon packages, and there I want just an upsmon package, and over there I want just a libupsclient to build my custom project against"). Which is cumbersome for others, who want to say "I already have libsnmp, so install me the NUT umbrella, and have it fulfill any achievable goals - e.g. server, upsmon, and SNMP drivers".

Then there's also an option of building everything, providing the products as one bundle having de-facto run-time dependency on everything, and "recommending" further packages as dependencies. You get the entirety of NUT installed in one shot, but then need to know to install more if some binaries to not run because they refer to missing SO files. Clumsy too, but some systems go for it as their take on simplicity.

Anyhow, diversity is good and challenging too; whatever choice a project makes, someone is happy and someone is not. So as distros (or OS-agnostic packaging projects in your example) tend to collect people of same mind, they are sort of advocates on how to best satisfy their friends' needs :)

As always, "PRs are welcome" -- there are lots of docs that did not get attention they deserve, in a while, and other loose ends that need tying in NUT, and "just one me" on some nights a month. So a write-up with your packaging project's perspective sounds useful (especially as it impacts the many OSes pkgsrc is available for, so users have options beside stock distro packaging).

jimklimov, 2022-08-16:
> Right now we have a base package, and then usb/snmp/cgi.

BTW, of things that are relatively not-niche, with this list it seems you are missing out on devices talking NetXML (MGE/Eaton) and modbus (more and more lately, at least per contributed drivers).

commit 17eeef00 (2022-08-17): configure.ac: add --with-nut-scanner (also as part of --with-all)

Closes: #1560
closed 2022-08-17 by jimklimov