networkupstools/nut #1567: NUT-Monitor: properly handle icon locations on Linux

Perlovka, 2022-08-14:
Currently icon file is being looked up like this:
```
path = QStandardPaths.locate(QStandardPaths.AppDataLocation, os.path.join( "icons", "hicolor", "256x256", "apps", filename ) )
```
But AppDataLocation points to /usr/share/\<app name\>, and there is no directory structure like icons/hicolor/256x256/apps/
Such directory structure is located in /usr/share, which is referenced by GenericDataLocation, so this line should look like this:
```
path = QStandardPaths.locate(QStandardPaths.GenericDataLocation, os.path.join( "icons", "hicolor", "256x256", "apps", filename ))
```
https://github.com/networkupstools/nut/blob/775ef9fe80c2225c62ecbd1f5e9745f669dbf24f/scripts/python/app/NUT-Monitor-py3qt5.in#L223

Perlovka, 2022-08-14:
Patch:
```
--- a/scripts/python/app/NUT-Monitor-py3qt5.in	2022-08-15 00:46:08.376655254 +0300
+++ b/scripts/python/app/NUT-Monitor-py3qt5.in	2022-08-15 00:46:35.723005624 +0300
@@ -220,7 +220,7 @@
         path = os.path.join( os.path.dirname( sys.argv[0] ), "icons", "256x256", filename )
         if os.path.exists(path):
             return path
-        path = QStandardPaths.locate(QStandardPaths.AppDataLocation, os.path.join( "icons", "hicolor", "256x256", "apps", filename ) )
+        path = QStandardPaths.locate(QStandardPaths.GenericDataLocation, os.path.join( "icons", "hicolor", "256x256", "apps", filename ) )
         if os.path.exists(path):
             return path
         raise RuntimeError("Cannot find %s resource %s" % ('icon', filename))
```

jimklimov, 2022-08-15:
Thanks!

CCing @luke-jr who contributed the porting to py3/qt5 : which OS did you use? Is it a typo or platforms-dependent thing? Should this code probe for one location and fail over to another?

Also wondering now where NUT `make install` puts the icons - its own dataroot (/usr/share/nut/icons...) or systemic. And non-NUT package recipes in distria.

luke-jr, 2022-08-15:
I am not sure why I used `AppDataLocation` there. It looks like I am actually patching my own `NUT-Monitor` script in the same way (changing it to use `GenericDataLocation`. It's possible that I fixed it when patching out the "look in program directory" code (the above 4 lines), and simply failed to notice it needed patching in the unmodified code.

>Also wondering now where NUT `make install` puts the icons - its own dataroot (/usr/share/nut/icons...) or systemic. And non-NUT package recipes in distria.

I don't think it installs the Python stuff at all - `scripts/python/Makefile.am` simply includes the files in `EXTRA_DIST`. From what I've seen, Python appears to have its own custom build systems, and I'm not familiar with them. However, the included `README` does say to put it in "`something like /usr/share/pixmaps/`".

>which OS did you use?

Not relevant since I apparently need the same patch, but FWIW I run Gentoo.

jimklimov, 2022-08-15:
Thank you both, will post and merge the PR as proposed, then :)

PR #1572 (2022-08-15): scripts/python/app/NUT-Monitor-py3qt5.in: fix __find_icon_file()...
...to look also for package-installed icons (in OS location)

Thanks to @Perlovka for noticing and proposing a fix, and to @lukejr for confirming it as correct.

Closes: #1567
commit 7c978e66 (2022-08-15): scripts/python/app/NUT-Monitor-py3qt5.in: fix __find_icon_file()...

...to look also for package-installed icons (in OS location)

Thanks to @Perlovka for noticing and proposing a fix, and to @lukejr
for confirming it as correct.

Closes: #1567
closed 2022-08-15 by jimklimov