networkupstools/nut #1569: Eaton 5PX with snmp-ups does not shut down when FSD issued

crowl0gic, 2022-08-14:
I'm running an Eaton 5PX with the Eaton Network M2 card and nut v2.7.4. I noticed that the functionality I can access with the snmp-ups driver via upsrw varies by the MIB specified. 

ups.conf
```
maxretry = 3
[Eaton5PX]
driver=snmp-ups
port=upstest:161
desc = "Eaton5PX"
community=private
snmp_version = v1
mibs=ietf
```
The card is cited by Eaton as being 100% IETF compatible, so I specified `mibs=ietf`. I'm able to shut down the UPS with `upsrw -s ups.timer.shutdown=5 Eaton5PX`

## upsc output for `mibs=ietf`:
```
Init SSL without certificate database
battery.charge: 89.00
battery.runtime: 11280.00
battery.voltage: 53.60
device.mfr: EATON
device.model: Eaton 5PX 1500
device.type: ups
driver.name: snmp-ups
driver.parameter.mibs: ietf
driver.parameter.pollinterval: 2
driver.parameter.port: upstest:161
driver.parameter.snmp_version: v1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: ietf MIB 1.5
driver.version.internal: 0.97
input.current: 0.10
input.frequency.nominal: 60.00
input.phases: 1.00
input.transfer.high: 151.00
input.transfer.low: 89.00
input.voltage: 122.00
input.voltage.nominal: 120.00
output.current: 0.00
output.frequency: 60.00
output.frequency.nominal: 60.00
output.phases: 1.00
output.power.nominal: 1500.00
output.realpower: 0.00
output.realpower.nominal: 1500.00
output.voltage: 122.00
output.voltage.nominal: 120.00
ups.beeper.status: enabled
ups.firmware: 14
ups.firmware.aux: 2.1.5
ups.load: 0.00
ups.mfr: EATON
ups.model: Eaton 5PX 1500
ups.start.auto: yes
ups.status: OL TEST
ups.test.result: done and passed
ups.timer.reboot: 3
ups.timer.shutdown: -1
ups.timer.start: 3
```
Note that "TEST" is always included with the value of `ups.status`
The 3 `ups.timer.(reboot|shutdown|start)` parameters can be modified and appear to be working as expected. 

If I don't specify `mibs`, "pw" is automatically selected by nut. I get some different parameters in the upsc output and no ability to shut down the UPS manually (is this expected?).

### upsc output for `mib=pw` (nut selects this by default):
```
Init SSL without certificate database
battery.charge: 92.00                 
battery.charger.status: floating                              
battery.runtime: 11687.00                    
battery.voltage: 54.00                                                                                                                             
device.mfr: EATON                                                                                                                                  
device.model: Eaton 5PX 1500                                                                                                                       
device.serial: 1234                                                                                                                          
device.type: ups                                                                                                                                   
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: upstest:161
driver.parameter.snmp_version: v1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: pw MIB 0.87
driver.version.internal: 0.97
input.frequency: 59.90
input.phases: 1.00
input.voltage.nominal: 120.00
output.current: 0.00
output.frequency: 59.90
output.frequency.nominal: 60.00
output.phases: 1.00
output.realpower: 0.00
output.voltage: 122.00
output.voltage.nominal: 120.00
ups.firmware: 14
ups.firmware.aux: 2.1.5
ups.load: 0.00
ups.mfr: EATON
ups.model: Eaton 5PX 1500
ups.power: 0.00
ups.power.nominal: 1500.00
ups.realpower.nominal: 1500.00
ups.serial: 1234
ups.start.auto: yes
ups.status: OL
ups.type: normal
```
Note: `ups.start.auto` is the only field that can be written to with the "pw" mib

## Questions
1. The "ietf" mib has more functionality than "pw", even though "pw" is chosen by nut by default. Conversely, some of the values in the upsc output of "ietf" aren't entirely correct. Is this the expected behavior? 
2. If I run `upsmon -c fsd`, my system will shut down (I disabled this for testing purposes by commenting the `SHUTDOWNCMD` parameter out).  However, the UPS won't shut down at all. Will I need to implement something like `upsrw -s ups.timer.shutdown=30 Eaton5PX` in the /bin/upssched-cmd script to shut the UPS down, or am I missing something?

upsd.users
```
[admin]
password = yyy
actions = set
actions = fsd
instcmds = all

[upsmon]
password = xxx
actions = fsd
upsmon master
```
upsmon.conf
```
RUN_AS_USER root
MONITOR Eaton5PX@localhost 1 admin yyy master
```

jimklimov, 2022-08-15:
For good measure, Eaton devices may also expose `mge` and/or `pxgx_ups` mapping tables, and IIRC none was a subset of any other. Also there is a netxml-ups driver for these network cards, as well as serial/usb connectivity. And in some cases the inner firmware passed values with different precision into different output comms streams. 

This actually led to discussions for a multi-pathed sort of NUT driver that could combine data of same device over many protocols, but so far nobody wrote it.

crowl0gic, 2022-08-15:
@jimklimov that makes sense, thanks for the explanation

Should the `upsmon -c fsd` command be sufficient to shut off the UPS, or will I need to implement that functionality myself (by way of the "/bin/upssched-cmd" script)?

closed 2022-08-15 by crowl0gic