networkupstools/nut #1571: Eaton 5PX + snmp-ups: upsdrvctl error

crowl0gic, 2022-08-15:
I'm running an Eaton 5PX with the Eaton Network M2 card and nut v2.7.4. Regardless of the mib I choose, the `upsmon -c fsd` fails to shut the UPS down. I ran `/lib/nut/snmp-ups -DDDDD -a Eaton5PX -k` and observed the following with each mib:

The "pw" mib returned:
```
   0.011236     Detected Eaton 5PX 1500 on host upstest:161 (mib: pw 0.87)
   0.011262     su_find_info: "load.off.delay" found
   0.011274     send_to_all: ADDCMD load.off
   0.011288     su_find_info: "load.on.delay" found
   0.011311     send_to_all: ADDCMD load.on
   0.011324     su_find_info: "load.off.delay" found
   0.011337     su_find_info: "load.on.delay" found
   0.011348     send_to_all: ADDCMD shutdown.return
   0.011361     send_to_all: ADDCMD shutdown.stayoff
   0.011383     Initiating UPS shutdown
   0.011393     upsdrv_shutdown...
   0.011402     Unknown template type: shutdown.return
   0.011411     entering su_instcmd(shutdown.return, (null))
   0.011430     su_find_info: "shutdown.return" found
   0.011442     entering nut_snmp_set(1.3.6.1.4.1.534.1.9.6, i, 0)
   0.012722     [Eaton5PX] nut_snmp_set: can't set 1.3.6.1.4.1.534.1.9.6: (noSuchName) There is no such variable name in this MIB.
   0.012745     su_instcmd: cannot set value for shutdown.return
   0.012754     Unknown template type: shutdown.reboot
   0.012761     entering su_instcmd(shutdown.reboot, (null))
   0.012773     su_find_info: unknown info type (shutdown.reboot)
   0.012791     su_instcmd: shutdown.reboot unavailable
   0.012802     Unknown template type: load.off.delay
   0.012815     entering su_instcmd(load.off.delay, (null))
   0.012828     su_find_info: "load.off.delay" found
   0.012839     entering nut_snmp_set(1.3.6.1.4.1.534.1.9.1, i, 30)
   0.013870     [Eaton5PX] nut_snmp_set: can't set 1.3.6.1.4.1.534.1.9.1: (noSuchName) There is no such variable name in this MIB.
   0.013893     su_instcmd: cannot set value for load.off.delay
   0.013905     Shutdown failed!
```
Note: Looks identical to "pxgx_ups"

The "pxgx_ups" mib returned:
```
   0.011236     Detected Eaton 5PX 1500 on host upstest:161 (mib: pxgx_ups 0.87)                                                                                                                                                                                                                       
   0.011263     su_find_info: "load.off.delay" found                                                                                                                                                                                                                                                   
   0.011275     send_to_all: ADDCMD load.off                                                                                                                                                                                                                                                           
   0.011289     su_find_info: "load.on.delay" found                                                                                                                                                                                                                                                    
   0.011305     send_to_all: ADDCMD load.on                                                                                                                                                                                                                                                            
   0.011317     su_find_info: "load.off.delay" found                                                                                                                                                                                                                                                   
   0.011328     su_find_info: "load.on.delay" found                                                                                                                                                                                                                                                    
   0.011359     send_to_all: ADDCMD shutdown.return                                                                                                                                                                                                                                                    
   0.011376     send_to_all: ADDCMD shutdown.stayoff                                                                                                                                                                                                                                                   
   0.011392     Initiating UPS shutdown                                                                                                                                                                                                                                                                
   0.011405     upsdrv_shutdown...                                                                                                                                                                                                                                                                     
   0.011416     Unknown template type: shutdown.return                                                                                                                                                                                                                                                 
   0.011429     entering su_instcmd(shutdown.return, (null))                                                                                                                                                                                                                                           
   0.011441     su_find_info: "shutdown.return" found                                                                                                                                                                                                                                                  
   0.011453     entering nut_snmp_set(1.3.6.1.4.1.534.1.9.6, i, 0)                                                                                                                                                                                                                                     
   0.012826     [Eaton5PX] nut_snmp_set: can't set 1.3.6.1.4.1.534.1.9.6: (noSuchName) There is no such variable name in this MIB.                                                                                                                                                                     
   0.012848     su_instcmd: cannot set value for shutdown.return
   0.012858     Unknown template type: shutdown.reboot
   0.012865     entering su_instcmd(shutdown.reboot, (null))
   0.012877     su_find_info: unknown info type (shutdown.reboot)
   0.012891     su_instcmd: shutdown.reboot unavailable
   0.012900     Unknown template type: load.off.delay
   0.012909     entering su_instcmd(load.off.delay, (null))
   0.012923     su_find_info: "load.off.delay" found
   0.012935     entering nut_snmp_set(1.3.6.1.4.1.534.1.9.1, i, 30)
   0.014074     [Eaton5PX] nut_snmp_set: can't set 1.3.6.1.4.1.534.1.9.1: (noSuchName) There is no such variable name in this MIB.
   0.014096     su_instcmd: cannot set value for load.off.delay
   0.014105     Shutdown failed!
```

The "ietf" mib returned:
```
   0.011796     Detected Eaton 5PX 1500 on host upstest:161 (mib: ietf 1.5)
[... 1 lines cut ...]
   0.011827     su_find_info: unknown info type (load.on.delay)
   0.011835     su_find_info: unknown info type (load.off.delay)
   0.011843     Initiating UPS shutdown
   0.011852     upsdrv_shutdown...
   0.011859     Unknown template type: shutdown.return
   0.011867     entering su_instcmd(shutdown.return, (null))
   0.011879     su_find_info: unknown info type (shutdown.return)
   0.011888     Unknown template type: load.on.delay
   0.011895     entering su_instcmd(load.on.delay, (null))
   0.011905     su_find_info: unknown info type (load.on.delay)
   0.011912     su_instcmd: load.on.delay unavailable
   0.011917     Unknown template type: shutdown.reboot
   0.011926     entering su_instcmd(shutdown.reboot, (null))
   0.011936     su_find_info: unknown info type (shutdown.reboot)
[... cut ...]
   0.011956     entering su_instcmd(load.off.delay, (null))
   0.011966     su_find_info: unknown info type (load.off.delay)
   0.011973     su_instcmd: load.off.delay unavailable
   0.011979     Shutdown failed!
```

I can issue `upsrw -s ups.timer.shutdown=X UPS` from the "ietf" mib and it shuts the UPS down successfully. So communication is good, the shutdown functionality is present, but snmp-ups seems to be missing something.

Any thoughts or suggestions on how I could get this to work? Thanks in advance

jimklimov, 2022-08-15:
One thing I forgot to mention in earlier issue #1569 on this subject: there were quite a few changes since NUT v2.7.4 release several years ago, and many were related to improving Eaton hardware support. Not sure quickly if shutdowns were addressed specifically, but are you in position to try NUT v2.8.0 or better yet Git HEAD (probably as a custom build)?

Note that it may suffice to just build NUT with prerequisites and run its driver and upsd from the build area and some envvars to point to custom config and run-time dirs (see e.g. `tests/NIT/nit.sh` for how the test suite sets this up), without overwriting the official installation until you decide you want to.

jimklimov, 2022-08-15:
CCing @aquette @dzomaya

crowl0gic, 2022-12-20:
> Not sure quickly if shutdowns were addressed specifically, but are you in position to try NUT v2.8.0 or better yet Git HEAD (probably as a custom build)?

@jimklimov I compiled v2.8.0 and although the driver's shutdown command (`/lib/nut/snmp-ups -DDDDD -a myups -k`) doesn't work, I used upscmd to issue a delayed shutdown via the `pw` mibs in /bin/upssched-cmd. (As a note, all available commands for `pw` show up via upscmd, while `ietf` seems to favor upsrw. `pw` appears to offer slightly more functionality). 

### /etc/nut/ups.conf
```
[myups]
driver=snmp-ups
port=ups:161
desc = "Eaton5PX"
community=private
snmp_version = v1
mibs=pw
offdelay=60
```

### snmp-ups output
```
   0.027584     [D3] Entering nut_snmp_get_str()                                                                                                   
   0.027620     [D3] nut_snmp_get(.1.3.6.1.2.1.1.6.0)                                                                                              
   0.027656     [D3] nut_snmp_walk(.1.3.6.1.2.1.1.6.0)                                                                                             
   0.027692     [D4] nut_snmp_walk: max. iteration = 1                                                                                             
   0.030114     [D2] Using IETF-MIB default to get and publish sysLocation for device.location (once)               
   0.030176     [D5] send_to_all: SETINFO device.location ""                                                                                       
   0.030232     [D2] entering su_setOID(setvar, ups.delay.shutdown, 60)                                                                            
   0.030288     [D3] su_find_info: unknown info type (ups.delay.shutdown)                                                                          
   0.030334     [D2] su_setOID: info element unavailable ups.delay.shutdown                                                                        
   0.030380     Initiating UPS shutdown                                                                                                            
   0.030425     [D1] upsdrv_shutdown...                                                                                                            
   0.030473     [D2] entering su_setOID(setvar, ups.delay.shutdown, 60)                                                                            
   0.030525     [D3] su_find_info: unknown info type (ups.delay.shutdown)                                                                          
   0.030570     [D2] su_setOID: info element unavailable ups.delay.shutdown                                                                        
   0.030615     [D2] entering su_setOID(instcmd, shutdown.return, (null))                                                                          
   0.030666     [D3] su_find_info: "shutdown.return" found                                                                                         
   0.030712     [D1] su_setOID: cannot execute command 'shutdown.return': value is not a number!                                                   
   0.030758     [D2] entering su_setOID(instcmd, shutdown.reboot, (null))                                                                          
   0.030809     [D3] su_find_info: unknown info type (shutdown.reboot)                                                                             
   0.030855     [D2] su_setOID: info element unavailable shutdown.reboot                                                                           
   0.030900     [D2] entering su_setOID(instcmd, load.off.delay, (null))                                                                           
   0.030951     [D3] su_find_info: "load.off.delay" found                                                                                          
   0.030996     [D1] su_setOID: cannot execute command 'load.off.delay': a provided or default value is needed!                                    
   0.031040     Shutdown failed!
```

Salvoxia, 2023-10-23:
I'd like to give this a bump, as I'm running into this very same issue with an Eaton 9PX and Network M2 management card using NUT v2.8.0.

jimklimov, 2023-10-23:
CC @aquette @dzomaya