networkupstools/nut #1588: Supported Eaton 5E650iUSB seems to be... not supported

mesmariusz, 2022-08-19:
Hi. 

Just look here -> https://forum.openmediavault.org/index.php?thread%2F44293-nut-scanner-not-available-nut-package-omv-not-possible-to-scan-connected-usb-ups%2F#post321934

It looks Eaton 5E650iUSB should be supported:

[image]

and the config should be:

driver = usbhid-ups
port = auto

But is not supported in a fact:

[image]

[image]

What do do? Could you support me? I bought this UPS especially for NUT. Because I found it is supported on your list :-(

Thank you and best regards
Mariusz

clepple, 2022-08-19:
Can you please post the following:
- output of `ls -l /dev/bus/usb/002/005` (if you have moved the UPS since the previous screenshot, adjust based on the current Bus and Device numbers in `lsusb`)
- OS version information (since few people working on NUT use OMV, more details about the underlying Debian version would be helpful)

jimklimov, 2022-08-19:
Also a question if it is possible for you to build and install custom NUT from sources, which fixes some bugs and enhances troubleshooting possibilities.

Also note some Linux distros had and later removed "quirks" for vendorid 0x0463 devices, you might be impacted by their presence or absence depending on firmware.