networkupstools/nut #1601: Error: Driver not connected

wkobiela, 2022-08-20:
Hi, I have a problem running nut on Ubuntu20:

```
ups.conf
[ups]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0501"
        product = "VP700ECD"
        vendor = "CPS"
        bus = "002"
```
```
$ cat /etc/nut/nut.conf
MODE=standalone
```
```
$ cat /etc/nut/upsd.conf
LISTEN localhost 3493
```

```
$ lsusb
Bus 002 Device 002: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
```

```
$ upsc ups@localhost
Init SSL without certificate database
Error: Driver not connected
```

```
$ sudo /lib/nut/usbhid-ups -a ups -u root -DD
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.000390     upsdrv_initups...
   0.262388     Checking device (0764/0501) (002/002)
   0.391959     - VendorID: 0764
   0.391979     - ProductID: 0501
   0.391983     - Manufacturer: CPS
   0.391987     - Product: VP700ECD
   0.391991     - Serial Number: unknown
   0.391996     - Bus: 002
   0.392000     - Device release number: 0001
   0.392004     Trying to match device
   0.392103     Device does not match - skipping
```

I am running out of idead how to run it, what else can I do?

jimklimov, 2022-08-21:
I don't see `VP700ECD` and `CPS` strings in `lsusb` output; can you try commenting them away in `ups.conf` so it would only try to match by numeric `vendorid` and `productid`?

A higher debug verbosity level (more `-D` options) can show more details into what it is probing and what did not match. Current codebase may be more informative about this (release 2.7.4 is rather aged, even if until recently it was the latest), so you might see better detail with a custom build from git.

Just in case, note that per #1512 there may be a regression for CPS support in 2.8.0/master codebase that we are trying to make sense of.

wkobiela, 2022-08-21:
Okay, so I changed `ups.conf`:

```
maxretry = 3
[ups]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0501"
        # product = "VP700ECD"
        # vendor = "CPS"
        bus = "002"
```

Tried now `$ sudo /lib/nut/usbhid-ups -a ups -u root -DDD` and now device matches, but still unable to get HID descriptor
```
0.262200     Checking device (0764/0501) (002/002)
   0.278970     - VendorID: 0764
   0.278988     - ProductID: 0501
   0.278991     - Manufacturer: CPS
   0.278994     - Product: VP700ECD
   0.278997     - Serial Number: unknown
   0.278999     - Bus: 002
   0.279002     - Device release number: 0001
   0.279005     Trying to match device
   0.279042     Device matches
   0.279081     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.280024     Unable to get HID descriptor (error sending control message: Protocol error)
   0.280063     HID descriptor length (method 1) -1
   0.280101     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 30 02
   0.280135     HID descriptor length (method 2) 560
   0.280141     HID descriptor length 560
   0.281017     Unable to get Report descriptor: Protocol error
```

wkobiela, 2022-08-21:
Tried to build nut from source, now have some permissions error with `libusb`

```
~/nut$ sudo drivers/upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0-431-g90093d281
Network UPS Tools - Generic HID driver 0.49 (2.8.0-431-g90093d281)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
Driver failed to start (exit status=1)
```

jimklimov, 2022-08-21:
That looks like you had not `configure`'d the user account to be same as in packaging. For a quick test, you can set the `root` option in `ups.conf` so driver won't drop privileges if started as root.

wkobiela, 2022-08-21:
Added `user = root` and tried to run `drivers/upsdrvctl start` as root, or with sudo but still libusb1 ha insufficient permissions

jimklimov, 2022-08-21:
For the later tests, are you sure earlier drivers are not running and holding the device? 

Also check if your udev.rules (or hwdb) got applied and mention this vid/pid pair so the OS does not hold it as a HOD device?..

jimklimov, 2022-08-22:
Also, adding some more `-D` flags to driver startup should expose more details (which devices it saw and why it rejected them).

wkobiela, 2022-08-22:
Yes, host is getting info ok - it is unRaid and I'm using build-in ups support. I was trying to deploy vm with Ubuntu20. UPS is recognized inside vm, but I just cannot run nut. Probably it just won't work with host and vm at the same time.

jimklimov, 2022-08-22:
If the host is using the USB connection, then probably guest VM can't. With
NUT you can set up one as the driver+data server (+client likely), and
another as just a client with upsmon e.g. for timely safe shutdowns.

Probably the host should be NUT server so it can tell the UPS to cut power
safely for the logical load.