networkupstools/nut #1618: Document protocol-level troubleshooting for USB drivers

jimklimov, 2022-08-25:
Posts by @nbriggs over the past year or so (ranging from #1024 to #1512) were quite enlightening about getting hold of and deciphering USB reports byte by byte, and then deciding what in the logic could be wrong (sloppy vendors or NUT bugs).

It would be great to summarize that wisdom into a developer documentation file among NUT sources, with general explanation and a how-to regarding detection of broken LogMin/LogMax processing.

jimklimov, 2022-08-25:
And while at it, a common question is why a driver does not report certain readings - some may be addressed by gen-usb-subdriver script (comparing its output with mappings in existing driver), but maybe something also hidden in reports we just don't recognize?

nbriggs, 2022-08-25:
This is not the actual documentation you need!  But, as someone who was new to the the process of debugging a USB/HID connected unit where it turned out to be a NUT bug (64/32 bit sensitivity issues in the code), and then helping out with a couple of other problems that turned out to be issues with the USB HID report descriptors the UPS units provide... this is how I'd go about it today:

1. turn the NUT USB driver debug level up to 5
2. look for reports from the driver of any problems it has already detected (LogMin/Max, report descriptor fixups)
3. get/build/install [regina-rexx](https://regina-rexx.sourceforge.io)
4. get/install [HIDRDD](https://github.com/abend0c1/hidrdd)
5. extract the hex dump of the report descriptor from USB driver output from 1. above, and run it through HIDRDD
6. Look at the HIDRDD output, with reference to any documents related to your device and the USB/HID power devices class available in https://networkupstools.org/ups-protocols.html
7. Especially look for inconsistencies in the RD between the min/max (logical and physical) values, the sizes of the report fields they apply to, the expected physical values (e.g., supply and output voltages, over-voltage/under-voltage transfer points, ...)
8. If you're seeing unexpected values for particular variables look at the raw data that is being sent, decide whether it makes sense in the context of the logical and physical min/max values from the report descriptor.
9. read the NUT code, tracing through how each value gets processed looking for where the result deviates from expectations...

jimklimov, 2022-08-27:
Thanks, even that is a decent starting content to nurture later =)

jimklimov, 2022-08-29:
Will post initial version of the doc chapter as part of PR #1635 - thanks!

closed 2022-08-30 by jimklimov