networkupstools/nut #1623: Salicru SPS One - Voltronic-QS H-variant ?

saper-2, 2022-08-26:
Hello,

I have got recently UPS: Salicru SPS ONE 900VA (they were super-cheap :satisfied: "box opened, returned by client" ).
I have slapped a small [program](https://github.com/saper-2/hidups-comm-test) in C#.net (because I don't stand java&http interface that came from viewPower software :nauseated_face: )... 

Anyway I tried to talk to it, and I found out (*also after analyzing a bunch of wireshark-ed packets from usb*😅), that the ups respond for ```M``` command with ```H``` protocol :thinking: - but it's almost identical to ```V``` variant (except the ```QI``` and it don't report temperature). 

I did not test only "shutdown" commands (because I have ups hooked already to my PC and I don't want to instantly kill my PC 🙃  - but I ordered one more so I'll do some testing on that one when it arrive in next week - I'll be updating my github and I'll put here info too).

jimklimov, 2022-08-26:
FWIW, did you try it with recent NUT (e.g. custom build from Github sources) and a Salicru contributed subdriver?

saper-2, 2022-08-26:
Hi,

No I didn't, but I'll be trying with next ups - it's going to work with raspberry pi. I just wanted for now to give a bit of info about that ```H``` protocol variation.

And it took me a while to find-out that you have GitHub repo 😅

saper-2, 2022-09-03:
Hello, a little update.

I did get 2nd ups, so I could experiment more - seems like it accept all command for ```V``` protocol and respond to them with expected results. Maybe this H it's just for missing (empty, only: ```--.-```) temperature readings in ```QS```.

So UPS: Salicru SPS ONE 900VA , I can say now that it use: ```Voltronic-QS V-protocol``` .

I've also updated a bit my repo with short description of proto.

I have also manage to build nut from git, here is a bit debug, driver does fallback to mustek protocol (is it so called ```blazer_usb```? because I did not find it in docs on nut webpage)
```
root@raspberrypi:~ # /usr/local/ups/bin/nutdrv_qx -a spsone -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (Windows-v2.8.0-alpha3-801-g713da62be)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.005852     [D1] upsdrv_initups...
   0.031697     [D2] Checking device 1 of 6 (174C/1153)
   0.032009     [D1] Failed to open device (174C/1153), skipping: Access denied (insufficient permissions)
   0.032196     [D2] Checking device 2 of 6 (1D6B/0003)
   0.032408     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.032587     [D2] Checking device 3 of 6 (17EF/6053)
   0.032795     [D1] Failed to open device (17EF/6053), skipping: Access denied (insufficient permissions)
   0.032975     [D2] Checking device 4 of 6 (0665/5161)
   0.040664     [D2] - VendorID: 0665
   0.040857     [D2] - ProductID: 5161
   0.041028     [D2] - Manufacturer: HID UPS
   0.041199     [D2] - Product: HID UPS
   0.041370     [D2] - Serial Number: HID UPS
   0.041540     [D2] - Bus: 001
   0.041710     [D2] - Device: unknown
   0.041882     [D2] - Device release number: 0002
   0.042051     [D2] Trying to match device
   0.042223     [D3] match_function_regex: matching a device...
   0.042544     [D2] Device matches
   0.042728     [D2] Reading first configuration descriptor
   0.042930     [D3] libusb_kernel_driver_active() returned 0
   0.043171     [D2] Claimed interface 0 successfully
   0.043363     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.045430     [D3] send: QGS
   0.302956     [D3] read: QGS
   0.303223     [D2] qx_process_answer: short reply (input.voltage)
   0.304209     [D3] send: QGS
   0.431007     [D3] read: QGS
   0.431214     [D2] qx_process_answer: short reply (input.voltage)
   0.432968     [D3] send: QGS
   0.687009     [D3] read: QGS
   0.687194     [D2] qx_process_answer: short reply (input.voltage)
   0.689748     [D3] send: M
   0.815048     [D3] read: H
   0.815273     [D2] voltronic_qs_protocol: invalid protocol [H]
   0.816514     [D3] send: M
   1.071052     [D3] read: H
   1.071240     [D2] voltronic_qs_protocol: invalid protocol [H]
   1.073293     [D3] send: M
   1.199074     [D3] read: H
   1.199259     [D2] voltronic_qs_protocol: invalid protocol [H]
   1.202058     [D3] send: M
   1.455102     [D3] read: H
   1.455327     [D2] voltronic_qs_hex_protocol: invalid protocol [H]
   1.456839     [D3] send: M
   1.583114     [D3] read: H
   1.583301     [D2] voltronic_qs_hex_protocol: invalid protocol [H]
   1.585606     [D3] send: M
   1.839151     [D3] read: H
   1.839342     [D2] voltronic_qs_hex_protocol: invalid protocol [H]
   1.840386     [D3] send: QS
   2.607256     [D3] read: (235.0 235.0 235.0 000 50.0 13.6 --.- 00001001
   2.607536     Using protocol: Mustek 0.07
   2.607753     [D1] upsdrv_initinfo...
   2.609232     [D3] send: QS
   3.375338     [D3] read: (235.0 235.0 235.0 000 50.1 13.6 --.- 00001001
   3.375705     [D2] ups_infoval_set: non numerical value [ups.temperature: --.-]
   3.378072     [D3] send: F
   3.887422     [D3] read: #230.0 003 12.00 50.0
   3.888883     [D3] send: I
   4.015436     [D3] read: I
   4.015624     [D2] qx_process_answer: short reply (device.mfr)
   4.015804     [D2] qx_process_answer: short reply (device.model)
   4.016018     [D2] qx_process_answer: short reply (ups.firmware)
   4.016307     No values for battery high/low voltages
   4.016523     Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   4.016718     Battery runtime will not be calculated (runtimecal not set)
   4.016920     [D1] upsdrv_updateinfo...
   4.017099     [D1] Quick update...
   4.019648     [D3] send: QS
   4.783496     [D3] read: (235.0 235.0 235.0 000 50.0 13.5 --.- 00001001

Fatal error: unable to create listener socket

bind /var/state/ups/nutdrv_qx-spsone failed: Permission denied

   4.784397     Exiting.
   4.784587     [D1] upsdrv_cleanup...
root@raspberrypi:~ #
```

jimklimov, 2022-09-04:
Last things first:

````
Fatal error: unable to create listener socket bind /var/state/ups/nutdrv_qx-spsone failed: Permission denied
````

seems like your `/var/state/ups` directory is owned by someone other than the `nut` user for privilege de-levation built into the driver (a configure script option). For the tests you can specify user=root to keep its access.

As for `mustek`, both the older blazer_ser/_usb and newer nutdrv_qx (inheriting from blazer and others) use subdrivers that user can request or driver can guess.

saper-2, 2022-09-04:
@jimklimov that last error is the last that I would care about 🙃 , I just grabbed repo to rpi4 , and compiled to see if it works at all (so not a bit I'd care for this, just want to see if driver will try to "kick-in"). As you can see, even paths are a bit strange 😅 ( like ``` /usr/local/ups/bin/``` )

Here is when I specify to use explicitly the ```voltronix-qs```, driver try requesting UPS for protocol , but get in response ```H``` which is unexpected (unsupported) and call it a day 😬 .
```
root@raspberrypi:~ # /usr/local/ups/bin/nutdrv_qx -a spsone -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (Windows-v2.8.0-alpha3-801-g713da62be)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.002080     [D1] upsdrv_initups...
   0.032764     [D2] Checking device 1 of 6 (174C/1153)
   0.032962     [D1] Failed to open device (174C/1153), skipping: Access denied (insufficient permissions)
   0.033029     [D2] Checking device 2 of 6 (1D6B/0003)
   0.033122     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.033190     [D2] Checking device 3 of 6 (17EF/6053)
   0.033282     [D1] Failed to open device (17EF/6053), skipping: Access denied (insufficient permissions)
   0.033345     [D2] Checking device 4 of 6 (0665/5161)
   0.039640     [D2] - VendorID: 0665
   0.039710     [D2] - ProductID: 5161
   0.039769     [D2] - Manufacturer: HID UPS
   0.039828     [D2] - Product: HID UPS
   0.039885     [D2] - Serial Number: HID UPS
   0.039948     [D2] - Bus: 001
   0.040007     [D2] - Device: unknown
   0.040064     [D2] - Device release number: 0002
   0.040122     [D2] Trying to match device
   0.040182     [D3] match_function_regex: matching a device...
   0.040391     [D2] Device matches
   0.040455     [D2] Reading first configuration descriptor
   0.040594     [D3] libusb_kernel_driver_active() returned 0
   0.040709     [D2] Claimed interface 0 successfully
   0.040774     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.040873     [D2] Skipping protocol Voltronic 0.06
   0.042400     [D3] send: M
   0.171517     [D3] read: H
   0.171600     [D2] voltronic_qs_protocol: invalid protocol [H]
   0.173129     [D3] send: M
   0.299568     [D3] read: H
   0.299639     [D2] voltronic_qs_protocol: invalid protocol [H]
   0.301935     [D3] send: M
   0.555606     [D3] read: H
   0.555679     [D2] voltronic_qs_protocol: invalid protocol [H]
   0.555748     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.555806     [D2] Skipping protocol Mustek 0.07
   0.555863     [D2] Skipping protocol Megatec/old 0.07
   0.555920     [D2] Skipping protocol BestUPS 0.06
   0.555976     [D2] Skipping protocol Mecer 0.07
   0.556033     [D2] Skipping protocol Megatec 0.06
   0.556090     [D2] Skipping protocol Zinto 0.06
   0.556146     [D2] Skipping protocol Masterguard 0.02
   0.556203     [D2] Skipping protocol Hunnox 0.01
   0.556255     [D2] Skipping protocol Ablerex 0.01
   0.556311     [D2] Skipping protocol Q1 0.07
   0.556365     Device not supported!
   0.556419     Device not supported!
```

Would be nice if I could specify exact protocol (with variant)... 

--- edit ---
So I did tinkered a little with the source and rewritten a bit this function (sorry, I'm no expert 😉 ):
```C
static int      voltronic_qs_protocol(item_t *item, char *value, const size_t valuelen)
{
        //char buf[SMALLBUF]; // DEV/DEBUG ONLY
        int ret=0;
        //int v = strcasecmp(item->value, "V"); // DEV/DEBUG ONLY
        //int h = strcasecmp(item->value, "H"); // DEV?DEBUG ONLY

        //snprintf(buf, sizeof(buf), "V=%i , H=%i", v,h); // DEV/DEBUGONLY
        //upsdebugx(2, "%s: protocol check: %s ", __func__, buf); //DEV/DEBUG ONLY
[... 66 lines cut ...]
   0.300163     [D3] send: QS
   1.066820     [D3] read: (232.9 232.9 232.9 000 50.1 13.6 --.- 00001001
   1.067283     Using protocol: Voltronic-QS 0.07
   1.067561     [D1] upsdrv_initinfo...
   1.069066     [D3] send: QS
   1.834912     [D3] read: (232.9 232.9 232.9 000 50.1 13.6 --.- 00001001
   1.835331     [D2] ups_infoval_set: non numerical value [ups.temperature: --.-]
   1.837912     [D3] send: F
   2.346979     [D3] read: #230.0 003 12.00 50.0
   2.347425     No values for battery high/low voltages
   2.347741     Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   2.348099     Battery runtime will not be calculated (runtimecal not set)
   2.348399     [D1] upsdrv_updateinfo...
   2.348855     [D1] Quick update...
   2.350733     [D3] send: QS
   3.115112     [D3] read: (232.9 232.9 232.9 000 50.1 13.6 --.- 00001001

Fatal error: unable to create listener socket

bind /var/state/ups/nutdrv_qx-spsone failed: Permission denied

   3.116356     Exiting.
   3.116579     [D1] upsdrv_cleanup...
root@raspberrypi:~ #
```

jimklimov, 2022-09-04:
If it works - cool! Can you post a PR? One nit seen cursorily - I'd rather not drop to `ret=0` always, but just in case the alternative H variant is seen (easier to read and extend later). Also chain with `else if` because current PoC code seems to ignore hits of V variant in the end?

jimklimov, 2022-09-04:
> Would be nice if I could specify exact protocol (with variant)...

What do you mean? You can specify it with `subdriver=...` option effectively. Then it is up to code in that subdriver (some added "quirk" options over time, or reacted to particular vendor/product IDs; others "forked" into new subdrivers) as you found later :)

saper-2, 2022-09-14:
@jimklimov see yourself 😁 , seems like it works:
```bash
root@raspberrypi:/home/pi/nut/hproto/nut-issue-1623/drivers # ./nutdrv_qx -a spsone -DDD -d1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (Windows-v2.8.0-alpha3-803-gbba09ca24)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.001977     [D1] upsdrv_initups...
   0.019194     [D2] Checking device 1 of 4 (1D6B/0003)
   0.019497     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.019684     [D2] Checking device 2 of 4 (0665/5161)
   0.026371     [D2] - VendorID: 0665
   0.026565     [D2] - ProductID: 5161
   0.026739     [D2] - Manufacturer: HID UPS
   0.026913     [D2] - Product: HID UPS
   0.027086     [D2] - Serial Number: HID UPS
   0.027259     [D2] - Bus: 001
   0.027433     [D2] - Device: unknown
   0.027606     [D2] - Device release number: 0002
   0.027777     [D2] Trying to match device
   0.027952     [D3] match_function_regex: matching a device...
   0.028262     [D2] Device matches
   0.028443     [D2] Reading first configuration descriptor
   0.028648     [D3] libusb_kernel_driver_active() returned 0
   0.028897     [D2] Claimed interface 0 successfully
   0.029095     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.029225     [D2] Skipping protocol Voltronic 0.06
   0.031135     [D3] send: M
   0.160386     [D3] read: H
   0.160549     [D2] voltronic_qs_protocol: detected H protocol [H]
   0.161875     [D3] send: QS
   0.928538     [D3] read: (230.8 230.8 230.8 009 50.0 13.6 --.- 00001001
   0.928777     Using protocol: Voltronic-QS 0.08
   0.928992     [D1] upsdrv_initinfo...
   0.930763     [D3] send: QS
   1.696623     [D3] read: (230.8 230.8 230.8 009 50.0 13.6 --.- 00001001
   1.696978     [D2] ups_infoval_set: non numerical value [ups.temperature: --.-]
   1.699621     [D3] send: F
   2.208717     [D3] read: #230.0 003 12.00 50.0
   2.209090     No values for battery high/low voltages
   2.209304     Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   2.209504     Battery runtime will not be calculated (runtimecal not set)
   2.209718     [D1] upsdrv_updateinfo...
   2.209899     [D1] Quick update...
   2.212443     [D3] send: QS
   2.976816     [D3] read: (230.8 230.8 230.8 009 50.0 13.6 --.- 00001001
   2.977096     [D1] upsdrv_updateinfo...
   2.977281     [D1] Quick update...
   2.979308     [D3] send: QS
   3.744935     [D3] read: (230.8 230.8 230.8 009 50.0 13.6 --.- 00001001
   3.745166     [D1] upsdrv_updateinfo...
   3.745346     [D1] Quick update...
   3.748159     [D3] send: QS
   4.641046     [D3] read: (230.8 230.8 230.8 009 50.0 13.6 --.- 00001001
   4.641290     [D3] Entering dstate_dump
battery.charge: 100
battery.voltage: 13.60
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.protocol: voltronic-qs
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: Windows-v2.8.0-alpha3-803-gbba09ca24
[... 86 lines cut ...]
driver.parameter.vendorid: 0665
driver.version: Windows-v2.8.0-alpha3-803-gbba09ca24
driver.version.data: Voltronic-QS 0.08
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.voltage: 16.9
input.voltage.fault: 16.9
output.current.nominal: 3.0
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 228.5
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware.aux: PM-H
ups.load: 10
ups.productid: 5161
ups.status: OB
ups.type: offline / line interactive
ups.vendorid: 0665
   4.642102     [D1] upsdrv_cleanup...
root@raspberrypi:/home/pi/nut/hproto/nut-issue-1623/drivers #
```
Seems like it works 😄 (I have old laptop connected to the ups for tests), 2nd command: I have disconnected ups from AC .

jimklimov, 2022-09-14:
Thanks! I'll keep it open as a reminder to myself

closed 2022-09-14 by jimklimov

saper-2, 2023-06-11:
HI, just wanted as info for people that would look for something in subject of Voltronic-QS and controling ups without using NUT - like me recently: I wanted to turn off beeper but didn't want to install whole NUT so, instead installing&config nut, I wrote those scripts 😂  : https://github.com/saper-2/py-upshid-voltqs

jimklimov, 2023-06-11:
Looks fun, thanks for the link :)

jimklimov, 2024-10-08:
Added both links to nut-website list of projects - began an "Other" category :)

PR #1653 (2022-09-14): drivers/nutdrv_qx_voltronic-qs.c: add support for "H protocol" [#1623]
Closes: #1623

CC @saper-2