networkupstools/nut #1634: Ultra Max 1000SC Device Works with ViewPower but not detected by nut-scanner

southalladam, 2022-08-28:
I have a 1000VA UPS branded as UltraMax which comes bundled with ViewPower monitoring software. I'm currently running a Debian based OS distro (Turnkey Linux), but the following is true of various other Linux distros running on bare-metal, LXC container, docker container and VMs. 

I can connect in Windows with the supplied ViewPower software (it took a few install attempt for this to work), but I want to get this to work in nuts so I can integrate with Home Assistant, for example.

lsusb give me the following:
`Bus 001 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial`

Other PowerWalker devices I know provide the same vendor/product id combination, so this seems to relate to the usb connector rather than the UPS device. 

nut-scanner finds no UPS hardware.

I have used the vendorid = 0665 and productid = 5161 along with the main drivers that support devices that work with ViewPower compatiple devices, but can't find any combination that works.

I have used the man pages for the 3 most likely drivers to support this UPS device, but haven't yet managed to find a combination of driver properties that work for this device. 

Has anyone managed to get nut working with this device, or could someone propose a way to help me find which protocol my device might be using and how I can find the correct combination of settings to make it work?

Many thanks
Adam

jimklimov, 2022-08-29:
"Couldn't open device, some information will be missing" looks suspect. Is some other program/driver/... holding it? In case of Linux, there would be udev rules to hand over the pid:vid from kernel and default HID driver to nut account. In case of usb pass-through, make sure host OS is not holding it.

Which NUT version? 2.7.4 or 2.8.0+/master?

southalladam, 2022-08-29:
Thanks for your reply!

I'm running the latest stable,. 2.8.0.

I've tried a few different hosts, currently running in a privileged LXC
containee in proxmox (latest version).

I also suspect something related to permissions/reserved device.

Will read up on udev to see how to make sure my container has priority.

Any pointers would be greatly appreciated!

Many thanks
Adam

jimklimov, 2022-08-30:
I think there were recent discussions where users detailed their passthrough containers, so gotta look in recently updated issues of 3-4 months maybe. Not sure if they had or addressed permissions issues though. Possibly that was about Dockerizing NUT...

NUT udev rules would usually be installed by packages or `make install` - but limited to the (guest) env where the package is, so host might not know it has to surrender the device.

jimklimov, 2022-08-30:
Maybe this: https://github.com/networkupstools/nut/wiki/Running-NUT-in-an-LXC-container

jimklimov, 2022-08-30:
Also, as you mentioned "bare-metal Linux" above - did you try running a NUT driver program on the Proxmox (Debian) host itself, perhaps as `root` and with `user=root` option so it would retain privileges while testing port access, and higher verbosity to see logged hints as to what else might prevent it from finding the device?

jimklimov, 2022-08-30:
Also, regarding "main drivers that support": per vendor ID, seems the older `blazer_usb` and newer `nutdrv_qx` (should be a superset of blazer and some other Qx protocol drivers) support a `subdriver=cypress` for the dialect. Assuming the problem is in OS permissions (or hardware comms - cable, port, ...) and you find a way to resolve that, these drivers should handle the UPS.

Then, it is generally also entirely possible that a vendor would cut corners so much as to market a new device or firmware talking a different protocol over the same chip IDs. For kicks, try if a query like this reports any USB HID interaction support (when other programs and drivers talking to that device are stopped):
````
sudo usbhid-ups -s testups -DDDDDD -x port=auto -x user=root -x explore -x vendorid=0665 -x productid=5161 -d1
````
If it prints something like
````
...
   0.491407     [D2] Trying to match device
   0.491411     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.491430     [D2] Device matches
   0.491434     [D2] Reading first configuration descriptor
   0.491456     [D2] Claimed interface 0 successfully
   0.557314     [D2] HID descriptor length 909
   3.731304     [D2] Report Descriptor size = 909
   3.731363     Using subdriver: EXPLORE HID 0.2
   3.731368     [D1] 94 HID objects found
   3.772356     [D1] Path: UPS.BatterySystem.Battery.AudibleAlarmControl, Type: Feature, ReportID: 0x20, Offset: 0, Size: 8, Value: 2
...
````
we might be on to something this way...

southalladam, 2022-08-31:
Hello Jim,

Just to confirm that I installed from scratch on a new Debian Stable VM and
made sure that the USB was assigned and this time it worked!

The new nutdrv_qx driver works for me.

I would prefer for this to be running in an LXC container, so I will follow
through with your recommendations on your other emails and try to get it
working that way.

In the meantime, I have all I need for now! It is already integrated with
Home Assistant.

Thank you so much for your help! If I discover anything useful as I work
through this I will contribute to the docs.

I'm likely to leave this as it is for now as I have what I need.

Thanks again!
Adam

jimklimov, 2022-08-31:
Super, great to hear! :)

closed 2022-09-02 by jimklimov

plittlefield, 2023-01-17:
Hi @southalladam 

Would you be kind enough to post your ups.conf here?

I have the same UPS and really struggling even after all this thread.

Here is mine:-

```
[ultramax]
        driver = nutdrv_qx
        subdriver = cypress
        protocol = voltronic
        port = auto
        bus = 003
        ddvice = 002
        vendorid = 0665
        productid = 5161
```

Thanks!

Paully

jimklimov, 2023-01-17:
Does the driver start? It may hit the `ddvice` typo and abort on unknown keyword - does it tell so in logs?..

jimklimov, 2023-01-18:
Oh wait, 2.7.2? That's a bit antique! Seems the option was not considered (nor matched) back then.

Please try building a newer (git master ideally) version. Or maybe your distro serves 2.7.4+? it should have at least some support for that, I hope.

plittlefield, 2023-01-18:
Fair comment!

I’ll try a git master build and report back.

jimklimov, 2023-01-18:
Good luck! Would be great to know if https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests helps a "newcomer" or better wording/detail is needed?

plittlefield, 2023-01-18:
The documentation is very good, but I'm a ways off replacing anything yet :)

I successfully built version 2.8.0 from the git master and tried the same options and config with this error output:-

```
$ ./drivers/nutdrv_qx -a ultramax -x subdriver=cypress -x protocol=voltronic -x port=auto -x bus=003 -x vendorid=0665 -x productid=5161 -DDDDDD
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-Windows-462-gf287126)
USB communication driver (libusb 1.0) 0.44
   0.000000     [D1] Network UPS Tools version 2.8.0-Windows-462-gf287126 configured with flags: --enable-Wcolor --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --enable-inplace-runtime
   0.000033     [D1] debug level is '6'
   0.000899     [D1] Succeeded to become_user(nut): now UID=111 GID=117
   0.000949     [D5] send_to_all: SETINFO device.type "ups"
   0.000987     [D5] send_to_all: SETINFO driver.state "init.device"
   0.001090     [D1] upsdrv_initups...
   0.005601     [D2] Checking device 1 of 8 (1D6B/0001)
   0.005638     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.005644     [D2] Checking device 2 of 8 (1D6B/0002)
   0.005654     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.005660     [D2] Checking device 3 of 8 (1D6B/0001)
   0.005670     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.005681     [D2] Checking device 4 of 8 (1D6B/0001)
   0.005695     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.005752     [D2] Checking device 5 of 8 (1D6B/0002)
   0.005769     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.005777     [D2] Checking device 6 of 8 (1D6B/0001)
   0.005796     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.005804     [D2] Checking device 7 of 8 (0665/5161)
   0.014911     [D2] - VendorID: 0665
   0.014944     [D2] - ProductID: 5161
   0.014953     [D2] - Manufacturer: WCHCH544
   0.014962     [D2] - Product: USB to Serial
   0.014970     [D2] - Serial Number: unknown
   0.014978     [D2] - Bus: 003
   0.014989     [D2] - Device: 002
   0.015000     [D2] - Device release number: 0002
   0.015007     [D2] Trying to match device
   0.015018     [D3] match_function_regex: matching a device...
   0.015053     [D2] Device matches
   0.015061     [D2] Reading first configuration descriptor
   0.015077     [D3] libusb_kernel_driver_active() returned 0
   0.015142     [D2] Claimed interface 0 successfully
   0.015153     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.015171     [D5] send_to_all: SETINFO ups.vendorid "0665"
   0.015217     [D5] send_to_all: SETINFO ups.productid "5161"
   0.016846     [D3] send: QGS
   1.017917     [D3] read: Operation timed out (-7)
   1.017964     [D4] qx_process: failed to preprocess answer [input.voltage]
   1.018901     [D3] send: QGS
   2.019912     [D3] read: Operation timed out (-7)
   2.019947     [D4] qx_process: failed to preprocess answer [input.voltage]
   2.020895     [D3] send: QGS
   3.021904     [D3] read: Operation timed out (-7)
   3.021936     [D4] qx_process: failed to preprocess answer [input.voltage]
   3.022022     [D2] Skipping protocol Voltronic-QS 0.08
   3.022044     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   3.022060     [D2] Skipping protocol Mustek 0.07
   3.022077     [D2] Skipping protocol Megatec/old 0.07
   3.022093     [D2] Skipping protocol BestUPS 0.06
   3.022109     [D2] Skipping protocol Mecer 0.07
   3.022126     [D2] Skipping protocol Megatec 0.06
   3.022148     [D2] Skipping protocol Zinto 0.06
   3.022232     [D2] Skipping protocol Masterguard 0.02
   3.022259     [D2] Skipping protocol Hunnox 0.01
   3.022284     [D2] Skipping protocol Ablerex 0.01
   3.022300     [D2] Skipping protocol Q1 0.07
   3.022314     Device not supported!
   3.022329     Device not supported!
   3.022357     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   3.022373     [D6] upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

How do I use the nut-scanner tool ?

Paully

plittlefield, 2023-01-18:
Sorry, I've just answered my own question :)

```
./tools/nut-scanner/nut-scanner -U
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        product = "USB to Serial"
        vendor = "WCHCH544"
        bus = "003"
        device = "002"
        ###NOTMATCHED-YET###bcdDevice = "0002"
```

Exciting!

So, if I change my config ...

plittlefield, 2023-01-18:
I get this ...

```
 ./drivers/nutdrv_qx -a nutdev1 -DDDDDD
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-Windows-462-gf287126)
USB communication driver (libusb 1.0) 0.44
   0.000000     [D1] Network UPS Tools version 2.8.0-Windows-462-gf287126 configured with flags: --enable-Wcolor --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --enable-inplace-runtime
   0.000065     [D1] debug level is '6'
   0.000905     [D1] Succeeded to become_user(nut): now UID=111 GID=117
   0.000954     [D5] send_to_all: SETINFO device.type "ups"
   0.000992     [D5] send_to_all: SETINFO driver.state "init.device"
   0.001020     [D1] upsdrv_initups...
   0.007206     [D2] Checking device 1 of 8 (1D6B/0001)
   0.007236     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.007241     [D2] Checking device 2 of 8 (1D6B/0002)
   0.007249     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.007254     [D2] Checking device 3 of 8 (1D6B/0001)
   0.007261     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.007266     [D2] Checking device 4 of 8 (1D6B/0001)
   0.007274     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.007283     [D2] Checking device 5 of 8 (1D6B/0002)
   0.007293     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.007299     [D2] Checking device 6 of 8 (1D6B/0001)
   0.007315     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.007321     [D2] Checking device 7 of 8 (0665/5161)
   0.016757     [D2] - VendorID: 0665
   0.016778     [D2] - ProductID: 5161
   0.016790     [D2] - Manufacturer: WCHCH544
   0.016808     [D2] - Product: USB to Serial
   0.016831     [D2] - Serial Number: unknown
   0.016841     [D2] - Bus: 003
   0.016857     [D2] - Device: 002
   0.016879     [D2] - Device release number: 0002
   0.016892     [D2] Trying to match device
   0.016906     [D3] match_function_regex: matching a device...
   0.017017     [D2] Device matches
   0.017031     [D2] Reading first configuration descriptor
   0.017059     [D3] libusb_kernel_driver_active() returned 0
   0.017134     [D2] Claimed interface 0 successfully
   0.017157     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.017182     [D5] send_to_all: SETINFO ups.vendorid "0665"
   0.017204     [D5] send_to_all: SETINFO ups.productid "5161"
   0.018656     [D3] send: QGS
   1.019709     [D3] read: Operation timed out (-7)
   1.019762     [D4] qx_process: failed to preprocess answer [input.voltage]
   1.020694     [D3] send: QGS
   2.021699     [D3] read: Operation timed out (-7)
   2.021734     [D4] qx_process: failed to preprocess answer [input.voltage]
   2.022661     [D3] send: QGS
   3.023652     [D3] read: Operation timed out (-7)
   3.023696     [D4] qx_process: failed to preprocess answer [input.voltage]
   3.024684     [D3] send: M
   4.025694     [D3] read: Operation timed out (-7)
   4.025759     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   4.026630     [D3] send: M
   5.027688     [D3] read: Operation timed out (-7)
   5.027725     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   5.028673     [D3] send: M
   6.029681     [D3] read: Operation timed out (-7)
   6.029726     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   6.030669     [D3] send: M
   7.031671     [D3] read: Operation timed out (-7)
   7.031728     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   7.032660     [D3] send: M
   8.033671     [D3] read: Operation timed out (-7)
   8.033709     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   8.034605     [D3] send: M
   9.035664     [D3] read: Operation timed out (-7)
   9.035703     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   9.036668     [D3] send: QS
[... 78 lines cut ...]
  35.087471     [D3] read: Operation timed out (-7)
  35.087517     [D4] qx_process: failed to preprocess answer [input.voltage]
  35.088462     [D3] send: Q1
  36.089465     [D3] read: Operation timed out (-7)
  36.089514     [D4] qx_process: failed to preprocess answer [input.voltage]
  36.090451     [D3] send: Q1
  37.091420     [D3] read: Operation timed out (-7)
  37.091482     [D4] qx_process: failed to preprocess answer [input.voltage]
  37.092438     [D3] send: Q1
  38.093457     [D3] read: Operation timed out (-7)
  38.093508     [D4] qx_process: failed to preprocess answer [input.voltage]
  38.094441     [D3] send: Q1
  39.095448     [D3] read: Operation timed out (-7)
  39.095496     [D4] qx_process: failed to preprocess answer [input.voltage]
  39.095507     Device not supported!
  39.095516     Device not supported!
  39.095538     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
  39.095629     [D6] upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

...so close I am sure!

Any ideas?

Paul

plittlefield, 2023-01-30:
I have finally fixed it and now have my UltraMax 1000SC working with nut!

I've uploaded all my notes to this pastebin:-

https://pastebin.com/nLdnKPRS

Hope it helps someone.

:-)

Paully

jimklimov, 2023-01-30:
Thanks!

I'm not sure if batt volt min/max of 1/24 are right for you (with current
being 27.2), and IIRC they play a role in current charge level
calculations. 13.6V per pack sounds plausible for a well-charged 12V
nominal.

Also you ought to look into systemd integration rather than hacking
rc.local, if possible. Relying on `sleep` is asking for trouble, generally
:)

Jim

plittlefield, 2023-01-31:
Hi @jimklimov 

Thanks for your reply, much appreciated!

I am a 'nut newbie' so my skills and settings are possibly wrong :)

I got the figure of 24 from the official web page (click on the DETAILS tab) ...

https://ultramax.co.uk/catalog/product/view/id/1708/s/ultra-max-1000sc-usb-1000va-600w-line-interactive-battery-backup-ups-system/category/145/

... so that line is **"Battery Type : 2 x 12volt 7Amps (INCLUDED)"**

Regarding the systemd integration, is there a script for this anywhere or some page I can find it?

Regards,

Paully

jimklimov, 2023-01-31:
For test rebuilds, a most recent cheat-sheet is https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

The configure script should detect if it can install systemd units into your OS, and then `sudo make install` would place the files. You need to activate them like in Wiki samples though.

plittlefield, 2023-02-01:
@jimklimov so, do I change the ups.conf to

default.battery.charge = 27
nominal.battery.charge = 24

What shall I put for the low figure?

Thanks for the systemd hint, I’ll let you know how I get on!

jimklimov, 2023-02-01:
Mixing oranges and flies here :)

"Charge" is percentage of battery (100 means full, 0 means depleted, in-between may be non-linear), as divined by guesstimate or reported by device. It is loosely related to its voltage, with typical 6-cell PbAc 12V+ being close to 100%, and some 10.5V being 0% (can't emit electrons). See e.g. https://en.wikipedia.org/wiki/Comparison_of_commercial_battery_types

What you are after here is "voltage" `override`s (or indeed `default`s, if the device is expected to report the value eventually and you trust that more), e.g.:
````
override.battery.voltage.high: 27.2
override.battery.voltage.nominal: 24.0
override.battery.voltage.low: 21.0
````

For more taxonomy details see https://github.com/networkupstools/nut/blob/master/docs/nut-names.txt

Hope this helps,
Jim

plittlefield, 2023-02-01:
OK!

I have edited my ups.conf and it now looks like this:-

```
$ cat /etc/nut/ups.conf
user = nut
maxretry = 3
[ultramax]
  driver = "nutdrv_qx"
  port = "auto"
  vendorid = "0665"
  productid = "5161"
  product = "USB to Serial"
  vendor = "WCHCH544"
  bus = "003"
  override.battery.voltage.high = 27.2
  override.battery.voltage.nominal = 24.0
  override.battery.voltage.low = 21.0
  override.battery.packs = 2
  runtimecal = 300,100,900,50
  chargetime = 21600
  idleload = 10
```

... and upsc ultramax@localhost reports ...

```
$ upsc ultramax@localhost
battery.charge: 100
battery.packs: 2
battery.runtime: 3000
battery.voltage: 27.40
battery.voltage.high: 27.2
battery.voltage.low: 21.0
battery.voltage.nominal: 24.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 003
driver.parameter.chargetime: 21600
driver.parameter.idleload: 10
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB to Serial
driver.parameter.productid: 5161
driver.parameter.runtimecal: 300,100,900,50
driver.parameter.synchronous: auto
driver.parameter.vendor: WCHCH544
driver.parameter.vendorid: 0665
driver.state: quiet
driver.version: 2.8.0-Windows-542-gccfb356
driver.version.data: Mustek 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.20 (API: 0x1000104)
input.frequency: 50.0
input.voltage: 240.3
input.voltage.fault: 240.3
output.voltage: 240.3
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665
```

Thanks for your help.

I think I can go and buy another one of these UPS units now :-)

Regards,

Paully

plittlefield, 2023-02-01:
My bad on the systemd scripts - they were there all the time - RTFM :-)

```
sudo systemctl start nut-driver-enumerator.service 
sudo systemctl enable nut-driver-enumerator.service 
sudo systemctl start nut-server.service 
sudo systemctl enable nut-server.service 
sudo systemctl start nut-monitor.service 
sudo systemctl enable nut-monitor.service 
sudo ps faux
sudo upsc ultramax@localhost
```

Yay!

Paully

jimklimov, 2023-02-01:
Sounds great! :)

FWIW also note issue #1279 and PR #1652 (currently in review and community testing, think there are some mis-behaviors that hold it back so far, which may get resolved and merged before next NUT release or some time later).

Also, for `systemctl` I think you should `enable` first (to add symlinks) and `start` later. But maybe those links are only important for dependency chains e.g. during boot. Also note that if you fiddle with dependencies, a `disable` removes the symlinks (so you need to both `disable` and then `enable` to remove obsoleted ones and keep intended ones). And `systemctl daemon-reload` to reliably slurp-in new files for the running system, if you edit (or re-install) them.

plittlefield, 2023-02-01:
OK so, after a github master install, the correct sequence is ...

```
sudo systemctl daemon-reload
sudo systemctl enable nut-driver-enumerator.service
sudo systemctl start nut-driver-enumerator.service
sudo systemctl enable nut-server.service
sudo systemctl start nut-server.service
sudo systemctl enable nut-monitor.service
sudo systemctl start nut-monitor.service
sudo ps faux
sudo upsc ultramax@localhost
```

?

jimklimov, 2023-02-01:
yes, should be

plittlefield, 2023-02-01:
Lovely, thanks.

Job, done :-)